Infineon/optiga-trust-m #3: DL_MAX_FRAME_SIZE not respected

creiter32, 2019-07-01:
Hi,

for the nrf5x PAL I need to set a lower level for the `DL_MAX_FRAME_SIZE` defined in [ifx_i2c_config.h:69](https://github.com/Infineon/optiga-trust-m/blob/master/optiga/include/optiga/ifx_i2c/ifx_i2c_config.h#L69), because the hardware doesn't support transfer sizes this large.
However, the macro doesn't seem to have any effect on the number of bytes in one I2C transfer.

I verified this by looking for the error message produced [here](https://github.com/Infineon/optiga-trust-m/blob/master/pal/nrf5x/pal_i2c.c#L270).

For debugging, I think this should also be reproducable on other platforms by adding the same check inside their PAL and setting  `DL_MAX_FRAME_SIZE=250` for example.

ayushev, 2019-07-01:
The Frame configuration will become part of the next release. For the timebeing, I could propose to you to apply a similar fix from the Trust X https://github.com/Infineon/optiga-trust-x/blob/7ae1031b8b7fdc5ce21c0baae5dc377edbd9b3ef/optiga/comms/ifx_i2c/ifx_i2c_config.c#L61-L65
to the Trust M1, can you please confirm, whether this does work for you?

creiter32, 2019-07-01:
Thank you for your quick response, yes this fix ported from the Trust X solves the problem for me.

ayushev, 2019-08-09:
Fixed in the latest release

closed 2019-08-09 by ayushev