Infineon/optiga-trust-m #7: About the Manifest data

toshi-onion, 2019-12-15:
Hi expert,

I'm reading solution reference manual of trust m but I can't understand clarify one contents. it's "Manifest data". What does Manifest data? How does Manifest data use to?

toshi-onion, 2019-12-15:
> Hi expert,
> 
> I'm reading solution reference manual of trust m but I can't understand clarify one contents. it's "Manifest data". What does Manifest data? How does Manifest data use to?

pwiegele, 2019-12-16:
You can find more Information about it the [Solution Reference Manual](https://github.com/Infineon/optiga-trust-m/blob/master/documents/OPTIGA_Trust_M_V1_Solution_Reference_Manual_v1.13.pdf)
> The Manifest is a top level construct that ties all other structures together and is signed by an authorized entity whose identity is represented by a trust anchor installed at the OPTIGA™. The trust anchor is addressed by its unique ID (OID), which is contained in the metadata of the manifest. Manifest consists of the metadata in plain text, the payload binding and the signature over metadata and payload binding.

closed 2019-12-19 by pwiegele