Infineon/optiga-trust-m #8: Optiga Trust M Public Key Generation

AeAscher, 2020-01-16:
Hi,

The chip is able to generate RSA **key pairs** and it is possible to export the private and the public key. Would it be also possible to extract only the public key from an already generated private key, which is stored in a key slot of the Optiga?

Best regards

ayushev, 2020-01-16:
Hi @AeAscher , yes this is possible, if you give `FALSE` to the `export_private_key` parameter of the [optiga_crypt_rsa_generate_keypair](https://github.com/Infineon/optiga-trust-m/wiki/Trust-M-Crypt-API#optiga_crypt_rsa_generate_keypair) function, you get the public key only.

AeAscher, 2020-01-16:
Hi @ayushev,

Ah sorry I know that I can export the public key alone. 
But my question is:
Let’s for example assume I generated a key pair and lost my public key , but the private key is still stored in the Optiga Trust M (in one of the key slots).
Is it then possible to use this specific private key in the Optiga and generate only a new public key or do i have to generate a new private and a new public key?

ayushev, 2020-01-16:
This isn't available, but you can export the public key and store it in one of Arbitrary Data Objects ([here](https://github.com/Infineon/optiga-trust-m/wiki/Data-and-Key-Store-Overview), right bottom corner).
Also is you generate a certificate out of this key pair, you can store it in the certificate storage, this certificate will also include the public key.

AeAscher, 2020-01-16:
Ok thank you very much for your fast answer. Storing the key in the Arbitrary Data Objects is probably the best solution for us.

closed 2020-01-16 by AeAscher

makerstorage, 2021-02-19:
>if you give FALSE to the export_private_key parameter of the optiga_crypt_rsa_generate_keypair function, you get the public key only.

I do not want to generate key each time. I want to get the public key already generated and stored. I also want the private key will be stored safely. Can you help me with the getpublickey function?

ayushev, 2021-02-19:
hi @makerstorage 

do you mean a function to get a corresponding public key of the private key you have in the [first key slot 0xe0f0](https://github.com/Infineon/optiga-trust-m/wiki/Data-and-Key-Store-Overview) ?

makerstorage, 2021-02-19:
I am working on the Arduino lib and looking at the examples. I test the encrypt decrypt [example](https://github.com/Infineon/arduino-optiga-trust-m/tree/master/examples/encryptDecryptData)
Every time I run the code it generates new key pairs. I want to do that once and get the publickey

My project is to encrypt a string in the web server with PHP using the trustM public key. First thing is to get the device public key and store it in the web server (this will be done manually and keys must not change in the field.) 

Server will send an encrypted message to the iot(build with trustM) the Arduino code will decrypt the message with the device private key.