Infineon/optiga-trust-m #9: How to use only TRNG in Truat M.

inot1222, 2020-02-02:
Hi expert,

The chip contains TRNG. I plan to use only this feature.
If only this feature is used, I would be glad if I could just initialize the interface and TRNG.
Even if you initialize the extra functions, the software may be optimized, but I want to know the minimum functional unit first.

ayushev, 2020-02-03:
Hi @inot1222 

for this you need just create a crypt_instance and run the random function. [Here](https://github.com/Infineon/optiga-trust-m/blob/master/examples/optiga/example_optiga_crypt_random.c) is the example.

You can disable all non used modules from the software side by disabling them in the [config file](https://github.com/Infineon/optiga-trust-m/blob/v1.30.885/optiga/include/optiga/optiga_lib_config.h).
[Here](https://github.com/Infineon/optiga-trust-m/wiki/Code-Footprint#other-functions) is some guidelines on this.

inot1222, 2020-02-06:
Hi @ayushev

Does Trust M I2C and IOs have configurations?
I’ll not use XMC and we’ll use only random function and shield connection.
Is there any other component of Trust M that needs to be initialized?

And OPTIGA_CRYPT_RANDOM and OPTIGA_COMMS_SHIELDED_CONNECTION need to be ported to platform agnostic APIs, right?

ayushev, 2020-02-06:
> Does Trust M I2C and IOs have configurations?
> I’ll not use XMC and we’ll use only random function and shield connection.
> Is there any other component of Trust M that needs to be initialized?

If I understood your question correctly, then the only thing you need to do in term of porting is to follow this [guidance ](https://github.com/Infineon/optiga-trust-m/tree/master/pal#port-crypto-module-for-platfrom-abstraction-layer). You would need both parts, the hardware porting and the crypto porting.

Shielded connection on the host side uses a crypto library to support crypto operations on the host side (encrypt data sent to Trust M, and decrypt data comming from Trust M). We provide mbedTLS along with this repository, so if you don#t have special requirements you can take it as it is (don#t forget to include mbedtls module in your build from [here](https://github.com/Infineon/optiga-trust-m/tree/master/externals/mbedtls)).

[GPIO configuration ](https://github.com/Infineon/optiga-trust-m/blob/master/pal/NEW_PAL_TEMPLATE/pal_gpio.c) is required only in case you need to bring the chip into the hibernation mode or implement a hard reset. In other cases you can use software reset. For this you can adjust this [line](https://github.com/Infineon/optiga-trust-m/blob/v1.30.885/optiga/include/optiga/optiga_lib_config.h#L119).

I2C Configuration is available as well, for instance i2c address, frame size (if the mcu i2c driver supports only small i2c frames), etc.

> And OPTIGA_CRYPT_RANDOM and OPTIGA_COMMS_SHIELDED_CONNECTION need to be ported to platform agnostic APIs, right?

You just need to define them in config, and others comemnt out. The part which this functionality is implemented is platfrom agnostic. As mentioned above the only thing you need to port is the pal (Platfrom Abstraction Layer)

I hope this will help.

ayushev, 2020-02-10:
Hi @inot1222 can we close the issue, or do you need further assistance?

inot1222, 2020-02-10:
Hi @ayushev

Thank you for your advice.
I will close once.

closed 2020-02-10 by inot1222