Infineon/optiga-trust-m #10: Need help for porting the TrustM on STM32

Darsh-Dev, 2020-02-10:
Hi

We are following the porting steps as mentioned 

> https://github.com/Infineon/optiga-trust-m/wiki/Porting-Guide

I2C communication verified using "test_optiga_communication()" example, and its working fine.

Please find the modified attached pal_os_event.c and pal_os_timer.c
[pal_os_event.txt](https://github.com/Infineon/optiga-trust-m/files/4179493/pal_os_event.txt)
[pal_os_timer.txt](https://github.com/Infineon/optiga-trust-m/files/4179499/pal_os_timer.txt)

We are facing issue with below mentioned call, 
> while (OPTIGA_LIB_BUSY == optiga_lib_status)
>         {
>             //Wait until the optiga_util_read_data operation is completed
>         }

Thanks,
Darshak

ayushev, 2020-02-10:
hi @Darsh-Dev 
you need to make sure function `pal_os_event_registered_callback` is called regularly. The framework is async, so this while loop is to make sure in the background your callbacks are called.
How is the timer configured? Could it be that it's not running?

Darsh-Dev, 2020-02-10:
Hi @ayushev 

Thanks for the pointer. I moved further after resolving the timer configuration. 
I have also changed the Response Type, its working
```
From
OPTIGA_UTIL_SET_COMMS_PROTECTION_LEVEL(me, OPTIGA_COMMS_RESPONSE_PROTECTION);
To
OPTIGA_UTIL_SET_COMMS_PROTECTION_LEVEL(me, OPTIGA_COMMS_NO_PROTECTION);

``` 
Now, I am trying with following examples in sequence,
```
test_optiga_communication();
example_optiga_init_m();
example_optiga_util_read_uid();
example_optiga_util_read_data();
util_handler = optiga_util_create(0, optiga_util_callback, NULL);
status = optiga_util_open_application(util_handler, 0);
status = example_authenticate_chip();
```
I have added some debug print, Please see the attached debug logs.  
[Optiga_TrustM_Debug.txt](https://github.com/Infineon/optiga-trust-m/files/4180730/Optiga_TrustM_Debug.txt)

I am not getting response from "read_chip_cert()" func, example is example_authenticate_chip().

Thanks,
Darshak

ayushev, 2020-02-11:
@Darsh-Dev hard to say for sure. Is there any limitation of the i2c frame size the underlying i2c driver can handle. 
Can it handle 277 bytes at once, if not you need to redefine this macro `IFX_I2C_FRAME_SIZE` 
Could you also please enable [this](https://github.com/Infineon/optiga-trust-m/blob/master/optiga/include/optiga/optiga_lib_config.h#L101) logging. This should trigger quite extensive logging, but this can help.

Darsh-Dev, 2020-02-11:
@ayushev We have enabled the debug log as you have suggested. Please find the attached debug log file [Optiga_TrustM_Debug_enabled.txt](https://github.com/Infineon/optiga-trust-m/files/4185906/Optiga_TrustM_Debug_enabled.txt)
```
mbedtls_x509_crt_parse_der ca cert failed(status=ffffd780)
Error: verify cert failed. (status=0x3002)
```
We have temporarily commented "while (OPTIGA_LIB_BUSY == open_app_status)" line 274 & 264 of file optiga-trust-m/examples/utilities/authenticate_chip/example_authenticate_chip.c

ayushev, 2020-02-11:
@Darsh-Dev you need to uncomment this again.
Trust M is an async device. It measn the software stack is done in a way that you send a request to the device ("read certificate") and without waiting the reply the function `optiga_util_read_data()` exists (in case the i2c transmission was successful). This while loop is required to wait till you get the responce from the chip with the certificate, as it is around 500 bytes and it sinply takes some time to read out this data frame by frame.
Please enable it again and provide the log. 
Alos please check internal stm32 i2c documentation (of the software driver) which frame size can it carry.

Darsh-Dev, 2020-02-11:
@ayushev I have uncomment it again, error "mbedtls_x509_crt_parse_der ca cert failed" comes due to memory allocation failed in mbedtls_calloc, just replaced with "pvPortMalloc" heap_5.c is used in our project.

Updated results after above changes,
```
12 18643 [TrstXHndlr] >read_chip_cert()
[optiga util]     : optiga_util_read_data
[optiga cmd]      : optiga_cmd_get_data_object
�,[optiga cmd]      : Sending read data command...
[optiga comms]    : >>>>
                    Length of data - 0x0011
[optiga comms]    : <<<<0E 00 0C 08 20 81 00 00 06 E0 E0 00 00 02 00 76 7C 
�p                    Length of data - 0x0115
[optiga comms]    : <<<<F 39 7E 4E 68 81 61 5E EC F2 54 C2 80 DD 27 A3 64 EB BA C6 CE 3D 03 01 07 03 42 00 04 44 0E  02 
�                    Length of data - 0x00E1
[optiga cmd]      : Processing response for read data command... 55 0E 95 BD 7A E8 B6 8D AD A4 CE 44 AA 4B 40 44 5B  A2 
[optiga cmd]      : Response of read data command is processed...
13 20767 [TrstXHndlr]  pal_crypt_verify_certificate 

``` 

Now I am again stuck at "pal_crypt_verify_certificate" call. I have debug further, I found the pointer stuck at func "ecp_check_pubkey_sw" file ecp.c
```
static int ecp_check_pubkey_sw( const mbedtls_ecp_group *grp, const mbedtls_ecp_point *pt )
{
    int ret;
    mbedtls_mpi YY, RHS;

    /* pt coordinates must be normalized for our checks */
    if( mbedtls_mpi_cmp_int( &pt->X, 0 ) < 0 ||
        mbedtls_mpi_cmp_int( &pt->Y, 0 ) < 0 ||
        mbedtls_mpi_cmp_mpi( &pt->X, &grp->P ) >= 0 ||
        mbedtls_mpi_cmp_mpi( &pt->Y, &grp->P ) >= 0 )
        return( MBEDTLS_ERR_ECP_INVALID_KEY );

    mbedtls_mpi_init( &YY ); mbedtls_mpi_init( &RHS );

    /*
     * YY = Y^2
     * RHS = X (X^2 + A) + B = X^3 + A X + B
     */
    MBEDTLS_MPI_CHK( mbedtls_mpi_mul_mpi( &YY,  &pt->Y,   &pt->Y  ) );  MOD_MUL( YY  );
    MBEDTLS_MPI_CHK( mbedtls_mpi_mul_mpi( &RHS, &pt->X,   &pt->X  ) );  MOD_MUL( RHS );
```
[Noted : I will check stm32 i2c documentation for I2C frame size limitation]

ayushev, 2020-02-11:
> [Noted : I will check stm32 i2c documentation for I2C frame size limitation]

No need, it looks like you could succesfully read out the certificate.

Before going deep into this problem. Could you skip chip authetnication and run all other examples from [here](https://github.com/Infineon/optiga-trust-m/tree/master/examples/optiga) similar to [this](https://github.com/Infineon/getstarted-optiga-trust-m/blob/master/xmc4800_iot_kit/optiga_shell.c) example.

Do you face any problems?

Darsh-Dev, 2020-02-12:
@ayushev I tried with few example, some example have 
```
OPTIGA_UTIL_SET_COMMS_PROTECTION_LEVEL(me, OPTIGA_COMMS_COMMAND_PROTECTION);  OPTIGA_UTIL_SET_COMMS_PROTECTION_LEVEL(me, OPTIGA_COMMS_RESPONSE_PROTECTION);
```
If I changed it to
```
OPTIGA_UTIL_SET_COMMS_PROTECTION_LEVEL(me, OPTIGA_COMMS_NO_PROTECTION);
```
then its working fine. Please find the debug log of [example_optiga_util_read_uid](https://github.com/Infineon/optiga-trust-m/files/4190203/OPTIGA_COMMS_NO_PROTECTION.txt)

I have TrustM module and I wants to validate the "amazon-freertos-optiga-trust-x" demo with **TrustM**, so I am trying to enable TrustM on **STM32L4 Discovery kit**. Next step for me is, validating _example_authenticate_chip_.

Please suggest me the next debug points.

Darsh-Dev, 2020-02-12:
@ayushev Further update, I have run "optiga_shell_selftest()" to run all the examples with my repo source. Please find the logs [optiga_shell_selftest_LOG.txt](https://github.com/Infineon/optiga-trust-m/files/4190571/optiga_shell_selftest_LOG.txt) and stuck at Starting TLS PRF SHA256 (Key Deriviation) Example.

I am suspecting the issue because of the "OPTIGA_UTIL_SET_COMMS_PROTECTION_LEVEL".

ayushev, 2020-02-12:
> @ayushev I tried with few example, some example have
>
> ```
> OPTIGA_UTIL_SET_COMMS_PROTECTION_LEVEL(me, OPTIGA_COMMS_COMMAND_PROTECTION);  OPTIGA_UTIL_SET_COMMS_PROTECTION_LEVEL(me, [...]

You need to pair devices before running the examples which have shielded connection enabled (see Shorlded Connection 101) for more information. Did you run the pairing example?

You also don’t need necessarily to execute the “example_authenticate_chip” if you have shielded connection enabled. 

Last but not least I’d recommend to use the official release of the Trust X for Amazon FreeRTOS (from their official repository), the one you can find on our GitHub is for previous versions of Amazon FreeRTOS (1.4.3) which didn’t support seamless integration of security modules.

Darsh-Dev, 2020-02-17:
@ayushev Yes, I ran paring example "_example_pair_host_and_optiga_using_pre_shared_secret.c_" during init sequence.  Default example contains, **OPTIGA_COMMS_NO_PROTECTION** and I am not facing any issue with it.
```
**  2. Initialize the protection level and protocol version for the instances    **
OPTIGA_UTIL_SET_COMMS_PROTOCOL_VERSION(me_util,OPTIGA_COMMS_PROTOCOL_VERSION_PRE_SHARED_SECRET);
OPTIGA_UTIL_SET_COMMS_PROTECTION_LEVEL(me_util,OPTIGA_COMMS_NO_PROTECTION);
OPTIGA_CRYPT_SET_COMMS_PROTOCOL_VERSION(me_crypt,OPTIGA_COMMS_PROTOCOL_VERSION_PRE_SHARED_SECRET);
OPTIGA_CRYPT_SET_COMMS_PROTECTION_LEVEL(me_crypt,OPTIGA_COMMS_NO_PROTECTION);
```

Not found the any latest release of TrustX + AmazonFreeRTOS, I will try with latest AmazonFreeRTOS and will port the trustM on it.

ayushev, 2020-02-17:
https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_infineon_trust_x.html

and you can also use [this](https://github.com/Infineon/personalize-optiga-trust/tree/master/using-amazon-root-ca) AppNote to provision the device

Do we have something pending here or shall we close the issue?

Darsh-Dev, 2020-02-17:
@ayushev Thanks for your support. 

I have resolved the issue here just now. Earlier I have used the mbedTLS library which is provided with AmazonFreeRTOS SDK (older version). I tried with **_optiga-trust-m/externals/mbedtls/_** , all the examples are working fine.

closed 2020-02-17 by Darsh-Dev

ayushev, 2020-03-18:
@Darsh-Dev hi, you might be interested in this
https://github.com/Infineon/amazon-freertos-optiga-trust
https://github.com/Infineon/amazon-freertos-optiga-trust/tree/feature/trustm/vendors/infineon/boards/xmc4800_plus_optiga_trust_m