Infineon/optiga-trust-m #12: How to get random numbers without going through the OS scheduler

inot1222, 2020-02-28:
Basically, this code seemed to call the handler with the scheduler from the OS event to access the device, and call the callback function after accessing the device data. Since we only use random numbers, we want to create a device that directly obtains random numbers. Is there a feasible method at the Physical Abstract Layer level?

ayushev, 2020-02-28:
The Secure Element understands only Commands; i.e. **GetRandom**. The Commands are prepared in the optiga_cmd.c file and then transfered to the ifx_i2c protocol implementation (presentation layer, transport layer, data link layer, and the physical layer at the end). All layers have their headers which the chip understands and unwrap. So in order to understnd your command the chip requires headers from the ifx_i2c protocol and the command.
However, I'd say there is a way to make a very basic impelemntation with minimum effort.
Checkout the [Datasheet](https://github.com/Infineon/optiga-trust-m/blob/master/documents/OPTIGA_Trust_M_V1_Datasheet_v1.72.pdf), there in Section 11 "OPTIGA™ Trust M Command/Response I2C Sample Logs" you can find an example on complete i2c messages you need to send to the chip in order to get the coprocessor uid, you can do the same for the random number. Though please carefully read the "Pre-requisites" there.

inot1222, 2020-03-05:
Sorry for the delayed response.
With reference to Section 11, the project has begun to be managed.
So closing this issue.

closed 2020-03-05 by inot1222