Infineon/optiga-trust-m #13: Coding for shield connection

inot1222, 2020-04-07:
Hi expert,

Files related to shield connection are in the following path.
\OPTIGA_Trust_M_V1_1.30.885\examples\optiga\usecases\example_pair_host_and_optiga_using_pre_shared_secret.c

Should I implement this file unchanged if I want to enable shield connection?

For example, if object metadata is read by “3. Read Platform Binding Shared secret (0xE140) data object metadata from OPTIGA using optiga_util_read_metadata.”(comment 3) and LcsO is LcsO ≧ LCSO_STATE_OPERATIONAL(comment 4), the random number generation in comment 5 and the random number writing to the platform binding secret object in comment 7 were not executed.

Is this correct behavior?

When this code is executed after startup (cold reset) and the Platform Binding Secret object metadata metadata is "LcsO ≥ LCSO_STATE_OPERATIONAL", the random number used in Sheild Connection is reset and the RAM has been erased by reset.

Does Shiled Connection work properly in that state?
I do not know the normal operation, so please let me know.

ayushev, 2020-04-07:
hi @inot1222 

>>Should I implement this file unchanged if I want to enable shield connection?
Not necessarily, it is just an example on how to implement it. Some information can be found [here](https://github.com/Infineon/optiga-trust-m/wiki/Shielded-Connection-101) as well

>>Is this correct behavior?
Correct, you can also find in the code the following statement
```
 * Note:
 * If the below example is executed once, the LcsO of Platform Binding shared secret is set to Operational.
 * The LcsO can't be reverted to previous states.
 * Please ensure the access conditions and other required settings in the metadata must be accordingly 
 * before setting the LcsO to operational state.
```
The pairing should happen only once, when you pair your MCU with the Security Element. They negotiate a shared secret. The Trust M stores this secret internally, and the MCU stores this secret somehwere in FLASH (it is recommneded to protect this area). Then in the real worl scenario you need to lock the possibility to pair once more. This is done thorugh the change of Lifecycle state of the object where you store this shared secret on the Trust M, if this oject indicates LcsO is Operational, then it is locked.

By default and in the standard example, the last step isn't performed, to let developers update this shared secret as many times as they want.

If you pair you MCU and your Trust M and lost access to the shared secret on your MCU (or removed this data) + you have locked the Shared Secret object than the chip is only partially functional. It depends on your setup what functionas are available and what are not.

inot1222, 2020-04-08:
Hi @ayushev 

After cold reset, RAM were erased, so I thought I had to pair  MCU with the Security Element.
If pairing has already been performed once, is it unnecessary even if it is reset?

>  The Trust M stores this secret internally, and the MCU stores this secret somehwere in FLASH (it is recommneded to protect this area). 

Does the MCU store the shared secret in FLASH instead of RAM?

ayushev, 2020-04-09:
> After cold reset, RAM were erased, so I thought I had to pair MCU with the Security Element.
> If pairing has already been performed once, is it unnecessary even if it is reset?

as explained [here](https://github.com/Infineon/optiga-trust-m/wiki/Shielded-Connection-101#pairing) it depends on `pal_os_datastore.c` implementation. In the real life scenario it is typically should be stored in a protected area of FLASH. As typically you need to do the pairing only once (it depends on the usecase of course, but in many cases you do this once).

The [example](https://github.com/Infineon/optiga-trust-m/blob/master/pal/xmc4800/pal_os_datastore.c) in this repository indeed stores the secret in RAM, but the [pairing example](https://github.com/Infineon/optiga-trust-m/blob/master/examples/optiga/usecases/example_pair_host_and_optiga_using_pre_shared_secret.c) also doesn't modify the Lifecycle state of the Object (LcsO) to Operational by default. This allows to redo the pairing as many times as you want. 

As soon as you put the LcsO in Operational, the object is locked and for all the consequitive shielded connection sessions you have to use the shared secret you have negotiated right before you put the LcsO in Operational state, if its lost, then shielded connection is not possible.
This corresponds to a real-life usecase, when you pair you MCU with the secure element, they work for some time, then a malicous user unsolder the Secure Element from you MCU motherboard and tries to immitiate your MCU with a differernt MCU. As the fake MCU doesn't have required shared secret the shielded connection isn't possible thus many operations aren't possible (this is just a very generic example, which might vary greatly)

closed 2020-04-15 by inot1222