Infineon/optiga-trust-m #15: Status after closing the optiga_util_close_application function

inot1222, 2020-04-15:
Hi expert,

After the application is opened using the optiga_util_open_application function, when the application is closed using the optiga_util_close_application function, the status may return 0x800B through the callback.

0x0B of the lower 1bye indicated by this status is "Command out of sequence (0x0B)" written in Table 6-Error Codes of "OPTIGA ™ Trust M External Interface Response Status Codes" in "OPTIGA_Trust_M_V1_Solution_Reference_Manual_v1.13.pdf, page 87". I think.

Could you tell me the conditions and factors that set this status?

Currently, this status is not set every time, but the status may return 0 by the number of times the optiga_util_close_application function is executed or by inserting another command.

- Below is the Sequence
Case of returning status 0x800B
1. Application open with optiga_util_open_application function
2. Shield Connection;
Get Shared Secret with optiga_crypt_random function.
Shared Secret writing with optiga_util_write_data function.
The interface is not locked by the optiga_util_write_metadata function.
3. Close the application with the optiga_util_close_application function. At this time, status 0x800B is returned.

- Case of returning status 0x800B as above
1. Application open with optiga_util_open_application function
2. Shield Connection;
Get Shared Secret with optiga_crypt_random function.
Shared Secret writing with optiga_util_write_data function.
The interface is not locked by the optiga_util_write_metadata function.
3. Get a 256-byte random number with the optiga_crypt_random function. At that time, status 0x0000 is returned.
4. Get a 256-byte random number with the optiga_crypt_random function. At that time, status 0x0000 is returned.
5. Application CLOSE with optiga_util_close_application function At that time, status 0x0000 is returned.

- Case of returning status 0x0000
1. Application open with optiga_util_open_application function
2. Shield Connection;
Get Shared Secret with optiga_crypt_random function.
Shared Secret writing with optiga_util_write_data function.
The interface is not locked by the optiga_util_write_metadata function.
3. Close the application with the optiga_util_close_application function. At that time, it returns status 0x800B.
4. Close the application with the optiga_util_close_application function. At that time, it returns status 0x800B.
5. Close the application with the optiga_util_close_application function. At that time, status 0x0000 is returned.

There is a few seconds between each step. (Command transfer from Host terminal)
OPTIGA_COMMS_FULL_PROTECTION is set for OPTIGA_UTIL_SET_COMMS_PROTECTION_LEVEL at random number acquisition and application closing.
As a supplement, there is no requirement to open / close the application in real time, so open the application with Power ON and do not close the application until Power OFF.

ayushev, 2020-04-15:
Hi @inot1222 

would it be possible to paste here code snippets for three of cases you mentioned?

inot1222, 2020-04-20:
It is difficult for me to open the source code to you .

I was concerned about the code implementation of the application CLOSE, so when I tried to fix it, the return value of the status also changed.
In the implemented code, the parameter "perform_hibernate" of "optiga_util_close_application" was set to TRUE. When it is changed to FALSE, 0 "OPTIGA_LIB_SUCCESS" is always returned after executing CLOSE.

Also, even if the parameter "perform_hibernate" is set to TRUE, it has been found that 0 "OPTIGA_LIB_SUCCESS" will be returned if the interval from OPEN to CLOSE is taken sufficiently.
I think the parameter "perform_hibernate" is a flag that puts the device into hibernation. It seems that a certain time interval is required after executing OPEN when the device is paused.

ayushev, 2020-04-23:
Sure, I understand, I was refering more to the code snippet which includes mainly optiga source code, which shouldn't be a problem generally.
Regarding you concern. To the best of my knowedge as soon as you get the responce (from the chip), you can peform a next operation on the chip. There is no wait-time.
YOu should, though be aware, that du to asynchronous nature of the chip, you need to wait for the "completion" event from the chip before performing a next operation.

This is done by following expression
```c
optiga_lib_status = OPTIGA_LIB_BUSY;
return_status = optiga_crypt_xxxx()

if (OPTIGA_LIB_SUCCESS != return_status) { return return_status;  }

// Wait until the optiga_crypt_xxxx operation is completed
while (OPTIGA_LIB_BUSY == optiga_lib_status) {}

if (OPTIGA_LIB_SUCCESS != optiga_lib_status) {  return optiga_lib_status; }

return_status = OPTIGA_LIB_SUCCESS;

return return_status;
```

ayushev, 2020-05-18:
Can you please update the issue, or shall I close it?

inot1222, 2020-05-20:
Thank you for your help.
I'm increasing the understanding of Shield Connection and I'm working on it, so please close this issue once.

closed 2020-05-20 by inot1222