Infineon/optiga-trust-m #16: shared secret based handshake

inot1222, 2020-04-15:
Hi expert,

Please allow me to ask about shared secret based handshake.

Please allow me to ask about "6.3.5 Shared secret based version (xxxx xxx1)" in "Infineon_I2C_Protocol_v2.02.pdf".
I have read this description, the handshake between master and slave looks like only once.

Is it correct?

1.Power On

2. Open the application and execute Shield Connection.
There is no handshake, because
OPTIGA_CRYPT_SET_COMMS_PROTECTION_LEVEL == OPTIGA_COMMS_NO_PROTECTION.

3. Get random numbers (first)
Parameters;
protection_level == OPTIGA_COMMS_FULL_PROTECTION
It's a handshake.

Four. Get random numbers (second time)
Parameter: protection_level ==
OPTIGA_COMMS_FULL_PROTECTION
Do not handshake.

Sequence of No.3 and No.4
-Instance creation (optiga_crypt_create)
-Initialize instance protection level with OPTIGA_CRYPT_SET_COMMS_PROTECTION_LEVEL (OPTIGA_COMMS_FULL_PROTECTION)
-Initialize protocol version with OPTIGA_CRYPT_SET_COMMS_PROTOCOL_VERSION
-Get random number (secret) (optiga_crypt_random)

<No.3 log>
optiga_cmd_execute_handler
OPTIGA_CMD_EXEC_PREPARE_COMMAND
optiga_cmd_get_random_handler
Sending get random command ... ________________
[IFX-PRL]: Transceive txlen 6.
[IFX-PRL]: ifx_i2c_prl_event_handler 2.
[IFX-PRL]: PRL_STATE_START protection_level = 0x3 negotiation_state = 0x0
[IFX-PRL]: PRL_STATE_HANDSHAKE
[IFX-PRL]: Do Handshake.
// Handshake occurs
// The negotiation_state of context ifx_i2c_context_t is PRL_NEGOTIATION_NOT_DONE

optiga_crypt_destroy

<No.4 log>
optiga_cmd_execute_handler
OPTIGA_CMD_EXEC_PREPARE_COMMAND
optiga_cmd_get_random_handler
Sending get random command ... ________________
exit_status = 0x0
[IFX-PRL]: Transceive txlen 6.
[IFX-PRL]: ifx_i2c_prl_event_handler 2.
[IFX-PRL]: PRL_STATE_START protection_level = 0x3
negotiation_state = 0x1
[IFX-PRL]: PRL_STATE_TXRX
// No handshake occurs.
// The negotiation_state of context ifx_i2c_context_t is PRL_NEGOTIATION_DONE.

optiga_crypt_destroy

ayushev, 2020-04-23:
The logic seems to be correct, however logs show it differently. For log three the handshake ("negotiation") isn't completed, whereas for the second time it is.

ayushev, 2020-05-18:
Can you please update the issue, or shall I clsoe it?

inot1222, 2020-05-20:
Thank you for your help.
The problem seems to be improved. Please close this issue once.

closed 2020-05-20 by inot1222