Infineon/optiga-trust-m #20: PAL: template: alternative for unused variables

thopiekar, 2020-05-28:
I noticed the following lint line, which shall ignore unused variables

```
//lint --e{715} suppress "The unused p_i2c_context variable is kept for future enhancements"
_STATIC_H void pal_i2c_release(const void * p_i2c_context)
{
```

However, if you compile your application with an GCC compiler using `-Werror` (warnings as errors), which you do in some RTOSs by default, you will get compiler errors again.
So you will end up with `(void) my_unused_variable` anyways. I'm not familiar with your lint'ing system, but I can imagine it could eat `(void) p_i2c_context` and you could avoid the error poping up at two places :)

ayushev, 2020-05-28:
https://github.com/Infineon/optiga-trust-m/commit/f86043eb1ee83c5d35a4b6bca3feb9c4cb9963bf
https://github.com/Infineon/optiga-trust-m/commit/389bbde38a6ca632ee2a230018b79345b45469a0
should solve this

closed 2020-05-28 by ayushev

thopiekar, 2020-06-02:
Great! Thanks :)