Infineon/optiga-trust-m #22: How to verify the certificate?

Darsh-Dev, 2020-07-20:
Hi,

Is there any API available to verify the certificate?

For time being I want to verify the certificate present in "0xE0E0" with test CA provided by Infineon if this verifies successfully then in future I want to add and verify my device certificate sing with my intermediate CA.

Regards,
Arjun

ayushev, 2020-07-20:
Hello Arjun,

[here](https://github.com/Infineon/optiga-trust-m/wiki/Trust-M-Crypt-API) you can find all available API from the chip itself. There is an example on how to do this on the host side. The chip does internal verification implicitly during some security relevant operations such as protected update of data objects

Darsh-Dev, 2020-07-20:
Hi @ayushev 

I can't find any API specific to verify the certificate from [here.](https://github.com/Infineon/optiga-trust-m/wiki/Trust-M-Crypt-API) and Not getting your point

> The chip does internal verification implicitly during some security relevant operations such as protected update of data objects

Can you please explain in detail.

Our use case is we have two devices one server and client we need to verify both device certificate with each other signed by common CA. if certificate verification is successful then we want to enable further communication. if fail we want to disable the communication.

So is there any single API available in trustM code or external third party library mbedtls to verify the certificate it's helpful.

Regards,
Arjun

ayushev, 2020-07-20:
>>I can't find any API specific to verify the certificate from here. and Not getting your point
sorry for this. I meant that API you are looking for doesn't exist. All supported API form the chip are available in that Wiki Section

>>So is there any API available in trustM code or external third party library mbedtls to verify the certificate?
You can have a look [here](https://github.com/Infineon/optiga-trust-m/blob/master/examples/utilities/authenticate_chip/example_authenticate_chip.c). this PAI is used as part of the [mbedtls ApplicationNote](https://github.com/Infineon/mbedtls-optiga-trust-m)
Please keep in mind, that the host decideds on how to deal with the verification status it has received form the security chip

ayushev, 2020-07-30:
Can we close this one, or do you need further suport here?

closed 2020-08-31 by ayushev