Infineon/optiga-trust-m #23: Certificates overwrite in OPTIGA at every startup

spalishchuk, 2020-07-29:
At every startup certificates located in OPTIGA overwritten by certificate defined in code. It's ok, but memory in OPTIGA alow only 100cycles of write (https://github.com/Infineon/optiga-trust-m/blob/master/documents/OPTIGA_Trust_M_V1_Solution_Reference_Manual_v1.13.pdf Page 113).

ayushev, 2020-07-29:
@spalishchuk is this for [the getting started guide](https://github.com/Infineon/getstarted-optiga-trust-m) or for a different code? Why do you think it happens at each startup?

spalishchuk, 2020-07-29:
for getting started guide. It's a normal flow by code.

ayushev, 2020-07-29:
it should be user controlled, only if a user explicetly triggers the specific example call, then it will be updated, or did I miss something? Can you please point ot the code?

spalishchuk, 2020-07-29:
https://github.com/Infineon/amazon-freertos/blob/df70469f234ff74318baefeeabae082c534d731a/demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c#L640

This function call every startup, vDevModeKeyProvisioning()->vAlternateKeyProvisioning()->xProvisionDevice()->xProvisionCertificate().

spalishchuk, 2020-07-29:
Internally it calls PKCS11_PAL_DestroyObject().

By the way, why JITP certificate not write into OPTIGA. It is normal or not?

ayushev, 2020-07-29:
> https://github.com/Infineon/amazon-freertos/blob/df70469f234ff74318baefeeabae082c534d731a/demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c#L640
> 
> This function call every startup, vDevModeKeyProvisioning()->vAlternateKeyProvisioning()->xProvisionDevice()->xProvisionCertificate().

This call (xProvisionCertificate) [shouln't happen](https://github.com/Infineon/amazon-freertos/blob/be454587d8648cfa768068524e68ad6772b96678/demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c#L1035) if the certificate isn't defined in the [configuration file](https://github.com/Infineon/amazon-freertos/blob/be454587d8648cfa768068524e68ad6772b96678/demos/include/aws_clientcredential_keys.h#L40) which shouldn't be the case, as the certificate comes from the chip itself

ayushev, 2020-07-29:
You can also clone the repository, and try from the scratch this guadance https://github.com/Infineon/amazon-freertos/tree/feature/trustm/vendors/infineon/boards/xmc4800_plus_optiga_trust_m#getting-started-with-the-infineon-optiga-trust-m-and-xmc4800-iot-connectivity-kit
Let me know if there are some difficulties with this. The code you have hear did pass the IDT, so it should be fully working.

spalishchuk, 2020-07-29:
Yes, I use this demo step by step with onboard private key generation. But didn't found in ReadMe that I should replace keyJITR_DEVICE_CERTIFICATE_AUTHORITY_PEM and keyCLIENT_CERTIFICATE_PEM to "" after first start. Should I replace them to exclude overwriting?

ayushev, 2020-07-29:
but you don't have it by default after checking the repository out. I can add this line if you thuink this is neccesary.
Is the issue resolved then?

ayushev, 2020-07-29:
Addressed here https://github.com/Infineon/amazon-freertos/commit/8924fea833e21379ca88d30d263088ecd00ae07e

closed 2020-07-29 by ayushev

spalishchuk, 2020-07-29:
Yes, it's resolved. It's would be better to say in ReadMe that user must define to "" after first provisioning. Certificates and keys already copied into OPTIGA after provisioning. Otherwise slots in OPTIGA deads (I have one. Solution is to switch pkcs11config IODs to other slots )