Infineon/optiga-trust-m #24: C_FindObjects return error if object not found

spalishchuk, 2020-07-29:
In case object not found, function returns xResult = CKR_FUNCTION_FAILED. 
But it's normal, if seached object not found, in that case must returns CKR_OK and *pulObjectCount = 0.

ayushev, 2020-07-29:
@spalishchuk 
Thanks for the issue. About which code snippet and repository are you talking here? Is it the https://github.com/Infineon/amazon-freertos ?

spalishchuk, 2020-07-29:
Yes, https://github.com/Infineon/amazon-freertos/tree/feature/trustm/vendors/infineon/boards/xmc4800_plus_optiga_trust_m 
Branch feature/trustm

spalishchuk, 2020-07-29:
https://github.com/Infineon/amazon-freertos/blob/feature/trustm/vendors/infineon/secure_elements/pkcs11/iot_pkcs11_trustm.c
line 2511

ayushev, 2020-07-30:
Thanks! It seems like you are right!
I need to run the pkcs11 test and IDT to verify whether the fix will still work with the bundle. So this will become part of the next version.