Infineon/optiga-trust-m #25: How to resolve error MBEDTLS_ERR_X509_INVALID_FORMAT ?

Darsh-Dev, 2020-07-29:
Hi,
I have used trust-m HSM for AWS freeRTOS to a new platform I referred freertos programming guide and porting guide of trust-m to do that.

Now my issue is, I m facing certificate parsing error in mbedtls_x509_crt_parse() function call while performing TLS_Connect() in iot_tls.c file , Mbedtls version string is "mbed TLS 2.16.0" one from optiga source it self.

Further deep debugging found that from mbedtls_asn1_get_tag() function receveing MBEDTLS_ERR_ASN1_OUT_OF_DATA .

I have attached my mbedtls/config.h here for reference and after debugging I enabled MBEDTLS_X509_ALLOW_EXTENSIONS_NON_V3 and MBEDTLS_X509_ALLOW_UNSUPPORTED_CRITICAL_EXTENSION macro but still no luck.

I have created a certificate from the one-click method from AWS console and then used tools/certificate_configuration/CertificateConfigurator.html tool to get freeRTOS supported aws_clientcredential_keys.h file and replace in my code.

Also if I apply Linux parse command with OpenSSL,

`openssl x509 -in x.y.z-certificate.pem.crt -text -noout`

it's working and showing output as an attached file.

is there anything I m missing here?

Any guidance will be appreciable.
[openssl_output.txt](https://github.com/Infineon/optiga-trust-m/files/4995165/openssl_output.txt)
[config.txt](https://github.com/Infineon/optiga-trust-m/files/4995167/config.txt)

Thanks

ayushev, 2020-07-29:
Hi @Darsh-Dev 
I didn't quite get you problem. Do you have troubles in parsing the certificate retrieved from the trust m? If so can you please past it here, please exactly as you give to the mbedtls, and also with the function you use to parse this certificate.

Darsh-Dev, 2020-07-29:
Hi,
Thank for the quick reply.

`#define keyCLIENT_CERTIFICATE_PEM \
"-----BEGIN CERTIFICATE-----\n"\
"MIIDWjCCAkKgAwIBAgIVAJ3wzBnLSnQvYi31rNVQRAXDUO/zMA0GCSqGSIb3DQEB\n"\
"CwUAME0xSzBJBgNVBAsMQkFtYXpvbiBXZWIgU2VydmljZXMgTz1BbWF6b24uY29t\n"\
"IEluYy4gTD1TZWF0dGxlIFNUPVdhc2hpbmd0b24gQz1VUzAeFw0yMDA3MjgxMTMz\n"\
"MTJaFw00OTEyMzEyMzU5NTlaMB4xHDAaBgNVBAMME0FXUyBJb1QgQ2VydGlmaWNh\n"\
"dGUwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQDHc2tmezGoekLjkQlb\n"\
"+YOBKFyPswYR+GLq/JRVbFX2k4OrHF5js4GTfbHm1oQ733KbcnIugdejtQnRhtnr\n"\
"1HRk3pqedVhRKGRo2DFDYyuX3K1UR6xna1poJF+6WNy6vXGxIQYKi7SNS5LtzkRT\n"\
"1FCziOLBaxfcCRNgR1NBHjlcFsUWyL4evMok6h/wU7HA3/dfKEisyLdh3sMy7Yox\n"\
"Im/ldvyX+9pH7Hj0TrGGTd5f8GtX8npNuSKdkntuag95r+vAaAPp6bQVyPWm8T/G\n"\
"SUN8N7Nvc9DOcJ8ZhvB/Ubq+Fa/eoUnr3SgXInufLHhrfxJW7dyrBTlw/1kdXgYw\n"\
"YiKnAgMBAAGjYDBeMB8GA1UdIwQYMBaAFP4UzdqnzQ4l89+D7UhXC5MKWnOJMB0G\n"\
"A1UdDgQWBBSn95OHFqTn3DrE3anpNq5RoOsT+DAMBgNVHRMBAf8EAjAAMA4GA1Ud\n"\
"DwEB/wQEAwIHgDANBgkqhkiG9w0BAQsFAAOCAQEA2Hvrxy2N0xt3I/w/7JIyoTH4\n"\
"ixUKMaD1QXe+g6LrsQSCVVsaq0L468OpyydVzQLQONXvDDRv3rqIEel1hPAJNG0y\n"\
"dp3g+WC1dPl7E44btM+59gBf1369lFwV6FbJMwCltVBUJ4hFAjt3QTkWRHq6DlFQ\n"\
"wa896aSr5UUiVNAJjf/hLVjERlVG4wDjPN7YifQssRqlNcYDgok3UhVsBfKIGnct\n"\
"WFbisX+0ONMyNnE1Qq6bX5g4sLN7VlwFhADiz1Xp2rUtLECR1NSPutYibWyvJJ8d\n"\
"htYYV1a0FSkg7JKyvOIJ8IYKEPsKE+UYo1Z8DwkmHHcap+h0OMWAnKQgRXn6QQ==\n"\
"-----END CERTIFICATE-----"`

Above is a certificate generated by AWS CertificateConfigurator.html 

it is being used to save in NV Memory i.e trust -m with vDevModeKeyProvisioning() and I don't see any errors here.

After this call, it begins to communication and in doing so it creates Socket (which is okay ) then it tries to connect(socket_connect).

This call intern calls TLS_connect where it performs Handshaking in an early stage but before that, it parses this certificate by getting into RAM from NV memory(trust-m) using prvReadCertificateIntoContext() call and fails in mbedtls_x509_crt_parse() with an above-mentioned error.

Please let me know if more info is needed.

Thanks

closed 2020-07-29 by ayushev