Infineon/optiga-trust-m #28: How to use the Trust-M in U-Boot

yoshi-ohta, 2020-08-26:
Hi,

We want to use Trust-M in U-Boot for Secure Boot.
We want to get a certificate from Trust-M in Secure Boot.
Please tell me how to install Trust-M driver in U-Boot.

Do you need all or some of the files in this repository?
Is there any sample code or helpful code to use in U-Boot?

We use u-boot-imx.
https://source.codeaurora.org/external/imx/uboot-imx/tree/?h=imx_v2018.03_4.14.98_2.0.0_ga

ayushev, 2020-08-26:
First of all, you need to port the Trust M Host [library](https://github.com/Infineon/optiga-trust-m) to your environment. We do have an example on how to run trust M from the user mode (see [here](https://github.com/Infineon/optiga-trust-m/tree/master/pal/linux) and [here](https://github.com/Infineon/cli-optiga-trust-m)). However u-Boot runs from the kernel mode, it means that you need to use this [Porting Guide](https://github.com/Infineon/optiga-trust-m/tree/master/pal) and this [template](https://github.com/Infineon/optiga-trust-m/tree/master/pal/NEW_PAL_TEMPLATE). Major challenges are: Kernel Timers for the ['pal_os_event.c'](https://github.com/Infineon/optiga-trust-m/blob/master/pal/NEW_PAL_TEMPLATE/pal_os_event.c) and i2c accessin ['pal_i2c.c'](https://github.com/Infineon/optiga-trust-m/blob/master/pal/NEW_PAL_TEMPLATE/pal_i2c.c)

yoshi-ohta, 2020-09-14:
Thank you for your answer, and I'm sorry for the late reply.
We are porting for U-Boot, but we are facing the problem of I2C communication failing.

We are using the DM_I2C driver, but the driver probe() processing fails.
The driver used is as follows.
https://source.codeaurora.org/external/imx/uboot-imx/tree/drivers/i2c/i2c-uclass.c?h=imx_v2018.03_4.14.98_2.0.0_ga
https://source.codeaurora.org/external/imx/uboot-imx/tree/drivers/i2c/mxc_i2c.c?h=imx_v2018.03_4.14.98_2.0.0_ga

When I searched for other drivers, I found an Infineon driver as a TPM driver.
https://source.codeaurora.org/external/imx/uboot-imx/tree/drivers/tpm/tpm-uclass.c?h=imx_v2018.03_4.14.98_2.0.0_ga
https://source.codeaurora.org/external/imx/uboot-imx/tree/drivers/tpm/tpm_tis_infineon.c?h=imx_v2018.03_4.14.98_2.0.0_ga

Which driver should I use to I2C communication for Trust-M?
Or which driver should I refer to?

pwiegele, 2020-09-14:
Hi yoshi-ohta,

the probing can fail because OPTIGA Trust M needs a special weakup sequence. A simple probing (like i2c-detect and similar) will not work.
You should use the generic I2C driver and add the software framework for OPTIGA Trust M as a module to the kernel. This will also require you to implement the PAL layer as suggested by @ayushev .

You can use the repository [here](https://github.com/Infineon/i2c-utils-optiga-trust) to check if you can physically access the device via i2c.

yoshi-ohta, 2020-09-16:
Hi, pwiegele

Thank you for your answer.
I'm not familiar with drivers, so please let me know.
What kind of driver is "generic i2 driver"?
I think the generic dirver in U-Boot is an i2c-uclass driver. Is it correct?
In addition, when I tried with trustm-cli on linux, I was able to communicate with trustm.

Another question.
TrustM-Lib uses asynchronous events (pal_os_event.c) for processing, but U-Boot cannot use asynchronous events.
Therefore, it is necessary to change the processing flow of Trust-M Lib.
I try to wait in a loop after processing one event until the next event.
In this case, the processing speed may be slow and the execution of the next event may be delayed (for example, the next event start time is set to 2ms later, but it is executed 5ms later).
Is it possible that I2C communication will fail if the processing is delayed?

aoune-ayoub, 2024-11-08:
Thank you for reaching out to us.

Due to inactivity, this ticket will be closed now. Feel free to contact us if the issue is not resolved yet.

Thanks,
Ayoub

closed 2024-11-08 by aoune-ayoub