Infineon/optiga-trust-m #34: Integration with ESP32

stefanomologni, 2020-12-11:
Good morning,
  I'm working in a project with SLS32AIA cryptochip module and I'm integrating "Open Source Host Code for OPTIGA™ Trust M libraries" into my ESP32 source.
I'm started from the example attached to the library but I have some troubles:

1)  Calling the "optiga_util_create" initialization function cause an ESP32 hard fail (guru meditation error), to avoid it I need to comment row 212 from pal_os_event.c.

2) Solved this first one I call "optiga_util_open_application" function and wait for an initialization, but the related callback is never called. Both I2C lines remains in idle.

I've already checked that the SLS32AIA is working properly, writing to that directly from ESP32 ecosystem.

Maybe could you support me?

Best regards,
Stefano

ayushev, 2020-12-11:
Hello @stefanomologni 
did you try to prepare the code using only a given PAL layer, or using a sample project here https://github.com/Infineon/azure-optiga-trust-m ?

stefanomologni, 2020-12-11:
Dear @ayushev,
  I used only the PAL layer supplied with OPTIGA library (optiga\pal\esp32_freertos folder).

To test it I followed the sample code in optiga-trust-m-master\examples\optiga\example_optiga_init_deinit.c.

stefanomologni, 2020-12-11:
If it is useful I can share with you my source code...

ayushev, 2020-12-11:
Ok, I see, then please bring that line 212 from the pal_os_event.c back (this line is very important), and try to include somewhere in the code **before** you init the optiga the folloiwng function pal_os_event_init(); from the pal_os_event.h
This should make it work

stefanomologni, 2020-12-11:
Ok, with this trick I solved the first problem, but I still have the second one.

This is a snapshot of my application, this stops in "optiga_wait_and_check_status" routine... 

{
optiga_lib_status_t return_status = !OPTIGA_LIB_SUCCESS;
	
OPT_DEBUG_PRINTF("Initialize OPTIGA");
	
pal_os_event_init();

/* Create an instance of optiga_util to open the application on OPTIGA */
optiga_hdl = optiga_util_create(OPTIGA_INSTANCE_ID_0, optiga_util_callback,
									NULL);
	
/* Open the application on OPTIGA */
optiga_lib_status = OPTIGA_LIB_BUSY;
return_status = optiga_util_open_application(optiga_hdl, 0);
	
**optiga_wait_and_check_status(return_status, optiga_lib_status);**

OPT_DEBUG_PRINTF("Initialization complete!");
}

ayushev, 2020-12-14:
COuld you please check this appnote, to check wether your hardware setup is the same as the PAL. For ths you need to follow the first setp from here https://github.com/Infineon/azure-optiga-trust-m#step-1-download-and-install-missing-components and then directly proceed with the  https://github.com/Infineon/azure-optiga-trust-m#creating-a-test-x509-device-certificate

Check wether you see a similar output mentioned here (Public Key Extraction)

If it works, then please have a closer look on your sdkconfig file, a working example can be found here https://github.com/Infineon/azure-optiga-trust-m/blob/master/examples/provision_test_certificate/sdkconfig

stefanomologni, 2020-12-14:
I followed your links.
I downloaded "azure-optiga-trust-m" repository (it is different from optiga-trust-m repository?).
I went to "azure-optiga-trust-m\examples\provision_test_certificate" folder.
I launched "idf.py menuconfig".

In the menù I haven't parameters related to OPTIGA M as you can see in my screenshot:

[image]
[image]
[image]

I'm sorry...

On the other way I'm trying to compare my sdkconfig with your...

ayushev, 2020-12-14:
It shoul dbe in the Top menu. It could be becasue of the IDF version, probably. The one described in the AppNote is 4.1
REgarding your question, the framework optiga-trust-m is included there as a [component](https://github.com/Infineon/azure-optiga-trust-m/tree/master/components/optiga) this is the reason why you need to close with recursive option

stefanomologni, 2020-12-14:
I'm using the same 4.1 version for IDF...

ayushev, 2020-12-14:
Strange, did you have any warnings? Can you list here the log output of your menuconfig?
For instance mine is following:

```console
C:\Users\Yushev\git\azure-optiga-trust-m\examples\provision_test_certificate>idf.py menuconfig
Executing action: menuconfig
Running cmake in directory c:\users\yushev\git\azure-optiga-trust-m\examples\provision_test_certificate\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 c:\users\yushev\git\azure-optiga-trust-m\examples\provision_test_certificate"...
Warn about uninitialized values.
-- Found Git: C:/Users/Yushev/AppData/Local/Programs/Git/cmd/git.exe (found version "2.21.0.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/Yushev/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/Yushev/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/Yushev/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/Yushev/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/Yushev/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
-- Found PythonInterp: C:/Users/Yushev/.espressif/python_env/idf4.1_py3.8_env/Scripts/python.exe (found version "3.8.1")
-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- Adding linker script C:/Users/Yushev/git/azure-optiga-trust-m/examples/provision_test_certificate/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/Yushev/git/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/Yushev/git/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/Users/Yushev/git/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs-time.ld
-- Adding linker script C:/Users/Yushev/git/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/Yushev/git/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/Yushev/git/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/Yushev/git/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/Yushev/git/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl optiga partition_table perfmon port protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/Yushev/git/esp-idf/components/app_trace C:/Users/Yushev/git/esp-idf/components/app_update C:/Users/Yushev/git/esp-idf/components/asio C:/Users/Yushev/git/esp-idf/components/bootloader C:/Users/Yushev/git/esp-idf/components/bootloader_support C:/Users/Yushev/git/esp-idf/components/bt C:/Users/Yushev/git/esp-idf/components/cbor C:/Users/Yushev/git/esp-idf/components/coap C:/Users/Yushev/git/esp-idf/components/console C:/Users/Yushev/git/esp-idf/components/cxx C:/Users/Yushev/git/esp-idf/components/driver C:/Users/Yushev/git/esp-idf/components/efuse C:/Users/Yushev/git/esp-idf/components/esp-tls C:/Users/Yushev/git/esp-idf/components/esp32 C:/Users/Yushev/git/esp-idf/components/esp_adc_cal C:/Users/Yushev/git/esp-idf/components/esp_common C:/Users/Yushev/git/esp-idf/components/esp_eth C:/Users/Yushev/git/esp-idf/components/esp_event C:/Users/Yushev/git/esp-idf/components/esp_gdbstub C:/Users/Yushev/git/esp-idf/components/esp_http_client C:/Users/Yushev/git/esp-idf/components/esp_http_server C:/Users/Yushev/git/esp-idf/components/esp_https_ota C:/Users/Yushev/git/esp-idf/components/esp_https_server C:/Users/Yushev/git/esp-idf/components/esp_local_ctrl C:/Users/Yushev/git/esp-idf/components/esp_netif C:/Users/Yushev/git/esp-idf/components/esp_ringbuf C:/Users/Yushev/git/esp-idf/components/esp_rom C:/Users/Yushev/git/esp-idf/components/esp_serial_slave_link C:/Users/Yushev/git/esp-idf/components/esp_websocket_client C:/Users/Yushev/git/esp-idf/components/esp_wifi C:/Users/Yushev/git/esp-idf/components/espcoredump C:/Users/Yushev/git/esp [... cut ...]
Executing "ninja menuconfig"...
[0/1] cmd.exe /C "cd /D C:\Users\Yushev\git\azure-optiga-t...iga-trust-m/examples/provision_test_certificate/sdkconfig"
Loaded configuration 'C:/Users/Yushev/git/azure-optiga-trust-m/examples/provision_test_certificate/sdkconfig'
No changes to save (for 'C:/Users/Yushev/git/azure-optiga-trust-m/examples/provision_test_certificate/sdkconfig')

C:\Users\Yushev\git\azure-optiga-trust-m\examples\provision_test_certificate>idf.py --version
ESP-IDF v4.1-beta1-106-g9a1c5a903-dirty
```

stefanomologni, 2020-12-14:
I'm trying to re-download the entire packet, but I have some problems with Azure SDK submodule. I'll try tomorrow and do a new test.

stefanomologni, 2020-12-15:
Also today I have this error during clone:

...
Failed to clone 'deps/azure-macro-utils-c' a second time, aborting
Failed to recurse into submodule path 'components/esp-azure/azure-iot-sdk-c/c-utility/deps/umock-c'
Failed to recurse into submodule path 'components/esp-azure/azure-iot-sdk-c/c-utility/testtools/testrunner'

...

ayushev, 2020-12-15:
Strange, just cloned a fresh repo and it worked, could you try to do the following in the root azure-optiga-trust-m repo:
```console
git submodule foreach --recursive 'git submodule sync'
git submodule update --recursive
```

stefanomologni, 2020-12-15:
I tryed...

..
Failed to recurse into submodule path 'components/esp-azure/azure-iot-sdk-c/deps/uhttp'
Failed to recurse into submodule path 'components/esp-azure/azure-iot-sdk-c/provisioning_client/deps/utpm'
Failed to recurse into submodule path 'components/esp-azure/azure-iot-sdk-c/uamqp'
Failed to recurse into submodule path 'components/esp-azure/azure-iot-sdk-c/umqtt'
Failed to recurse into submodule path 'components/esp-azure/azure-iot-sdk-c'
...

ayushev, 2020-12-15:
If you like you can paste here the complete log starting from the beggining. It looks like a network or github issue.
Alternativly lets go back to you menuconfig result. Can you please list here what menuconfig shows you, like the log I listed above.

stefanomologni, 2020-12-15:
Thi is the complete log...

git.exe submodule update --progress --init --recursive -- "components/esp-azure/azure-iot-sdk-c"
Cloning into 'C:/Users/stefano.mologni/Desktop/azure-optiga-trust-m/components/esp-azure/azure-iot-sdk-c/c-utility/deps/umock-c/deps/ctest/deps/azure-macro-utils-c'...
remote: Enumerating objects: 90, done.
remote: Counting objects: 100% (90/90), done.
remote: Compressing objects: 100% (61/61), done.
remote: Total 657 (delta 34), reused 45 (delta 18), pack-reused 567
Receiving objects: 100% (657/657), 512.26 KiB | 1.52 MiB/s, done.
Resolving deltas: 100% (381/381), done.
fatal: cannot write keep file 'C:/Users/stefano.mologni/Desktop/azure-optiga-trust-m/.git/modules/components/esp-azure/azure-iot-sdk-c/modules/azure-c-shared-utility/modules/deps/umock-c/modules/deps/azure-ctest/modules/deps/azure-macro-utils-c/objects/pack/pack-b3b86793901ae57809038fab7d7d15152ebf8a0e.keep': Filename too long
fatal: index-pack failed
fatal: clone of 'https://github.com/Azure/azure-macro-utils-c.git' into submodule path 'C:/Users/stefano.mologni/Desktop/azure-optiga-trust-m/components/esp-azure/azure-iot-sdk-c/c-utility/deps/umock-c/deps/ctest/deps/azure-macro-utils-c' failed
Failed to clone 'deps/azure-macro-utils-c'. Retry scheduled
Cloning into 'C:/Users/stefano.mologni/Desktop/azure-optiga-trust-m/components/esp-azure/azure-iot-sdk-c/c-utility/deps/umock-c/deps/ctest/deps/azure-macro-utils-c'...
remote: Enumerating objects: 90, done.
remote: Counting objects: 100% (90/90), done.
remote: Compressing objects: 100% (61/61), done.
remote: Total 657 (delta 34), reused 45 (delta 18), pack-reused 567
Receiving objects: 100% (657/657), 512.26 KiB | 1.40 MiB/s, done.
Resolving deltas: 100% (381/381), done.
fatal: cannot write keep file 'C:/Users/stefano.mologni/Desktop/azure-optiga-trust-m/.git/modules/components/esp-azure/azure-iot-sdk-c/modules/azure-c-shared-utility/modules/deps/umock-c/modules/deps/azure-ctest/modules/deps/azure-macro-utils-c/objects/pack/pack-b3b86793901ae57809038fab7d7d15152ebf8a0e.keep': Filename too long
fatal: index-pack failed
fatal: clone of 'https://github.com/Azure/azure-macro-utils-c.git' into submodule path 'C:/Users/stefano.mologni/Desktop/azure-optiga-trust-m/components/esp-azure/azure-iot-sdk-c/c-utility/deps/umock-c/deps/ctest/deps/azure-macro-utils-c' failed
Failed to clone 'deps/azure-macro-utils-c' a second time, aborting
Cloning into 'C:/Users/stefano.mologni/Desktop/azure-optiga-trust-m/components/esp-azure/azure-iot-sdk-c/c-utility/deps/umock-c/deps/testrunner/deps/azure-macro-utils-c'...
remote: Enumerating objects: 90, done.
remote: Counting objects: 100% (90/90), done.
remote: Compressing objects: 100% (61/61), done.
remote: Total 657 (delta 34), reused 45 (delta 18), pack-reused 567
Receiving objects: 100% (657/657), 512.26 KiB | 705.00 KiB/s, done.
Resolving deltas: 100% (381/381), done.
fatal: cannot write keep file 'C:/Users/stefano.mologni/Desktop/azure-optiga-trust-m/.git/modules/components/esp-azure/azure-iot-sdk-c/modules/azure-c-shared-utility/modules/deps/umock-c/modules/deps/testrunner/modules/deps/azure-macro-utils-c/objects/pack/pack-b3b86793901ae57809038fab7d7d15152ebf8a0e.keep': Filename too long
fatal: index-pack failed
fatal: clone of 'https://github.com/Azure/azure-macro-utils-c' into submodule path 'C:/Users/stefano.mologni/Desktop/azure-optiga-trust-m/components/esp-azure/azure-iot-sdk-c/c-utility/deps/umock-c/deps/testrunner/deps/azure-macro-utils-c' failed
Failed to clone 'deps/azure-macro-utils-c'. Retry scheduled
Cloning into 'C:/Users/stefano.mologni/Desktop/azure-optiga-trust-m/components/esp-azure/azure-iot-sdk-c/c-utility/deps/umock-c/deps/testrunner/deps/azure-macro-utils-c'...
remote: Enumerating objects: 90, done.
remote: Counting objects: 100% (90/90), done.
remote: Compressing objects: 100% (61/61), done.
remote: Total 657 (delta 34), reused 45 (delta 18), pack-reused 567
Receiving objects: 100% (657/657), 512.26 KiB | 1.45 MiB/s, done.
Resolving deltas: 100% (381/381), done.
fatal: cannot write keep file 'C:/Users/stefano.mologni/Desktop/azure-optiga-trust-m/.git/modules/components/esp-azure/azure-iot-sdk-c/modules/azure-c-shared-utility/modules/deps/umock-c/modules/deps/testrunner/modules/deps/azure-macro-utils-c/objects/pack/pack-b3b86793901ae57809038fab7d7d15152ebf8a0e.keep': Filename too long
fatal: index-pack failed
fatal: clone of 'https://github.com/Azure/azure-macro-utils-c' into submodule path 'C:/Users/stefano.mologni/Desktop/azure-optiga-trust-m/components/esp-azure/azure-iot-sdk-c/c-utility/deps/umock-c/deps/testrunner/deps/azure-macro-utils-c' failed
Failed to clone 'deps/azure-macro-utils-c' a second time, aborting
Failed to recurse into submodule path 'components/esp-azure/azure-iot-sdk-c/c-utility/deps/umock-c/deps/ctest'
Failed to recurse into submodule path 'components/esp-azure/azure-iot-sdk-c/c-utility/deps/umock-c/deps/testrunner'
Cloning into 'C:/Users/stefano.mologni/Desktop/azure-optiga-trust-m/components/esp-azure/azure-iot-sdk-c/c-utility/testtools/testrunner/deps/azure-macro-utils-c'...
remote: Enumerating objects: 90, done.
remote: Counting objects: 100% (90/90), done.
remote: Compressing objects: 100% (61/61), done.
remote: Total 657 (delta 34), reused 45 (delta 18), pack-reused 567
Receiving objects: 100% (657/657), 512.26 KiB | 1.37 MiB/s, done.
Resolving deltas: 100% (381/381), done.
[... cut ...]
Failed to recurse into submodule path 'components/esp-azure/azure-iot-sdk-c/deps/uhttp'
Failed to recurse into submodule path 'components/esp-azure/azure-iot-sdk-c/provisioning_client/deps/utpm'
Failed to recurse into submodule path 'components/esp-azure/azure-iot-sdk-c/uamqp'
Failed to recurse into submodule path 'components/esp-azure/azure-iot-sdk-c/umqtt'
Failed to recurse into submodule path 'components/esp-azure/azure-iot-sdk-c'

git did not exit cleanly (exit code 1) (1278703 ms @ 15/12/2020 16:52:20)

ayushev, 2020-12-15:
A potential issue is that at the very beggining of your log I see Windows related error "Filename too long", you can try either to install Linux for Windows (WSL) if you have Win10. Alternativly you can download the [release package](https://github.com/Infineon/azure-optiga-trust-m/releases/download/202005.00/202005.00_azure-optiga-trust-m.zip) and work from it.

ayushev, 2020-12-16:
Great, it measn that your hardware setup is fine and it works. Now you need to migrate configuration to your target board.
The original issue was that the open_application command hang up, this is most probably due to timers/events don't work properly.
In yur target system/application can you please uncomment [this line](https://github.com/Infineon/optiga-trust-m/blob/d57dd0a8a858c11c7754c6aac83b8e560756db60/pal/esp32_freertos/pal_os_event.c#L118) and see, whether during optiga_util_open_application you see or don't see these messages. Also could you please copy pastre here a code snipper of your initialisation routine for the optiga

stefanomologni, 2020-12-16:
Done. It prints:

I (31) boot: ESP-IDF v4.1-dirty 2nd stage bootloader
I (31) boot: compile time 15:43:20
I (31) boot: chip revision: 3
I (34) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (42) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 3, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0d9dc ( 55772) map
I (135) esp_image: segment 1: paddr=0x0001da04 vaddr=0x3ffbdb60 size=0x02614 (  9748) load
I (140) esp_image: segment 2: paddr=0x00020020 vaddr=0x400d0020 size=0x566c8 (353992) map
I (277) esp_image: segment 3: paddr=0x000766f0 vaddr=0x3ffc0174 size=0x014f0 (  5360) load
I (279) esp_image: segment 4: paddr=0x00077be8 vaddr=0x40080000 size=0x00404 (  1028) load
I (284) esp_image: segment 5: paddr=0x00077ff4 vaddr=0x40080404 size=0x14a08 ( 84488) load
I (340) boot: Loaded app from partition at offset 0x10000
I (340) boot: Disabling RNG early entropy source...
I (341) cpu_start: Pro cpu up.
I (345) cpu_start: Application information:
I (349) cpu_start: Project name:     esp32
I (354) cpu_start: App version:      e393ef4-dirty
I (360) cpu_start: Compile time:     Dec 16 2020 14:16:59
I (366) cpu_start: ELF file SHA256:  784690ce4ca6f469...
I (372) cpu_start: ESP-IDF:          v4.1-dirty
I (377) cpu_start: Starting app cpu, entry point is 0x40081390
I (0) cpu_start: App cpu up.
I (387) heap_init: Initializing. RAM available for dynamic allocation:
I (394) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (400) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (406) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (412) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (419) heap_init: At 3FFC7658 len 000189A8 (98 KiB): DRAM
I (425) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (431) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (437) heap_init: At 40094E0C len 0000B1F4 (44 KiB): IRAM
I (444) cpu_start: Pro cpu start user code
I (462) spi_flash: detected chip: generic
I (463) spi_flash: flash io: dio
W (463) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (473) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (484) OPT: Initialize OPTIGA
I (584) pal_os_event: Init : Create Timer
vTaskCallbackHandler
I (584) pal_os_event: Init : Create Timer successful
I (584) OPT: Executing command
I (584) OPT: Open-Acquire-lock command
I (584) OPT: Completed.

stefanomologni, 2020-12-16:
I also tryed to copy the optiga component from the new repository to my original project but I have this:

Files/mbedcrypto.dir/ecdh.c.obj   -c C:/esp/esp-idf/components/mbedtls/mbedtls/library/ecdh.c
C:/esp/esp-idf/components/mbedtls/mbedtls/library/ecdh.c: In function 'ecdh_make_params_internal':
C:/esp/esp-idf/components/mbedtls/mbedtls/library/ecdh.c:309:17: error: implicit declaration of function 'ecdh_gen_public_restartable'; did you mean 'mbedtls_ecp_mul_restartable'? [-Werror=implicit-function-declaration]
     if( ( ret = ecdh_gen_public_restartable( &ctx->grp, &ctx->d, &ctx->Q,
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~
                 mbedtls_ecp_mul_restartable
C:/esp/esp-idf/components/mbedtls/mbedtls/library/ecdh.c: In function 'ecdh_calc_secret_internal':
C:/esp/esp-idf/components/mbedtls/mbedtls/library/ecdh.c:622:17: error: implicit declaration of function 'ecdh_compute_shared_restartable' [-Werror=implicit-function-declaration]
     if( ( ret = ecdh_compute_shared_restartable( &ctx->grp, &ctx->z, &ctx->Qp,
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors
[20/968] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/rsa.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

stefanomologni, 2020-12-16:
Last founded. The NimBLE stack forced a flag in menuconfig (MBEDTLS_ECP_RESTARTABLE) to use non blocking ECP functions.

Now I have:

../components/optiga/optiga-trust-m/pal/esp32_freertos/pal_os_event.c:94:6: warning: no previous prototype for 'vTimerCallback' [-Wmissing-prototypes]
 void vTimerCallback( TimerHandle_t xTimer )

I suppose something else in menuconfig...

stefanomologni, 2020-12-16:
Solved also this. Now I have:

I (466) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (477) OPT: Initialize OPTIGA
I (577) pal_os_event: Init : Create Timer
I (577) pal_os_event: Init : Create Timer successful
I (647) pal_i2c: Initialize pal_i2c_init
I (647) pal_i2c: init successful

And still here...

ayushev, 2020-12-17:
The essence of the stack is that you should see many of the "vTaskCallbackHandler" messages, if you see only on or nothing at all, it measn the timer in pal_os_event.c doesn't work correctly for whatever reason. Did you change something there?
The flow is following, this function 
https://github.com/Infineon/optiga-trust-m/blob/d57dd0a8a858c11c7754c6aac83b8e560756db60/pal/esp32_freertos/pal_os_event.c#L192 
is called from the communciation stack many times (depending on how many commands did you send and which commands). Then once the timer
https://github.com/Infineon/optiga-trust-m/blob/d57dd0a8a858c11c7754c6aac83b8e560756db60/pal/esp32_freertos/pal_os_event.c#L94 
is triggered the stack should continue from here
https://github.com/Infineon/optiga-trust-m/blob/d57dd0a8a858c11c7754c6aac83b8e560756db60/pal/esp32_freertos/pal_os_event.c#L120

 I would reccomend also to uncomment this line
https://github.com/Infineon/optiga-trust-m/blob/d57dd0a8a858c11c7754c6aac83b8e560756db60/pal/esp32_freertos/pal_os_event.c#L211 
and comment this line

https://github.com/Infineon/optiga-trust-m/blob/d57dd0a8a858c11c7754c6aac83b8e560756db60/pal/esp32_freertos/pal_os_event.c#L212 
 

So in the end you should see a couple (or many of them depending on the i2c communication) of vTaskCallbackHandler messages

stefanomologni, 2020-12-17:
Changing this line cause this error:

I (477) OPT: Initialize OPTIGA
I (577) pal_os_event: Init : Create Timer
vTaskCallbackHandler
I (577) pal_os_event: Init : Create Timer successful
C:/esp/esp-idf/components/freertos/timers.c:795 (prvProcessReceivedCommands)- assert failed!
abort() was called at PC 0x400913bf on core 0
0x400913bf: prvProcessReceivedCommands at C:/esp/esp-idf/components/freertos/timers.c:795 (discriminator 1)

ELF file SHA256: 17190faafdd7e1a0

Backtrace: 0x4008d72d:0x3ffbce90 0x4008dac1:0x3ffbceb0 0x400913bf:0x3ffbced0 0x400913ff:0x3ffbcf10 0x4008ed05:0x3ffbcf40
0x4008d72d: invoke_abort at C:/esp/esp-idf/components/esp32/panic.c:157

stefanomologni, 2020-12-17:
FreeRTOS has 100Hz tick rate.

stefanomologni, 2020-12-17:
With more printf:

W (455) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (465) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (671) pal_os_event: Init : Create Timer
I (671) pal_os_event: Init : Create Timer successful
[optiga util]     : OPTIGA Trust initialization
[optiga util]     : optiga_util_open_application
[optiga cmd]      : optiga_cmd_open_application
I (761) pal_i2c: Initialize pal_i2c_init  
I (761) pal_i2c: init successful
[optiga cmd]      : Sending open app command...
[optiga comms]    : >>>>
                    Length of data - 0x001B

ayushev, 2020-12-17:
Can you try to change tis line for your setup in the pal_os_event.c
Comment back this line

https://github.com/Infineon/optiga-trust-m/blob/eaf4203e88b5e8da8318e0d27c338e96c4064cd4/pal/esp32_freertos/pal_os_event.c#L211

and write instead the following

```c
xTimerChangePeriod(xTimer, pdMS_TO_TICKS(time_us / 1000), portMAX_DELAY);
```

stefanomologni, 2020-12-18:
Changing this line I have an assert:

I (467) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (672) pal_os_event: Init : Create Timer
vTaskCallbackHandler
I (672) pal_os_event: Init : Create Timer successful
[optiga util]     : OPTIGA Trust initialization
C:/esp/esp-idf/components/freertos/timers[optiga util]     : optiga_util_open_application
[optiga cmd]      : optiga_cmd_open_application.c:795 (prvProcessReceivedCommands)- assert failed!
abort() was called at PC 0x400913bf on core 0

ELF file SHA256: d04ec0ec927ebdc3

Backtrace: 0x4008d72d:0x3ffbce90 0x4008dac1:0x3ffbceb0 0x400913bf:0x3ffbced0 0x400913ff:0x3ffbcf10 0x4008ed05:0x3ffbcf40

ayushev, 2020-12-18:
it looks like this statement pdMS_TO_TICKS(time_us / 1000) is evalueated into 0.
Would it be possible to print out the value of time_ms before you call this line? AN post here.
Alternativly (just to see whether this is the reaosn) you can try to bring back this line
https://github.com/Infineon/optiga-trust-m/blob/eaf4203e88b5e8da8318e0d27c338e96c4064cd4/pal/esp32_freertos/pal_os_event.c#L212

and see wether there is an effect. Alos please don't forget to enable the "vTaskCallbackHandler" output as you did before

stefanomologni, 2020-12-18:
>>Would it be possible to print out the value of time_ms before you call this line?

It is 1000.

>> Alternativly (just to see whether this is the reaosn) you can try to bring back this line

With this works!! Finally!!! I have some errors in the next part but sounds good, thanks!!!

vTaskCallbackHandler
I (672) pal_os_event: Init : Create Timer successful
[optiga util]     : OPTIGA Trust initialization
[optiga util]     : optiga_util_open_application
[optiga cmd]      : optiga_cmd_open_application
I (762) pal_i2c: Initialize pal_i2c_init  
I (762) pal_i2c: init successful
[optiga cmd]      : Sending open app command...
[optiga comms]    : >>>>
                    Length of data - 0x001B
[optiga comms]    : >>>>0 16 08 20 F0 00 00 10 D2 76 00 00 04 47 65 6E 41 75 74 68 41 70 70 6C BE 40 
                    Length of data - 0x001B
[optiga comms]    : >>>>0 16 08 20 F0 00 00 10 D2 76 00 00 04 47 65 6E 41 75 74 68 41 70 70 6C BE 40 
                    Length of data - 0x001B
[optiga comms]    : >>>>0 16 08 20 F0 00 00 10 D2 76 00 00 04 47 65 6E 41 75 74 68 41 70 70 6C BE 40 
                    Length of data - 0x001B
[optiga comms]    : >>>>0 16 08 20 F0 00 00 10 D2 76 00 00 04 47 65 6E 41 75 74 68 41 70 70 6C BE 40 
                    Length of data - 0x001B
[optiga comms]    : <<<<0 16 08 20 F0 00 00 10 D2 76 00 00 04 47 65 6E 41 75 74 68 41 70 70 6C BE 40 
                    Length of data - 0x000B
[optiga cmd]      : Processing response for open app command...
[optiga cmd]      : Response of open app command is processed...
[optiga util]     : OPTIGA Trust initialization is successful
I (2102) OPT: ..........
Generating keypair in Oid E0F0
[optiga crypt]    : optiga_crypt_rsa_generate_keypair
[optiga cmd]      : optiga_cmd_gen_keypair
[optiga cmd]      : Sending generate keypair command..
[optiga comms]    : >>>>
                    Length of data - 0x0014
[optiga comms]    : <<<<0 0F 08 20 B8 41 00 09 01 00 02 E0 F0 02 00 01 11 38 31 
                    Length of data - 0x000B
[optiga cmd]      : Processing response for generate keypair command...
[optiga cmd]      : Error in processing generate keypair response...
[optiga comms]    : >>>>
                    Length of data - 0x0011
[optiga comms]    : <<<<0 0C 08 20 01 00 00 06 F1 C2 00 00 00 01 36 F6 
                    Length of data - 0x000C
[optiga crypt]    : Call back status error0 01 01 FE B4 
8001I (2372) DB: Read data from NVS storage...

ayushev, 2020-12-18:
I see, good that you have a progress here! 
regarding the issue, it looks like you are trying to generate a new key pair using a locked object (0xE0F0) this key object is used in a conjunction with a default Infineon  certificate populated in the object ID 0xE0E0 (for genetic samples which can be found on the market) 
You can also unlock this object and reuse it, but I’d suggest to select a different key if, for instance 0xE0F1
You can find more usage examples of the API in examples/optiga

ayushev, 2020-12-21:
Any update, did that advice help?

stefanomologni, 2020-12-22:
I need to make some tests, but I'm out of office forn now...

ayushev, 2021-01-18:
Hi there, any update? Or should I close the issue?

stefanomologni, 2021-01-19:
Hi, thank you.

You could close this issue.

Best regards,
Stefano

________________________________

Stefano
Mologni
Firmware Design and Micro Area

R&D Division - Senior Firmware Developer

+39 0346 750 337
stefano.mologni@comelit.it<mailto:stefano.mologni@comelit.it><file://com-fs-01/aree_comuni/Firma/signature.html#>
www.comelitgroup.com<http:\\www.comelitgroup.com><http://www.comelitgroup.com><http://www.comelitgroup.com/>
        [http://www.comelit.it/firma/comelit.png]  <http://www.comelitgroup.com/>

[https://www.comelitgroup.com/firma/banner.jpg]<https://www.comelitgroup.com/it-it/cctv-advance-la-nuova-gamma-di-videosorveglianza-professionale-di-comelit>

NORME DI ACCESSO E DI COMPORTAMENTO PER ESTERNI<https://www.comelitgroup.com/it-it/norme-accesso-comportamento-esterni>

AVVISO DI CONFIDENZIALITÀ E PRIVACY
Questo messaggio ed i suoi eventuali allegati possono contenere informazioni confidenziali, di proprietà, legalmente protette. È destinato all'utilizzo esclusivo del destinatario sopra indicato; la natura privata e la confidenzialità non sono modificati da eventuali errori di trasmissione, pur avendo il mittente assunto tutte le misure di sicurezza idonee e preventive per evitarli. Il trattamento dei dati ivi contenuti, e quelli che verranno forniti in risposta, è realizzato in conformità alla normativa sulla privacy Reg 679/2016, potrete prendere visione della informativa anche visitando il sito www.comelitgroup.com ovvero sulle app Comelit. Rispondendo alla presente mail dichiarate di aver preso visione della richiamata informativa e di aver prestato il consenso al trattamento dei dati personali. Se il messaggio non è a lei destinato, non deve utilizzarlo, diffonderlo, copiarlo con qualunque mezzo, o intraprendere azioni basandosi su di esso. Se ha ricevuto questo messaggio per errore, la preghiamo di volerlo distruggere (unitamente ad eventuali copie dello stesso e suoi allegati) e di volerci cortesemente informare del fatto scrivendo al mittente.
CONFIDENTIALITY NOTICE AND PRIVACY
This message and its attachments (if any) may contain confidential, proprietary or legally privileged information and it is intended only for the use of the addressee named above. No confidentiality or privilege is waived or lost by any erroneous transmission, despite having taken all the security and preventative measures to avoid it. The processing of personal data contained therein, and those that will be provided in response, is made in compliance with the privacy regulation Reg 679/2016, you can also read the information by visiting www.comelitgroup.com or on Comelit apps. Responding to this email, you declare that you have read the aforementioned information and have given consent to the processing of personal data If you are not the intended recipient of this message, you are hereby notified that you must not use, disseminate, copy it in any form or take any action in reliance on it. If you have received this message in error, please, delete it (and any copies of it and attachment) and kindly inform the sender.

closed 2021-01-27 by ayushev