Infineon/optiga-trust-m #35: Integration with Electric Imp - I2C Issues

rodrigo-lorenzo-leal, 2020-12-11:
To whom it might concern. 

We are currently developing the new generation of one of our products and we have selected the Optiga Trust M as our crypto co-processor. Our initial prototype uses the Optiga Trust X but, since for the final model we're switching to the Trust M, we decided to open the issue here. 

For the most part we want to use the Optiga Trust X/M to: 
- Encrypt/decrypt data as fast as possible using AES128
- Eventually use ECDH to perform a key exchange.

The biggest problem we're facing is that our [application processor](https://developer.electricimp.com/) does not support C as its programming language, but instead uses a [scripting language](https://developer.electricimp.com/squirrel). Because of this, we can't rely on Optiga's own's framework, but rather our own.

We're willing to write the code to support this, but currently we are struggling to have a simple communication going between our application processor and the TrustX in our prototype

For example, just trying to read the 0x82 register fails often, since the TrustX is unresponsive most of the time. We believe that this is probably because of the "Late Acknowledge Algorithm". We are confused with this algorithm since it says that the master should loop "as long as the slave doesn't respond with an ACK". How long can be the "as long as"? If we try to read continuously until the read is successful it can take 20ms sometimes, but, it can also take 1s. 

Is there any special timing requirements or configuration we need to set to the chip in order to have a reliable I2C communication? 

We wrote to Infineon support and they suggested we open and issue here.

ayushev, 2020-12-14:
Hello @rodrigo-lorenzo-leal 

thanks for posting a support request here, I'll try to assist you here.

I'll try to structurise my answer, it might be that you know most of the things described below, but to be complete.

First thing which might help you is to read and try this simple open application, read co-processor UID i2c snapshot ( [Trust M Datasheet](https://github.com/Infineon/optiga-trust-m/blob/trust_m1_m3/documents/OPTIGA_Trust_M_Datasheet_v3.10.pdf) Section 11, this part is also Trust X applicable, please carefully read the intro part).
In general most of the time Trust X/M sleeps, but wakes up from time to time to see whether there is an activity on the line (idle mode), it means typically the host processor needs to send several times a command within a specific time (in the framework this time is 1 ms [by default](https://github.com/Infineon/optiga-trust-m/blob/trust_m1_m3/optiga/include/optiga/ifx_i2c/ifx_i2c_config.h#L60) ), if you got an NACK, you didn't hit the target, otherwise you should get an ACK. For Data, the polling interval is higher (by default [5 ms](https://github.com/Infineon/optiga-trust-m/blob/trust_m1_m3/optiga/include/optiga/ifx_i2c/ifx_i2c_config.h#L64)), as the chip typically takes some time to prepare this data.

Let me know your progress.

rodrigo-lorenzo-leal, 2020-12-14:
@ayushev 

We'll test some more and come back to you. Is there a way we can configure the Trust X/M to avoid being asleep most of the time? The product is not battery powered so we don't have any current consumption requirements.

ayushev, 2020-12-14:
@rodrigo-lorenzo-leal 

once the chip enters a sleep mode you can't affect it, but it is possible to prolong the entrance from the default 20 ms to the max of 255 ([Sleep Activation Delay](https://github.com/Infineon/optiga-trust-m/wiki/Data-and-Key-Store-Overview) Register, more about sleep you can find in the [Solutions Reference Manual](https://github.com/Infineon/optiga-trust-m/blob/trust_m1_m3/documents/OPTIGA_Trust_M_Solution_Reference_Manual_v3.15.pdf) section 4.1.1 )

agostonpeter, 2020-12-17:
Hi @ayushev 

I'm working together with @rodrigo-lorenzo-leal on this issue and the good news is that we were able to implement a more stable communication with the Trust X (including the "Late Acknowledge Algorithm") so we can read/ write to/from the Trust X reliably. We expect this work just as good with the Trust M also.

I was going through the sample application that you have mentioned (Section 11 of the Trust M Datasheet) and I think there might be a step missing in Section 11.1.3 between Step 1 and Step 2, since the ACK from the Trust M in the DATA register should be read before the response is there. Similarly how it is in Section 11.1.2 Step 3. Note that I was doing this simple example still on the Trust X that we have available on our prototype at the moment.

Once I've added the missing step to read 5 byte from the 0x80 register, I was able to get the Coprocessor UID as expected.

ayushev, 2020-12-17:
@agostonpeter Thank you very much for this input! Let me discuss this with the devteam, if this would be confirmed then we need to change the documentation. I jope this didn't consumer a lot of time ob your side to figure this out.

ayushev, 2020-12-17:
If you need further technical support, don't hesitate to contact here. We will be happy to assist you.

ayushev, 2020-12-17:
Shall we close this issue?

agostonpeter, 2020-12-18:
We can close this issue as the I2C issues we were seeing are solved. We will let you know if we have some more questions as we start working with the Trust M in more details.

Thanks again for the help @ayushev! 💯

closed 2020-12-21 by ayushev