Infineon/optiga-trust-m #39: Trust M no answer

INfinnnnn, 2021-01-02:
Good evening,
I am not sure if it is the right place to ask this but:
I try to communicate with the chip on Shield2Go M using Atmega16(it is the master). The program is written in c.
I just try to address it using the standart address 0x30 (I also tried every single address from 0 to (255&0b11111110) but all I get is always NACK.
[image]

As for the Dev. Kit I broke it in half and connected it like this: (VCC to VCC, GND to GND, RST to VCC(I wont reset it), SDA to SDA with 10k pull-up and SCL to SCL using 10k pull-up; the SCL frequency is 100kHz)

I witnessed a similiar problem here: https://arduino.stackexchange.com/questions/51785/problem-using-optiga-trust-x-i2c
However, I still cannot find out what the problem is.

Has it sth to do with the "multiple-master-mode"?
Thank you for your time.

ayushev, 2021-01-03:
Hi @INfinnnnn 

it might not be directly visible in the [documentation](https://github.com/Infineon/optiga-trust-m/tree/trust_m1_m3/documents) (Datasheet or Solution Reference Manual), but here what you can find in the Datasheet Annex B (Section 11):
"The OPTIGA™ Trust M will not acknowledge the slave address sent by a host if it is either busy or in idle
state. Hence the host must retry or repeat the transaction until it is successful or timed out for 100
milliseconds (extreme case)."
The chip always goes into an idle or sleep state if no command is issued in a defined time period, unless a Sleep Activation Delay isn't changed.

Porting the library on Atmega16 might be a tough challenge, but if you don't need complex options supported by the communicaiton stack and chip it's managable, if you need help please place your question here.

INfinnnnn, 2021-01-03:
HI, thank you for your reply.
I have put the transaction in a loop as shown in "Infineon_I2C_Protocol_v2.02" on page 9 but the chip never answers.
Could it be that it is dead? Or are the SDA and SCL voltages maybe too low? I have a 5V source and measure around 3.7V as high SCL and SDA. As for the program I have written a small prototype but it gets stuck at addressing the chip:
while (ack != 1) ack = I2C_Start(ADDRESS | 0x00);    /*if ack = 1, ack was received; if ack = 0, nack was received*/

I also have tried to reset or cold reset it at start (RST(ad VCC) LOW for 2ms and then wait 15ms).

Regards

ayushev, 2021-01-04:
Is the communication log you attached above the one from your setup? Then I would suggest to shift left the address by 1 bit and "or" it with the write bit, it should be 0b01100000 not 0b00110000
Can you try this setup?

INfinnnnn, 2021-01-04:
Wow you are truly my hero. I apologize for this stupid mistake. Have a good day and once more, thank you.

closed 2021-01-04 by ayushev