Infineon/optiga-trust-m #40: Protected update failure when COUID is specified

dbalashenka, 2021-01-18:
Hello,

I'm verifying protected update feature for our use cases. I was able to use it successfully, however when I generate data set for a specific COUID, protected update is rejected with the Error 0x800F: OPTIGA device Invalid Manifest. 

I do not know how to approach in troubleshooting this. Can you confirm that 'couid' option is functional, because I did not see it in examples or official documentation, only in help of protected_update_data_set.exe. Am I reading and providing correctly couid content?

Here is an example of what I'm doing.
1. Read COID from the chip
trustm_data -r 0xe0c2
========================================================
UID                         [0xE0C2] [Size 0027] : 
        CD 16 33 4D 01 00 1C 00 05 00 00 0A 09 1B 5C 00 
        15 00 70 00 9A 80 10 10 70 24 40 
========================================================

2. Generate data set for protected update, here is a partial output of the tool protected_update_data_set.exe
Tool Version : 3.00.2490                                                                               
Info : Default values are set                                                                          
Info : User provided inputs                                                                            
        Payload version      : 1                                                                       
        Trust anchor oid     : E0EF                                                                    
        Target oid           : E140                                                                    
        Co-Processor OID     : CD16334D01001C000500000A091B5C00150070009A801010702440                  
        Signature Algorithm  : ES_256                                                                  
        Private key          : my.key
        Type of Payload      : metadata                       
        Shared secret        : secret.txt
        Shared secret OID    : F1DA    
        Label                : metadata                                              
        Encryption algorithm : AES-CCM-16-64-128
        Length of seed       : 64                                                                      
        Content Reset        : 0                                                                                 
        Meta Data            : new-metadata.txt                                         
                                                                                                       
Info : Setting value for data formatter                                                                                                                                                                                                       
        Payload version      : 1                                                                                                                                                                                                              
        Trust anchor oid     : E0EF
        Target oid           : E140
        Digest algorithm     : 29
        Co-Processor OID     : CD16334D01001C000500000A091B5C00150070009A801010702440
        Signature Algorithm  : FFFFFFF9
        Type of Payload      : FFFFFFFE
        Length of seed       : 40
        Label                : metadata
        Shared secret OID    : F1DA
....
        Key Derivation Function : FFFEFF48
        Content Reset        : 0

Best regards,
Dzmitry

ayushev, 2021-01-18:
Hello @dbalashenka 

am I right, that you would like to do the protected update so that it is valid only for a specific chip with a dedicated Coprocessor ID?

dbalashenka, 2021-01-19:
Yes, that the idea. It is essential feature. It allows, for example, to produce data protected update, which would only be valid for a specific device without being afraid that it can be reused on multiple devices.

ayushev, 2021-01-19:
Ok, understood, yes this feature is available, unfortunately the limitation is a bit hidden. A cddl representation is avilable but it's burried under many of other presentations https://github.com/Infineon/optiga-trust-m/blob/8c5610b497168e0e9da1d659dc9e2f64ecb243d6/examples/tools/protected_update_data_set/cddl/OPTIGA_Trust_M_V3_manifest.cddl#L175
YOu should give 25 bytes instead of 27 (two last bytes 2440 should be removed)

dbalashenka, 2021-01-20:
Thank you, I tried removing last two bytes as you suggested. But I still get the same error. What else can I check?

dbalashenka, 2021-01-22:
Hello, I still have the same issue. Can you send an example of functional manifest data - I would like to compare with what I have. 

Thanks,
Dzmitry

ayushev, 2021-01-22:
the couid option should be functional, it is enabled for the Trust M V3 (the chip you have). To get functional manifest only with this specific feature enabled I need some time. 
But what I noticed already now, if I generate the metadata manifest from the samples directory, my output doesn't mathc yours

```bash
C:\Users\Yushev\git\tmp\optiga-trust-m\examples\tools\protected_update_data_set\samples>..\bin\protected_update_data_set.exe payload_version=3 trust_anchor_oid=E0E3 couid=CD16334D01001C000500000A091B5C00150070009A80101070 target_oid=E0E1 sign_algo=ES_256 priv_key=..\samples\integrity\sample_ec_256_priv.pem payload_type=metadata metadata=..\samples\payload\metadata\metadata.txt content_reset=0 secret=..\samples\confidentiality\secret.txt label="test" enc_algo="AES-CCM-16-64-128" secret_oid=F1D1
Tool Version : 3.00.2490
Info : Default values are set
Info : User provided inputs
        Payload version      : 3
        Trust anchor oid     : E0E3
        Target oid           : E0E1
        Co-Processor OID     : CD16334D01001C000500000A091B5C00150070009A80101070
        Signature Algorithm  : ES_256
        Private key          : ..\samples\integrity\sample_ec_256_priv.pem
        Type of Payload      : metadata
        Shared secret        : ..\samples\confidentiality\secret.txt
        Shared secret OID    : F1D1
        Label                : test
        Encryption algorithm : AES-CCM-16-64-128
        Content Reset        : 0
        Meta Data            : ..\samples\payload\metadata\metadata.txt

Info : Setting value for data formatter
        Payload version      : 3
        Trust anchor oid     : E0E3
        Target oid           : E0E1
        Digest algorithm     : 29
        Co-Processor OID     : CD16334D01001C000500000A091B5C00150070009A80101070
        Signature Algorithm  : FFFFFFF9
        Type of Payload      : FFFFFFFE
        Length of seed       : 40
        Label                : test
        Shared secret OID    : F1D1
0x49, 0xC9, 0xF4, 0x92, 0xA9, 0x92, 0xF6, 0xD4, 0xC5, 0x4F, 0x5B, 0x12, 0xC5, 0x7E, 0xDB, 0x27,
        0xCE, 0xD2, 0x24, 0x04, 0x8F, 0x25, 0x48, 0x2A, 0xA1, 0x49, 0xC9, 0xF4, 0x92, 0xA9, 0x92, 0xF6,
        0x49, 0xC9, 0xF4, 0x92, 0xA9, 0x92, 0xF6, 0xD4, 0xC5, 0x4F, 0x5B, 0x12, 0xC5, 0x7E, 0xDB, 0x27,
        0xCE, 0xD2, 0x24, 0x04, 0x8F, 0x25, 0x48, 0x2A, 0xA1, 0x49, 0xC9, 0xF4, 0x92, 0xA9, 0x92, 0xF6,
        Encryption algorithm : A
        Key Derivation Function : FFFEFF48
        Content Reset        : 0
Manifest Data , size : [284]
        uint8_t manifest_data[] =
        {
        0x84, 0x43, 0xA1, 0x01, 0x26, 0xA1, 0x04, 0x42, 0xE0, 0xE3, 0x58, 0xCE, 0x86, 0x01, 0xF6, 0xF6,
        0x84, 0x21, 0x0D, 0x03, 0x82, 0x00, 0x00, 0x82, 0x82, 0x20, 0x58, 0x25, 0x82, 0x18, 0x29, 0x58,
        0x20, 0xA8, 0x47, 0x4D, 0x8D, 0x00, 0x35, 0xDF, 0xBE, 0x97, 0xA3, 0xF7, 0x04, 0x75, 0x15, 0x16,
        0xF9, 0xAA, 0x20, 0x5F, 0xCE, 0x56, 0x99, 0x42, 0x7D, 0x51, 0x44, 0x03, 0xDB, 0x4F, 0x82, 0xC4,
        0xFB, 0x82, 0x01, 0x83, 0x43, 0xA1, 0x01, 0x0A, 0x81, 0x82, 0x58, 0x54, 0xA3, 0x04, 0x42, 0xF1,
        0xD1, 0x01, 0x3A, 0x00, 0x01, 0x00, 0xB7, 0x05, 0x82, 0x44, 0x74, 0x65, 0x73, 0x74, 0x58, 0x40,
        0x9F, 0x4A, 0xF5, 0x22, 0x4D, 0x6C, 0xB0, 0x4B, 0x28, 0x62, 0xAD, 0x68, 0x69, 0xD6, 0xBB, 0xAF,
        0x2D, 0x18, 0x62, 0xEE, 0x86, 0xD4, 0xA7, 0x3F, 0x60, 0x9E, 0xCD, 0xDB, 0x8A, 0xAE, 0x5D, 0x6B,
        0xE1, 0x3C, 0xEA, 0x9E, 0xBD, 0xF5, 0xFB, 0x9A, 0xE4, 0x2C, 0x6B, 0x0D, 0x34, 0xA5, 0x71, 0xE4,
        0xC2, 0x7C, 0x42, 0x0E, 0xF6, 0xB2, 0x20, 0x6D, 0xC4, 0xCA, 0x14, 0x87, 0x45, 0x3B, 0xD5, 0xA6,
        0xF6, 0xF6, 0x82, 0x58, 0x32, 0x43, 0x44, 0x31, 0x36, 0x33, 0x33, 0x34, 0x44, 0x30, 0x31, 0x30,
        0x30, 0x31, 0x43, 0x30, 0x30, 0x30, 0x35, 0x30, 0x30, 0x30, 0x30, 0x30, 0x41, 0x30, 0x39, 0x31,
        0x42, 0x35, 0x43, 0x30, 0x30, 0x31, 0x35, 0x30, 0x30, 0x37, 0x30, 0x30, 0x30, 0x39, 0x41, 0x38,
        0x30, 0x31, 0x30, 0x31, 0x30, 0x37, 0x30, 0x42, 0xE0, 0xE1, 0x58, 0x40, 0x8A, 0xF2, 0x5A, 0x5F,
        0x95, 0x33, 0x1D, 0xDE, 0xA3, 0x7C, 0x2B, 0x80, 0xAA, 0x8D, 0x8D, 0x8C, 0xF3, 0x6F, 0x40, 0x1F,
        0x25, 0x62, 0xDE, 0x40, 0x21, 0x5F, 0x57, 0x02, 0xAE, 0xB3, 0x7E, 0x4F, 0xF0, 0x60, 0xF9, 0xC4,
        0xE4, 0x39, 0x10, 0xF3, 0xF7, 0x95, 0x1C, 0xE8, 0x6F, 0x29, 0x13, 0x73, 0x46, 0x1D, 0xFE, 0xB1,
        0xF2, 0xCB, 0xE3, 0xAB, 0xA9, 0x20, 0x34, 0x69, 0xAE, 0xA5, 0xC3, 0xA0,
        };

Fragment number:[01], size:[021]
        uint8_t fragment_01[] =
        {
        0x0E, 0x1C, 0x84, 0x4D, 0xFB, 0x3D, 0xD1, 0x1B, 0xB7, 0x24, 0x90, 0x94, 0x0B, 0xC5, 0x9A, 0x8F,
        0x68, 0xA4, 0xD0, 0xAF, 0xA1,
        };
```

In particular the payload version, the error you have might be triggerd by the version mismatch. Unicast update was enabled in the latest chip version (M3)
CAn you try to generate the data with the same sample script, adding couid=CD16334D01001C000500000A091B5C00150070009A80101070 as a parameter and check for the result

dbalashenka, 2021-01-22:
I tried to generate manifest with couid that you suggested. I got the same result. Then I applied directly your manifest for my case (I know that it was issued for different OID, but it should at least accept manifest) - I got the same "Invalid manifest" error.

Payload version is not related to the version of the chip, it is the next version of the target OID - as specified by a user. So, it is something else. 
My trustm device is actually engineering sample. Is it possible that it is not final version?

ayushev, 2021-01-26:
Apparantly you were right, the parsing of couid wasn't correctly done. The tool used octet string as an input for manifest construction, instead of a real byte array.
https://github.com/Infineon/optiga-trust-m/commit/86871be8f82d7d754fd4dc0c24b7f1b937f13591 should fix it, please revert if it doesn't for your case. Production team will need some time to test all corner cases for this, I added this commit here, so you an make it work for your setup right now, so theoretically it can fail, but my lcoal tests were all fine.

dbalashenka, 2021-01-27:
I was not able to use new executable, it seems compiled in debug mode and I'm missing VCRUNTIME140D.dll, ucrtbased.dll. Microsoft Visual C++ redistributable package comes only with release version of these libraries

ayushev, 2021-01-27:
You are right, my bad, thank you for spotting this.
I recompiled in release mode https://github.com/Infineon/optiga-trust-m/commit/32579623d1afe37710990b14e503917ce79ac32e 
Can you please try this one?

dbalashenka, 2021-01-27:
This one worked. Thank you! You can close the issue.

closed 2021-01-27 by ayushev