Infineon/optiga-trust-m #41: GenKeyPair: which Bytes to send

INfinnnnn, 2021-01-18:
Good evening,
If it is not a too complicated question to ask, I have the following problem:
after initializing the application context I need to generate a key pair for RSA-1024BIT for Encryption & Decryption.

I send the following Bytes (as described in table 18 in "Solution Reference Manual"):

 
[image]

Description of APDU:
**0x38**: GenKeyPair Command
**0x41**: Algorithm Identifier: RSA 1024 bit
**0x0009**: Length of InData
**0x01, 0x0002**
**0xE0FC**: (Ref.: Table 32) (Solution Reference Manual): Device Private RSA Key
**0x02, 0x0001**
**0x02**: Key Usage: Encryption / Decryption (Ref.: table: Table 22)

-But when I read the I2C register after receiving an ACK, I get following Bytes: 0x08 0x80 0x00 0x00 which basically means that the device is not busy (not executing a command) and it is not ready to give a response.

What do I do wrong?

Regards

ayushev, 2021-01-21:
it looks like you have an equipment ot record frames you sent, would it be possible to record (jsut bytes with timestamps) the overall transaction, from the beggining til the end (when you receive 0x08 0x80 0x00 0x00)

ayushev, 2021-01-21:
It could be that you wait before sending the command and getting the responce from the chip not long enough (if I recall corectly you use your custom communicaiton stack with teh chip). It takes for the chip in average about 5 ms to copy the content of the command you send to internal uffer and switch the internal state to inform the host that the operation is in progress. Can you try to wait for it (after you received the ack and the next attempt to read the content)?

INfinnnnn, 2021-01-23:
Hello,
looks like I fixed it. The frame umber was missing in FCTR. I have another question though:

Should I close an application after every command?
Or which sequence is correct:

a) OpenApplication->GenKeyPair->CloseAppl->OpenAppl->EncryptAsym->CloseAppl
b) OpenAppl->GenKeyPair->EncryptAsym->CloseAppl

Regards

closed 2021-01-23 by ayushev