Infineon/optiga-trust-m #43: Export ECC private key

stefanomologni, 2021-01-27:
Hi,
  I'm trying to use  **optiga_crypt_ecc_generate_keypair** function in order to generate keys. For public key no problem, but I need to export also private key to test it, but I can't.

I set **export_private_key** as TRUE, but in **public_key** buffer the function export only public key. Where I'm wrong?

Thanks,
Stefano

ayushev, 2021-01-27:
Hi @stefanomologni,

how you api call looks like (please including variables you pass to the function)?

stefanomologni, 2021-01-27:
Hi @ayushev, my call is this:

void optiga_generate_ecc_keypair(uint8_t * pub_key, uint8_t * pub_key_size)
{
    optiga_lib_status_t return_status = 0;
    optiga_key_id_t optiga_key_id;
    optiga_crypt_t * me = NULL;

    do
    {
        /* Create OPTIGA crypt Instance */
        me = optiga_crypt_create(0, optiga_util_callback, NULL);
        if (NULL == me)
        {
            break;
        }

        /* Generate ECC Key pair using ECC NIST P 256 Curve and expoort public */
        optiga_lib_status = OPTIGA_LIB_BUSY;
        optiga_key_id = OPTIGA_KEY_ID_E0F1;

        return_status = optiga_crypt_ecc_generate_keypair(me,
                                                          OPTIGA_ECC_CURVE_NIST_P_256,
                                                          (uint8_t)OPTIGA_KEY_USAGE_SIGN,
                                                          TRUE,
                                                          &optiga_key_id,
														  pub_key,
                                                          pub_key_size);

ayushev, 2021-01-27:
You need to provide the buffer to store your [private key](https://github.com/Infineon/optiga-trust-m/wiki/Trust-M-Crypt-API#optiga_crypt_ecc_generate_keypair).
Can you try this call? 

```
void optiga_generate_ecc_keypair(uint8_t * pub_key, uint8_t * pub_key_size)
{
    #define  OPTIGA_ECC_CURVE_NIST_P_256_SIZE 32
    #define  PRIVATE_KEY_TAG_SIZE 2
    /*
     * // DER OCTET String Tag (Private Key)
     * 04
     *  // Length of Tag
     *  20
     *    // Private Key
     *    20 DC 58 98 CF 51 31 44 22 EA 01 D4 0B 23 B2 45
     *    7C 42 DF 3C FB 0D 33 10 B8 49 B7 AA 0A 85 DE E7
     */
    optiga_lib_status_t return_status = 0;
    uint8_t private_key[OPTIGA_ECC_CURVE_NIST_P_256_SIZE + PRIVATE_KEY_TAG_SIZE];
    optiga_crypt_t * me = NULL;

    do
        /* Create OPTIGA crypt Instance */
        me = optiga_crypt_create(0, optiga_util_callback, NULL);
        if (NULL == me)
        {
            break;
        }

        /* Generate ECC Key pair using ECC NIST P 256 Curve and expoort public */
        optiga_lib_status = OPTIGA_LIB_BUSY;
        optiga_key_id = OPTIGA_KEY_ID_E0F1;

        return_status = optiga_crypt_ecc_generate_keypair(me,
                                                          OPTIGA_ECC_CURVE_NIST_P_256,
                                                          (uint8_t)OPTIGA_KEY_USAGE_SIGN,
                                                          TRUE,
                                                          private_key,
                                                          pub_key,
                                                          pub_key_size);
// ...
```
you might probably also give your private key as function parameter.
Please keep in mind, that if you export the key it won't be stored on the chip, as in this case it's not secure anymore.

stefanomologni, 2021-01-27:
Done. With this the value into private_key doesn't change. If I supply with all zeroes returns witho all zeroes, if I supply with all FF returns all FF.

>> Please keep in mind, that if you export the key it won't be stored on the chip, as in this case it's not secure anymore.

Yes, I know. It is only to test the chain. I need to generate keys and share them with openssl to double check.

Thanks,
Stefano

ayushev, 2021-01-27:
I've updated my answer, do you have also that three calls inf your code? What is the optiga_lib_status value at the end?

stefanomologni, 2021-01-27:
Ok, now works. I've checked immediately after call intead of wait for callback... :-D

Now I have two keys (public and private) but doesn't match. I suppose I'm using wrong part of public key. Which is the format of public key after the generation?

For example private key is: TAG[2] + KEY[32]

closed 2021-01-27 by ayushev

stefanomologni, 2021-01-27:
Ok, thanks!