Infineon/optiga-trust-m #45: Linux Port - Linux 4.9.11 GCC 7.3.0

Kitty-Hawk1, 2021-02-15:
Hello,

We’re running the Infineon Trust M attached to an i.MX6 running Linux 4.9.11 built with GCC version 7.3.0

We’re using yocto to build all the software and the recipe is :

CFLAGS_prepend = "-I${STAGING_INCDIR} -fPIC"

do_compile () { 
     oe_runmake -C ${WORKDIR}/git workaround_patch

     oe_runmake -C ${WORKDIR}/git 'CC=${CC}'
}

Now we’re seeing that to make the code work correctly we need to change:

optiga_lib_status needs to be a volatile 

which implies our compiler options don’t match what you’ve used in testing on the Raspberry PI.

Could you elaborate on what compiler options are used for the RPI3 or if the change is better [also any other changes ?] File details below of the changes.

--- git/trustm_helper/include/trustm_helper.h-old 2021-02-09 09:40:30.861357292 +0000
+++ git/trustm_helper/include/trustm_helper.h    2021-02-09 08:46:49.029489413 +0000
@@ -113,7 +113,7 @@
// *********** Extern
extern optiga_util_t * me_util;
extern optiga_crypt_t * me_crypt;
-extern optiga_lib_status_t optiga_lib_status;
+extern volatile optiga_lib_status_t optiga_lib_status;
extern uint16_t trustm_open_flag;
extern uint8_t trustm_hibernate_flag;

only in patch2:
unchanged:
--- git/trustm_helper/trustm_helper.c-old   2021-02-09 09:39:35.737359553 +0000
+++ git/trustm_helper/trustm_helper.c 2021-02-09 08:47:15.297488336 +0000
@@ -48,7 +48,7 @@
*************************************************************************/
optiga_util_t * me_util;
optiga_crypt_t * me_crypt;
-optiga_lib_status_t optiga_lib_status;
+volatile optiga_lib_status_t optiga_lib_status;
uint16_t trustm_open_flag = 0;
uint8_t trustm_hibernate_flag = 0;

ayushev, 2021-02-15:
Hello @Kitty-Hawk1 

I suppose you are well aware of this repository 
Make and build-in gcc (version 8.3) has the following output:
<details><summary>Output</summary>

```console
pi@raspberrypi:~/git/cli-optiga-trust-m $ gcc --version
gcc (Raspbian 8.3.0-6+rpi1) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
pi@raspberrypi:~/git/cli-optiga-trust-m $ make
+++++++ Generating lib object: trustm_lib/pal/linux/pal.c
gcc -c -I trustm_lib/optiga/include -I trustm_lib/optiga/include/optiga -I trustm_lib/optiga/include/optiga/ifx_i2c -I trustm_lib/optiga/include/optiga/comms -I trustm_lib/optiga/include/optiga/common -I trustm_lib/optiga/include/optiga/cmd -I trustm_lib/optiga/include/optiga/pal -I trustm_lib/pal/linux -I trustm_helper/include -I trustm_engine -Wall -DENGINE_DYNAMIC_SUPPORT trustm_lib/pal/linux/pal.c -o trustm_lib/pal/linux/pal.o
+++++++ Generating lib object: trustm_lib/pal/linux/pal_gpio.c
gcc -c -I trustm_lib/optiga/include -I trustm_lib/optiga/include/optiga -I trustm_lib/optiga/include/optiga/ifx_i2c -I trustm_lib/optiga/include/optiga/comms -I trustm_lib/optiga/include/optiga/common -I trustm_lib/optiga/include/optiga/cmd -I trustm_lib/optiga/include/optiga/pal -I trustm_lib/pal/linux -I trustm_helper/include -I trustm_engine -Wall -DENGINE_DYNAMIC_SUPPORT trustm_lib/pal/linux/pal_gpio.c -o trustm_lib/pal/linux/pal_gpio.o
+++++++ Generating lib object: trustm_lib/pal/linux/pal_i2c.c
gcc -c -I trustm_lib/optiga/include -I trustm_lib/optiga/include/optiga -I trustm_lib/optiga/include/optiga/ifx_i2c -I trustm_lib/optiga/include/optiga/comms -I trustm_lib/optiga/include/optiga/common -I trustm_lib/optiga/include/optiga/cmd -I trustm_lib/optiga/include/optiga/pal -I trustm_lib/pal/linux -I trustm_helper/include -I trustm_engine -Wall -DENGINE_DYNAMIC_SUPPORT trustm_lib/pal/linux/pal_i2c.c -o trustm_lib/pal/linux/pal_i2c.o
+++++++ Generating lib object: trustm_lib/pal/linux/pal_logger.c
gcc -c -I trustm_lib/optiga/include -I trustm_lib/optiga/include/optiga -I trustm_lib/optiga/include/optiga/ifx_i2c -I trustm_lib/optiga/include/optiga/comms -I trustm_lib/optiga/include/optiga/common -I trustm_lib/optiga/include/optiga/cmd -I trustm_lib/optiga/include/optiga/pal -I trustm_lib/pal/linux -I trustm_helper/include -I trustm_engine -Wall -DENGINE_DYNAMIC_SUPPORT trustm_lib/pal/linux/pal_logger.c -o trustm_lib/pal/linux/pal_logger.o
+++++++ Generating lib object: trustm_lib/pal/linux/pal_os_datastore.c
gcc -c -I trustm_lib/optiga/include -I trustm_lib/optiga/include/optiga -I trustm_lib/optiga/include/optiga/ifx_i2c -I trustm_lib/optiga/include/optiga/comms -I trustm_lib/optiga/include/optiga/common -I trustm_lib/optiga/include/optiga/cmd -I trustm_lib/optiga/include/optiga/pal -I trustm_lib/pal/linux -I trustm_helper/include -I trustm_engine -Wall -DENGINE_DYNAMIC_SUPPORT trustm_lib/pal/linux/pal_os_datastore.c -o trustm_lib/pal/linux/pal_os_datastore.o
+++++++ Generating lib object: trustm_lib/pal/linux/pal_os_event.c
gcc -c -I trustm_lib/optiga/include -I trustm_lib/optiga/include/optiga -I trustm_lib/optiga/include/optiga/ifx_i2c -I trustm_lib/optiga/include/optiga/comms -I trustm_lib/optiga/include/optiga/common -I trustm_lib/optiga/include/optiga/cmd -I trustm_lib/optiga/include/optiga/pal -I trustm_lib/pal/linux -I trustm_helper/include -I trustm_engine -Wall -DENGINE_DYNAMIC_SUPPORT trustm_lib/pal/linux/pal_os_event.c -o trustm_lib/pal/linux/pal_os_event.o
+++++++ Generating lib object: trustm_lib/pal/linux/pal_os_lock.c
gcc -c -I trustm_lib/optiga/include -I trustm_lib/optiga/include/optiga -I trustm_lib/optiga/include/optiga/ifx_i2c -I trustm_lib/optiga/include/optiga/comms -I trustm_lib/optiga/include/optiga/common -I trustm_lib/optiga/include/optiga/cmd -I trustm_lib/optiga/include/optiga/pal -I trustm_lib/pal/linux -I trustm_helper/include -I trustm_engine -Wall -DENGINE_DYNAMIC_SUPPORT trustm_lib/pal/linux/pal_os_lock.c -o trustm_lib/pal/linux/pal_os_lock.o
+++++++ Generating lib object: trustm_lib/pal/linux/pal_os_timer.c
gcc -c -I trustm_lib/optiga/include -I trustm_lib/optiga/include/optiga -I trustm_lib/optiga/include/optiga/ifx_i2c -I trustm_lib/optiga/include/optiga/comms -I trustm_lib/optiga/include/optiga/common -I trustm_lib/optiga/include/optiga/cmd -I trustm_lib/optiga/include/optiga/pal -I trustm_lib/pal/linux -I trustm_helper/include -I trustm_engine -Wall -DENGINE_DYNAMIC_SUPPORT trustm_lib/pal/linux/pal_os_timer.c -o trustm_lib/pal/linux/pal_os_timer.o
+++++++ Generating lib object: trustm_lib/pal/pal_crypt_openssl.c
gcc -c -I trustm_lib/optiga/include -I trustm_lib/optiga/include/optiga -I trustm_lib/optiga/include/optiga/ifx_i2c -I trustm_lib/optiga/include/optiga/comms -I trustm_lib/optiga/include/optiga/common -I trustm_lib/optiga/include/optiga/cmd -I trustm_lib/optiga/include/optiga/pal -I trustm_lib/pal/linux -I trustm_helper/include -I trustm_engine -Wall -DENGINE_DYNAMIC_SUPPORT trustm_lib/pal/pal_crypt_openssl.c -o trustm_lib/pal/pal_crypt_openssl.o
+++++++ Generating lib object: trustm_lib/pal/linux/target/rpi3/pal_ifx_i2c_config.c
gcc -c -I trustm_lib/optiga/include -I trustm_lib/optiga/include/optiga -I trustm_lib/optiga/include/optiga/ifx_i2c -I trustm_lib/optiga/include/optiga/comms -I trustm_lib/optiga/include/optiga/common -I trustm_lib/optiga/include/optiga/cmd -I trustm_lib/optiga/include/op [... cut ...]
******* Linking bin/trustm_engine.so
gcc -Lbin -Wl,-Rbin -ltrustm -lpthread -lssl -lcrypto -lrt trustm_engine/trustm_engine_rsa.o trustm_engine/trustm_engine_ec.o trustm_engine/trustm_engine.o trustm_engine/trustm_engine_rand.o -shared -o bin/trustm_engine.so
```
</details>

If this might help, we had in the past also a [port on GCC 8.3](https://github.com/Infineon/optiga-trust-m/pull/29/files) on some other yocto based platform. In order to build the setup they had o change slightly the core library.

ayushev, 2021-02-23:
any progress?

Kitty-Hawk1, 2021-02-25:
Yes, those patches do seem to improve things. Bit more testing required.

thanks.

ayushev, 2021-04-27:
SHould we close it? Does it work?

Kitty-Hawk1, 2021-04-27:
Yes it works and can be closed

closed 2021-04-27 by Kitty-Hawk1