Infineon/optiga-trust-m #50: pal: linux: open failure is not handled

svenschwermer, 2021-04-17:
When `open` in `pal_i2c_init` (pal/linux/pal_i2c.c) fails, this error is not handled. Even when handling the error, calling `optiga_util_open_application` freezes. This is what I changed in order to handle the error:

```diff
/usr/src/trustm_lib # git diff -- pal/linux/pal_i2c.c
diff --git a/pal/linux/pal_i2c.c b/pal/linux/pal_i2c.c
index 4d6d5ad..c1c7ba7 100644
--- a/pal/linux/pal_i2c.c
+++ b/pal/linux/pal_i2c.c
@@ -38,6 +38,7 @@
 #include <fcntl.h>
 #include <sys/ioctl.h>
 #include <unistd.h>
+#include <errno.h>
 
 #include "optiga/pal/pal_i2c.h"
 #include "pal_linux.h"
@@ -153,33 +154,47 @@ void i2c_master_arbitration_lost_callback(void)
 
 pal_status_t pal_i2c_init(const pal_i2c_t* p_i2c_context)
 {
-       int32_t ret = PAL_I2C_EVENT_ERROR;
-       pal_linux_t *pal_linux;
+       pal_status_t status = PAL_STATUS_FAILURE;
+       pal_linux_t *pal_linux = (pal_linux_t *) p_i2c_context->p_i2c_hw_config;
+    LOG_HAL("pal_i2c_init\n");
        do
        {
-               pal_linux = (pal_linux_t*) p_i2c_context->p_i2c_hw_config;
                pal_linux->i2c_handle = open(pal_linux->i2c_if, O_RDWR);
-               LOG_HAL("IFX OPTIGA TRUST X Logs \n");
-
+        if (pal_linux->i2c_handle == -1)
+        {
+                       LOG_HAL("open returned an error = %d (%s)\n", errno, strerror(errno));
+            break;
+        }
+
                // Assign the slave address
-               ret = ioctl(pal_linux->i2c_handle, I2C_SLAVE, p_i2c_context->slave_address);
-               if(PAL_STATUS_SUCCESS != ret)
+               int ret = ioctl(pal_linux->i2c_handle, I2C_SLAVE, p_i2c_context->slave_address);
+               if (ret == -1)
                {
-                       LOG_HAL((uint32_t)pal_linux->i2c_handle, "ioctl returned an error = ", ret);
+                       LOG_HAL("ioctl returned an error = %d (%s)\n", errno, strerror(errno));
                        break;
                }
-
-               //start_transceive_thread();
-       }while(0);
-    return ret;
+
+               status = PAL_STATUS_SUCCESS;
+       } while(0);
+    return status;
 }
 
 
 pal_status_t pal_i2c_deinit(const pal_i2c_t* p_i2c_context)
 {
-       LOG_HAL("pal_i2c_deinit\n. ");
-
-    return PAL_STATUS_SUCCESS;
+    pal_status_t status = PAL_STATUS_SUCCESS;
+    pal_linux_t *pal_linux = (pal_linux_t *) p_i2c_context->p_i2c_hw_config;
+       LOG_HAL("pal_i2c_deinit\n");
+       if (pal_linux->i2c_handle >= 0)
+    {
+        int ret = close(pal_linux->i2c_handle);
+        if (ret == -1)
+        {
+                       LOG_HAL("close returned an error = %d (%s)\n", errno, strerror(errno));
+            status = PAL_STATUS_FAILURE;
+        }
+    }
+    return status;
 }
```

svenschwermer, 2021-04-17:
It furthermore looks like `p_i2c_context->upper_layer_event_handler == NULL` isn't handled everywhere. That, however, is the case for the 2 included linux targets (rpi, ultra96).

svenschwermer, 2021-04-17:
I believe, the hanging might be caused by `ifx_i2c_tl_event_handler` not being called when `ifx_i2c_prl_init` fails.

ayushev, 2021-04-19:
@svenschwermer 

thanks for the proposal of the code change, this will be merged to the corresponding sample PAL.
But I'm not sure wheteher I understood your latest comment.

svenschwermer, 2021-04-20:
> thanks for the proposal of the code change, this will be merged to the corresponding sample PAL.

I will send you a pull request if you don't mind.

> But I'm not sure wheteher I understood your latest comment.

That was a little unclear, let me elaborate:

1. `optiga_util_create` is called with a callback function set, 0 (SUCCESS) is returned
2. `optiga_util_open_application` is called, 0 (SUCCESS) is returned
3. Underneath and asynchronously, `pal_i2c_init` fails which makes `ifx_i2c_{pr,t}l_init` return a non-zero status (ERROR)
4. `ifx_i2c_init` which had invoked `ifx_i2c_{pr,t}l_init`, never reports this error to the upper layer
5. The util callback is never called. This is unexpected since we must know when the application has been opened.

The following patch fixes the issue, but I'm not sure whether it's the right place for such an invocation of the upper layer handler.

```diff
diff --git a/optiga/comms/ifx_i2c/ifx_i2c.c b/optiga/comms/ifx_i2c/ifx_i2c.c
index 7fbd679..6a2dbc6 100644
--- a/optiga/comms/ifx_i2c/ifx_i2c.c
+++ b/optiga/comms/ifx_i2c/ifx_i2c.c
@@ -347,6 +347,9 @@ _STATIC_H optiga_lib_status_t ifx_i2c_init(ifx_i2c_context_t * p_ifx_i2c_context
 #endif
     }
 
+    if (api_status != IFX_I2C_STACK_SUCCESS)
+        ifx_i2c_tl_event_handler(p_ifx_i2c_context, api_status, 0, 0);
+
     return (api_status);
 }
 /// @endcond
```

ayushev, 2021-04-27:
Just to give you an update. Thanks for the PR you have pushed, we are evaluating internally right now this change and let you know as soon as possible.

aoune-ayoub, 2024-11-04:
Merge request was closed.

closed 2024-11-04 by aoune-ayoub