Infineon/optiga-trust-m #52: issues trying to port trust M on STM32

Simon47h, 2021-04-21:
Hello,

I am trying to port trust M v3 on STM32. 
For first attempt I am using bare metal architecture (without OS). 

I followed the porting guide to implement the PAL : https://github.com/Infineon/optiga-trust-m/wiki/Porting-Guide
and I take a look to the old issue on the same topic  : https://github.com/Infineon/optiga-trust-m/issues/10   (@darsh_patel)  

I am testing `optiga_communication()` example to read the I2C state. (So far I write and I read only 00 00 00 00)
I am facing issue on the receiver function error is --> `((upper_layer_callback_t)(p_i2c_context->upper_layer_event_handler))
                                                       (p_i2c_context->p_upper_layer_ctx , PAL_I2C_EVENT_ERROR);`
Please find the main attached PAL files. 

Do you agree that we don't need OS to establish basic communication with the trust M ?
Could you please help me to solve this issue ?

Thank you in advance. 

[pal_gpio.txt](https://github.com/Infineon/optiga-trust-m/files/6350226/pal_gpio.txt)
[pal_i2c.txt](https://github.com/Infineon/optiga-trust-m/files/6350227/pal_i2c.txt)
[pal_ifx_i2c_config.txt](https://github.com/Infineon/optiga-trust-m/files/6350229/pal_ifx_i2c_config.txt)
[pal_os_event.txt](https://github.com/Infineon/optiga-trust-m/files/6350231/pal_os_event.txt)
[pal_os_timer.txt](https://github.com/Infineon/optiga-trust-m/files/6350232/pal_os_timer.txt)

ayushev, 2021-04-21:
Hello @momo47f ,

Thanks for comming here once you have a problem. First of all answering your question ,the OS is not required for the succesfull communication.

Also I had a look into your code, is this the one you are actually using? To avoid lots of messages I will write here several things which might help you:
1) I'm not sure whether pal_ifx_i2c_config.txt is actually in use, but it seems like the `optiga_pal_i2c_context_0` structure instance is commented out. It is very important for the communication, for instance it is given to the pal_i2c_write function as a parameter.
2) To see whether things are moving forward you can set a breakpoint in the folloiwng place:
  a) you need to make sure that `pal_os_event_trigger_registered_callback` function in the the `pal_os_event.txt` file is actually called regularly (actually quite often at the beggining)

momo47f, 2021-04-21:
Thank you for your prompt reply. 

1. Actually, the optiga_pal_i2C_context_0 in

momo47f, 2021-04-21:
Thank you for your prompt reply.

1. Actually, the `optiga_pal_i2c_context_0 ` was defined elsewhere (main.c). But now I it's defined in pal_ifx_i2c_config.txt.
2. I double checked the `pal_os_event_trigger_registered_callback ` it's never called (i can't understand why right now). 

Knowing that i have done unitary test for my I2C in pooling mode and timer handler in interrupt mode. Everything is working fine but once I move to `optiga_communication()` I got issues. 

The first issue I got : 
In pal_i2c.txt file I got error in transmission in`pal_i2c_write` function (I think because the slave (TrustM) doesn't acknowledge its address ? )

Thanks again for your help.

I updated the pal_i2c.txt, pal_os_event and pal_ifx_i2c_config.txt.

[pal_gpio.txt](https://github.com/Infineon/optiga-trust-m/files/6353349/pal_gpio.txt)
[pal_i2c.txt](https://github.com/Infineon/optiga-trust-m/files/6353350/pal_i2c.txt)
[pal_ifx_i2c_config.txt](https://github.com/Infineon/optiga-trust-m/files/6353351/pal_ifx_i2c_config.txt)
[pal_os_event.txt](https://github.com/Infineon/optiga-trust-m/files/6353352/pal_os_event.txt)
[pal_os_timer.txt](https://github.com/Infineon/optiga-trust-m/files/6353353/pal_os_timer.txt)

momo47f, 2021-04-22:
@ayushev  Have you an idea please why `pal_os_event_trigger_registered_callback ` is never called ? The `pal_os_event_start` neither

ayushev, 2021-04-22:
Sorry, you were talking from the begginign about that sample code `test_optiga_communication``, which just check whether i2c write/read are working by sending in pollin gmode 0x82 command to the chip and waiting for a reply.

>> In pal_i2c.txt file I got error in transmission in pal_i2c_write function (I think because the slave (TrustM) doesn't acknowledge its address ? )
 `pal_i2c_write` usually at first always returns NACk as it sleeps most of the time, the host side needs to send its command several times. Usually after several unsecesfull attempts 

Lets try another way. Please include [this example](example_optiga_crypt_random) in your source (call `example_optiga_crypt_random` from your main), make sure that this function `example_optiga_init` is called (actually if the code will gone through this funciton you have a good communication to the chip).
Put breakpoint in the `pal_os_event_trigger_registered_callback ` function.

Waiting for your reply.

momo47f, 2021-04-22:
Sorry but the link to the example of `example_optiga_crypt_random` is empty (No results matched your search)

ayushev, 2021-04-22:
sorry, wrong link, the right one is this https://github.com/Infineon/optiga-trust-m/blob/trust_m1_m3/examples/optiga/example_optiga_crypt_random.c

momo47f, 2021-04-23:
Hello,

I have integrated and tested the `example_optiga_crypt_random `  in while loop (the example_optiga_init is called).

I put a breakpoint to `pal_os_event_trigger_registered_callback`  it's called 
Here is the log in debug mode for first and second iteration :
      [optiga example]  : example_optiga_init
      [optiga example]  : Passed
      [optiga example]  : example_optiga_crypt_random
      [optiga example]  : Passed
      [optiga example]  : example_optiga_Deinit --> Hardfault 

Are these logs show that the comm with chip it works ? I can't see the call to the i2c write/read ? 

After second iteration I got stuck when timer is elapsed and `pal_os_event_trigger_registered_callback ` is called. Please see the attached pal_os_event.txt

[pal_os_event.txt](https://github.com/Infineon/optiga-trust-m/files/6364790/pal_os_event.txt)

ayushev, 2021-04-26:
Not sure why the Hardfault happens, maybe it depends on how you call the code, but "Passed" would look like a working sample and working communication.
To see in details what is sent back and forth you can enable logging here https://github.com/Infineon/optiga-trust-m/blob/trust_m1_m3/optiga/include/optiga/optiga_lib_config_m_v3.h#L135-L144 

>> I can't see the call to the i2c write/read ?

what is your expectation, how should you see this? Did you put a brackpoint there?

momo47f, 2021-04-26:
> what is your expectation, how should you see this? Did you put a brackpoint there?
I put a breakpoint in `pal_i2c_write` in pal_I2C.txt the transmit function is not called !
Indeed, I am doing some unitary test on I2C bus between MCU and trustMv3 withoiut using OPTIGA stack. For example :

Parameters used are:
Frequency : 100 KHz
slave address : 0x30
Addressing Mode : 7 bits
Data sent : 0x82

 `while(HAL_I2C_Master_Transmit(&I2cHandle, (uint16_t)i << 1, (uint8_t*)data_buffer, sizeof(data_buffer), 100)!= HAL_OK)
    {
         printf("Polling data... %d \n\r", i);
      /* Error_Handler() function is called when Timeout error occurs.
         When Acknowledge failure occurs (Slave doesn't acknowledge its address)
         Master restarts communication */
      if (HAL_I2C_GetError(&I2cHandle) != HAL_I2C_ERROR_AF)
      {
        Error_Handler();
      }
    }`

I have tested all the possible address from 0 to 127, the trust M doesn't answer any one of theme ? Please do you have any idea about this issue i think this is my root cause...

ayushev, 2021-04-26:
> I put a breakpoint in `pal_i2c_write` in pal_I2C.txt the transmit function is not called !
> Indeed, I am doing some unitary test on I2C bus between MCU and trustMv3 withoiut using OPTIGA stack. For example :

Did you put it at the time of that random example execution? In general if you dont have a succesfull communication it should not show the "Passed" statement. You can check the communication if you come back to that random example and enable the Logging in the config, or alternativly put a breakpoint [here](https://github.com/Infineon/optiga-trust-m/blob/trust_m1_m3/examples/optiga/example_optiga_crypt_random.c#L112) and make sure you come to it.
If it is the casr you can use all the OPTIGA PAI freely, it is uíntegrated.

> I have tested all the possible address from 0 to 127, the trust M doesn't answer any one of theme ? Please do you have any idea about this issue i think this is my root cause...

Most of the time OPTIGA sleeps, you need to send several times the address to get an acknowedgment. Have a look on this issue https://github.com/Infineon/optiga-trust-m/issues/39

momo47f, 2021-04-26:
Finally, It works for me. I succeed to make work the `optiga_communication()`. I had to reset the chip and wait for up to 15 ms. (It's not said clearly in the trustM datasheet).

I will keep going the test for the other OPTIGA. i will contact you if needed.

Thanks a lot for your support. It was useful.

closed 2021-04-26 by ayushev