Infineon/optiga-trust-m #58: How to prevent metadata update?

svenschwermer, 2021-06-17:
Hi,

I have a question regarding locking down access to data objects: I understand from the solution reference manual that access to the actual object can be restricted by setting change/read/execute AC together with LcsO. However, I don't understand what prevents a future metadata update? Couldn't I just undo the restricting AC?

Do you have a guide on which objects should have their life cycle states set to "op" before devices with Trust M leave the factory?

ayushev, 2021-06-17:
Hi @svenschwermer glad to see you back!
The AC have their own predefined conditions (SRM Table 74). Change is allowed unless the LcsO is less than Operational. For M3 however it is possible to reset the metadata and LcsO using Protected Update procedure, but that would work only if you explicitly allow this thorugh configuration, othwise noone can revert the metadata.

svenschwermer, 2021-06-17:
> The AC have their own predefined conditions (SRM Table 74). 

I'm looking at OPTIGA_Trust_M_Solution_Reference_Manual_v3.15.pdf but can't find anything related on page 74.

ayushev, 2021-06-17:
It should be in the Table 74

svenschwermer, 2021-06-17:
> It should be in the Table 74

My bad, found it now. Thanks!

closed 2021-06-17 by svenschwermer

ayushev, 2021-06-17:
Sorry, I overlooked your second question. YOu might probably something different, so my answer would be pretty straigt-forward, let me know if you menat something different.

>> Do you have a guide on which objects should have their life cycle states set to "op" before devices with Trust M leave the factory?
For the non-customised chips, the first Certificate (0xe0e0 ) and the corresponing private key (0xe0f0) should be moved to an operational state. 
Additionally, in case any of the AC configured in a way that presribes usage of a shielded communication we recommend to pair a device with you host system (as a result you will have a new binding secret) and then lock the Binging Secret Object Id on the SE  0xE140 by moving its state to operational. In that case, **if** the binding secret on the host system is lost, AC can't be met.

Other AC depend on you setup, in case you use other objects and they should not be modified in the filed better to have them operational.

svenschwermer, 2021-06-18:
That's what I was planning, thank you!