Infineon/optiga-trust-m #61: Secure connection

stefanomologni, 2021-06-17:
### Discussed in https://github.com/Infineon/optiga-trust-m/discussions/60

<div type='discussions-op-text'>

<sup>Originally posted by **stefanomologni** June 17, 2021</sup>
Dear all,
  I'm not sure about using Secure Connection with Optiga TRUST M library on EIA 232 chip.

Starting from scratch I'll share a secret with OPTIGA using datastore, during first boot. After that I can set a protection level and execute some operations. After that, if i restart OPTIGA chip, the protection level is manteined? The communication still secure?

After this reboot the host can access to data into EIA 232 without the shared secret?

There are some examples of code using this feature?

Best regards,
Stefano</div>

pwiegele, 2021-09-13:
Hi stefanomologni,

> Starting from scratch I'll share a secret with OPTIGA using datastore, during first boot.
Yes 

> After that I can set a protection level and execute some operations
correct

>  After that, if i restart OPTIGA chip, the protection level is manteined? 
exactly, the protection level (in the form of access rules) is written into NVM and is not lost during power down.

> The communication still secure?
Yes the rules that you have setup (for example: access to object only under Secure Connection) is not lost. 

> After this reboot the host can access to data into EIA 232 without the shared secret?
Yes, once the access rules are in place and shared secret is written to Trust M access is only permitted if the access rules are satisfied. THis means access is only possible if you have the Secure Connection established (assuming that is your access condition).

BR Paul

stefanomologni, 2021-09-13:
Thanks a lot!

________________________________

Stefano
Mologni
Firmware Design and Micro Area

R&D Division - Senior Firmware Developer

+39 0346 750 337
***@***.******@***.***><file://com-fs-01/aree_comuni/Firma/signature.html#>
www.comelitgroup.com<http:\\www.comelitgroup.com><http://www.comelitgroup.com><http://www.comelitgroup.com/>
        [http://www.comelit.it/firma/comelit.png]  <http://www.comelitgroup.com/>

[https://www.comelitgroup.com/firma/banner.jpg]<https://www.comelitgroup.com/it-it/cctv-advance-la-nuova-gamma-di-videosorveglianza-professionale-di-comelit>

NORME DI ACCESSO E DI COMPORTAMENTO PER ESTERNI<https://www.comelitgroup.com/it-it/norme-accesso-comportamento-esterni>

AVVISO DI CONFIDENZIALITÀ E PRIVACY
Questo messaggio ed i suoi eventuali allegati possono contenere informazioni confidenziali, di proprietà, legalmente protette. È destinato all'utilizzo esclusivo del destinatario sopra indicato; la natura privata e la confidenzialità non sono modificati da eventuali errori di trasmissione, pur avendo il mittente assunto tutte le misure di sicurezza idonee e preventive per evitarli. Il trattamento dei dati ivi contenuti, e quelli che verranno forniti in risposta, è realizzato in conformità alla normativa sulla privacy Reg 679/2016, potrete prendere visione della informativa anche visitando il sito www.comelitgroup.com ovvero sulle app Comelit. Rispondendo alla presente mail dichiarate di aver preso visione della richiamata informativa e di aver prestato il consenso al trattamento dei dati personali. Se il messaggio non è a lei destinato, non deve utilizzarlo, diffonderlo, copiarlo con qualunque mezzo, o intraprendere azioni basandosi su di esso. Se ha ricevuto questo messaggio per errore, la preghiamo di volerlo distruggere (unitamente ad eventuali copie dello stesso e suoi allegati) e di volerci cortesemente informare del fatto scrivendo al mittente.
CONFIDENTIALITY NOTICE AND PRIVACY
This message and its attachments (if any) may contain confidential, proprietary or legally privileged information and it is intended only for the use of the addressee named above. No confidentiality or privilege is waived or lost by any erroneous transmission, despite having taken all the security and preventative measures to avoid it. The processing of personal data contained therein, and those that will be provided in response, is made in compliance with the privacy regulation Reg 679/2016, you can also read the information by visiting www.comelitgroup.com or on Comelit apps. Responding to this email, you declare that you have read the aforementioned information and have given consent to the processing of personal data If you are not the intended recipient of this message, you are hereby notified that you must not use, disseminate, copy it in any form or take any action in reliance on it. If you have received this message in error, please, delete it (and any copies of it and attachment) and kindly inform the sender.

pwiegele, 2021-09-13:
welcome

closed 2021-09-13 by pwiegele