Infineon/optiga-trust-m #64: Optiga Trust M not detected on i2c scanner app.

markandpathak, 2021-06-24:
Hello,

We are using Optiga Trust M SLS32AIA chip for one of our project. Problem is that we can not find the chip on any I2C scanner app. We tried to read register values (For example registers: 0x82). We also checked hardware:

ayushev, 2021-06-24:
Hello @markandpathak 

where the i2c detect app is located, is it a linux machine? Then [this](https://github.com/Infineon/i2c-utils-optiga-trust) can be usefull.
Otherwise this [reply](https://github.com/Infineon/optiga-trust-m/issues/39#issuecomment-753621314) also might be helpfull

markandpathak, 2021-06-24:
Sorry for the incomplete details. I am using I2Ctools by ESP-IDF on a ESP32-S2 module which includes a command i2cdetect.  Also written a small code to try to read register 0x82. As per your previous answer we checked that the I2C address is shifted to 7 bit and read byte is appended correctly.

markandpathak, 2021-06-24:
I am also following #34 to test a working example azure-optga-trust-m provision_test_certificate. I was able to successfully build the code for esp32-s2 but I could not figure out where to provide GPIO numbers for SCL, SDA and RST as I have different GPIOs allocated for I2C.

ayushev, 2021-06-24:
Unless you have a specific usecase in mind, I would recomend you to use the framework, for instance make the azure example work  (what you are trying to do)

closed 2021-09-27 by ayushev