Infineon/optiga-trust-m #65: ECDH Key agreement with default non-volatile keys

hb-kaertech, 2021-08-13:
**Question/Issue**:

We want to use `optiga_crypt_ecdh` with a non-volatile key of the chip.

I see the following, using a `OPTIGA_ECC_CURVE_NIST_P_256` public key:

- `optiga_crypt_ecdh` on `OPTIGA_KEY_ID_E0F0` -> error 0x8024
- `optiga_crypt_ecdh` on `OPTIGA_KEY_ID_E0F1`,  `OPTIGA_KEY_ID_E0F2`,  `OPTIGA_KEY_ID_E0F3` - -> error 0x800B

My guess is `OPTIGA_KEY_ID_E0F0` does not have `OPTIGA_KEY_USAGE_KEY_AGREEMENT` usage set. And possibly that the other keys are not of type. NIST P-256.

Could you confirm our guess?
Is there a way/workaround to do ECDH with a nonvolatile P-256 keys?

**Context**

For reference, `examples/optiga/example_optiga_crypt_ecdh.c` works well.
We are using the `SLS32AIA010MK` chip.

**Where did you expect to find the answer?**

I was expecting to either find:
 - a function to dump the current type / usage of a key ID
 - a documentation of the pre-preprogrammed keys
I was looking for this information in `OPTIGA_Trust_M_Solution_Reference_Manual_v3.15.pdf` .

Thanks!

ayushev, 2021-08-13:
Hello @hb-kaertech 

> optiga_crypt_ecdh on OPTIGA_KEY_ID_E0F0 -> error 0x8024

you are right, the prepopulated by default key doesn't have the OPTIGA_KEY_USAGE_KEY_AGREEMENT usage set. You can cross check it if you [read the metadata](https://github.com/Infineon/optiga-trust-m/blob/64929318d8db6dca0412e7c211a0ce5024e1fc4c/examples/optiga/example_optiga_util_read_data.c#L125-L133) and parse it, here is an example of the metadata which you most probably have, you can become familiar with parsing in SRM (Table 74):
```c
uint8_t e0f0_metadata[] = {
0x20,  // A Tag meaning beginning of the metadata 
        0x0f,  // Overall length of the metadata 
               0xc0, 0x01, 0x01,   // Lifecycle State (Lcso) is creation
               0xd0, 0x01, 0xff,   // Change Access Condition = never (NEV)
               0xd3, 0x01, 0x00,   // Execute Access Condition = always (ALW), execute means whether the chip is allowed to use it internally; e.g. for ecdh or ecdsa operations
               0xe0, 0x01, 0x03,   // Key algorithm associated with the key = NIST P256
               0xe1, 0x01, 0x01    // Key usage associated with the key = Authentication
};
```

> optiga_crypt_ecdh on OPTIGA_KEY_ID_E0F1, OPTIGA_KEY_ID_E0F2, OPTIGA_KEY_ID_E0F3 - -> error 0x800B

Please have a look on the [error description](https://github.com/Infineon/optiga-trust-m/wiki/Device-Error-Codes). "Another example is a usage of the optiga_crypt_ecdh() and optiga_crypt_tls_prf_sha256() functions in the row using the Session OID without optiga_crypt_ecc_generate_keypair(), this leads to failure "of out of sequence" due to a lack of private key in Session OID slot". It can be interpreted, that you don't have a private key in this [nonvolatile slot](https://github.com/Infineon/optiga-trust-m/wiki/Data-and-Key-Store-Overview), as by default only the 0xE0E0/0xE0F0 (a certificate and a private key) are populated. If you like you can populate this slot by calling a ['optiga_crypt_ecc_generate_keypair()'](https://infineon.github.io/optiga-trust-m/optiga__crypt_8h.html#a960387f95d10c5987e18b8eb67ca6197) with a given key_type (can be ORed), alternativly this slot can be provisioned during production.

> Is there a way/workaround to do ECDH with a nonvolatile P-256 keys?

yes, you need to either generate a new keypair (see above) on your side (an alternative would be to use our [python library](https://github.com/Infineon/python-optiga-trust) to automate this process) with required Key Usage, or configure this during the production on our side.

Let me know if that was helpful or if you need further support.

hb-kaertech, 2021-08-16:
@ayushev : All very clear. I was able to finish my PoC. Thanks!

closed 2021-08-16 by hb-kaertech