Infineon/optiga-trust-m #68: close application - with hardware RESET pin low

RaymWong, 2021-09-08:
**Question/Issue**:
The close application API once executed, will set the HW reset pin to LOW. For applications that are using HW reset pin, after perform some crypto operation(SEC counter > 0), will make the SEC counter not reducing after IDLE > tmax as the RESET pin is held low.
@pwiegele

pwiegele, 2021-09-13:
Hi RaymWong,

I would have 2 suggestions:
1. Close application shall be changed not to pull HW reset pin low.
2. As per default disable HW reset pin the the hostcode stack.

BR Paul

RaymWong, 2021-09-13:
> Hi RaymWong,
> 
> I would have 2 suggestions:
> 
> 1. Close application shall be changed not to pull HW reset pin low.
**[Q1]: Thanks a lot for your recommendation. Kindly advice or pin point the location in the source file of the host library to safely implement this solution.**
> 2. As per default disable HW reset pin the the hostcode stack.
[ANS]: https://github.com/Infineon/optiga-trust-m/blob/b172bcbd54571a5f4e006f299283b322099972a3/optiga/include/optiga/optiga_lib_config_m_v3.h#L114
> 
> BR Paul

sgsharath123, 2021-11-29:
Hi @RaymWong 

The Host side SW or the user Application shall take care of checking the SEC counter and ensuring it is counted down to 0, before calling the close application API. This is applicable in case OPTIGA_COMMS_DEFAULT_RESET_TYPE is 0U.

The above reset type will hold the OPTIGA in reset set via the HW pin and hence it is necessary that the Host side SW or user application ensures that OPTIGA is in a state where it can be shut down properly.

We will close the issue as this is already known to you from internal communication.

closed 2021-11-29 by sgsharath123