Infineon/optiga-trust-m #70: Metadata for changing access to registers on the optiga chip

KjetilSekse, 2021-10-04:
Hi. I am currently using the optiga chip as a security chip on a PCB-board for a big project. I have been reading through the libraries and documentation, but I dont find anywhere an explanation of metadata and what the different numbers indicate there but some through several examples. How do you for example set read/write access through writing metadata to an object etc? Also when I am trying to read from register 0xE0E0 after saving a generated ECC-key there I get no result and the error code: 0x0107 which means there was a presentation layer handshake error. I also get this exact same error when I use the example code from your github "example_read_data_from_optiga" with all relevant code through linking as well. Any ideas to what the cause of this error could be would be greatly appreciated. The error as written in the terminal can be seen below. Also a link to explanation of metadata for the optiga chip would be awesome (if it exists). Thanks in advance
[image]

KjetilSekse, 2021-10-04:
Hi, I found the document explaining metadata, and I believe I have found my problem. I put the optiga chip in operational mode for one runthrough, and after that the problem started. Is it is possible to put the relevant registers back in creational mode using optiga_util_write_metadata() with the correct metadata or is that chip now useless in the sense of testing?

ayushev, 2021-10-04:
Hi @KjetilSekse bringing back an object from an operational mode into a creational mode is possible only if a protected metadata update is configured, otherwise this isn't possible as one of protective mechanisms. 

If you would like to play around with metadata you might take an evaluation kit (or a RPi3/Embedded Linux) and use the [python library](https://github.com/Infineon/python-optiga-trust) for this. There is a supportive [documentation](https://infineon.github.io/python-optiga-trust/prov_mode.html) on how to bring your EvalKit into Provisioning mode. There is as well an experimental support for an `optigatrust` tool inside the package.
[image]

This might be also somehow usefull if you would like to play with metadata; as the tool allows to define metadata in json format
For example:

```python
{
    "lcso": "creation",
    "change": [
        "lcso",
        "<",
        "operational"
    ],
    "execute": "always",
    "algorithm": "secp256r1",
    "key_usage": [
        "authentication",
        "signature"
    ]
}
```

KjetilSekse, 2021-10-05:
Thanks for the quick and great answer! I also have another problem now I was hoping you could help out with. In the key generation stage we want to store both the private and public key on the optiga chip. I am as of now writing the public key to a free "ECC-key" slot (0XE0F1). The metadata is set as follows: const uint8_t metadata[] = {0x20, 0x0C, // Length of metadata excluding this line in elements (0x09)
     0xc0, 0x01, 0x01, //  State of the register is creation
     0xE1, 0xFB, 0x07, // set key container access as read in creation and init
     0xD0, 0x01, 0x00,   // Always allow to write to this register: WRITE = ALW
     0xD1, 0x01, 0x00, // Always allow to read from this register: READ = ALW 
     };  
The problem however is although I mean I have set read access properly, I get nothing out from reading it but zeros. I do manage to read other objects such as the ceritifcate at address 0XE0E0. Is it not possible to write a public key generated using ECC_generate_keypair to one of the keyslots: 0XE0F0-0XE0F3? How would I go about storing it on the optiga chip if that is the case in your opinion? Also is my metadata correctly set for if I want to be able to both change, and read the data stored in the relevant memory address?

ayushev, 2021-10-05:
The Private Key slot (0xe0f0-0xe0f3 + 0xe0fc, 0xe0fd for RSA) are special objects, which can't be updated unless a protected data update is confiured for that object. A User also is never allowed to read this private key, only use within the chip.

In general the flow is that you [generate a keypair](https://infineon.github.io/optiga-trust-m/optiga__crypt_8h.html#a960387f95d10c5987e18b8eb67ca6197) using one of the key objects as a parameter and then **don#t export the private key**, but only the public. In this case the Private key will be stored inside the key slot and the public could be stored in a different data object.
 
Answering your direct questions:

> Is it not possible to write a public key generated using ECC_generate_keypair to one of the keyslots: 0XE0F0-0XE0F3?
A keyslot contains only a private key, not public and a host can either send an instruction to the chip to use that key (for example to generate a signature), or can send a command to generate a new keypair.

> Also is my metadata correctly set for if I want to be able to both change, and read the data stored in the relevant memory address?
As I already mentioned you can't write directly into a key object (only as part of the protected data update procedure), but to help you with metadata, here is a corrected version of it if what you were looking for would be possible:

```
const uint8_t metadata[] = {
0x20, 
    0x0C, // Length of metadata excluding this line in elements
        0xc0, 0x01, 0x01, // State of the register is creation (ay: this is required only if you would like to change, if the object is already in this state you can omit this line)
        0xE1, 0x01,  0x30, // 0xE1 is a key usage tag, see Solution Reference Manual ver. 3.15 page 107. 0x30 = 0x10 | 0x20, Signature and Key Agreement
        0xD0, 0x01, 0x00, // Always allow to write to this register: WRITE = ALW
        0xD1, 0x01, 0x00, // Always allow to read from this register: READ = ALW
};
```

I'm not quite sue what did you mean with the "set key container access as read in creation and init", ut it doesn't look like what I've written to you, please help to clarify.

closed 2021-10-12 by ayushev