Infineon/optiga-trust-m #81: ECDSA Sign and External Verify

tejaChintalapatiPD, 2021-11-25:
**Question/Issue**:

How to generate signatures, send and verify them externally?

**Context**

I would like to verify the ECDSA Signature generated by the Optiga in the cloud and have been trying to verify the same using openssl but am unable. Here is the code:

```
ECCKey_E0F1 = objects.ECCKey(0xE0F1)
ECCPublicKey_E0F1, _ = crypto.generate_pair(ECCKey_E0F1, curve = secp256r1, export = False)
#Write the public key at arbitraty data object
arbitraryDataObject = optiga.Object(0xF1DB)
arbitraryDataObject.write(ECCPublicKey_E0F1)
.
.
.
.
.
#Code in other function
#Reading ECC Public Key stored in the arbitrary data object
arbitraryDataObject = optiga.Object(0xF1DB)
ECCPublicKey_E0F1 = arbitraryDataObject.read()

#Building the message
messagePayload = {}
messagePayload["comms_ecc_curve"] = DEF_ECC_CURVE
messagePayload["comms_ecc_public_key"] = ECCPublicKey_E0F1.hex()

#Sign the time to let the cloud know that this message indeed originated from the device itself
messageTime = int(time.time())
deviceMACBytes = bytes(str(Eth0MacAddress.rstrip()), 'utf-8')

ECCKey_E0F1 = objects.ECCKey(0xE0F1)
ECDSASignature_E0F1 = crypto.ecdsa_sign(ECCKey_E0F1, deviceMACBytes)
messagePayload["ecdsa_time"] = ECDSASignature_E0F1.signature.hex()
messagePayload["ecdsa_time_algorithm"] = ECDSASignature_E0F1.algorithm
```

I would like to know the following:

1. Is this the correct way of generating the signature? I am not sure if I am sending the Public key and the signature correctly in the following sections of the code:
```
messagePayload["comms_ecc_public_key"] = ECCPublicKey_E0F1.hex()
messagePayload["ecdsa_mac"] = ECDSASignature_E0F1.signature.hex()
```
**Output of the program**
"comms_ecc_public_key":"3059301306072a8648ce3d020106082a8648ce3d03010703420004c48cfae491ec9e3f3c76ad7643f932b2787a94988edba9b9b18bb6ffb8af463049d15785f835a579096e7b4e879e761ce39b9315bdae94818d2ea395730bd36e"

"ecdsa_mac":"3045022100e0deb8b2623dded60864cdd77cb961ab711455eeb16b350440d82f3056cae5f702202bb2e4e076304ff836d2dcb4174943d4e848114ab0ba2fa4195e6be740cb12b2"

2. Assuming the first step is correct, how can I externally verify if this signature is correct?

###### THANKS!
###### Teja

ayushev, 2021-11-25:
Hello @tejaChintalapatiPD 
The values seems to be correct, maybe only one difference is that most of crypto libraries assume input data is of `bytes` or `bytearray` type, so it is rather should be `b'3059301306072a8648ce3d020106082a8648ce3d03010703420004c48cfae491ec9e3f3c76ad7643f932b2787a94988edba9b9b18bb6ffb8af463049d15785f835a579096e7b4e879e761ce39b9315bdae94818d2ea395730bd36e' `  and `b'3045022100e0deb8b2623dded60864cdd77cb961ab711455eeb16b350440d82f3056cae5f702202bb2e4e076304ff836d2dcb4174943d4e848114ab0ba2fa4195e6be740cb12b2'`

Also many crypto libraries require a specific structure to be used for the verification function to be called. Have a look [here](https://github.com/Infineon/python-optiga-trust/blob/9feadd6dba7de58b52f91ac754e1ee9c841a3c35/tests/test_ecdsa.py#L66-L79):

tejaChintalapatiPD, 2021-11-25:
Thanks for such a quick answer! Appreciate it very much.

I am trying to verify the signature generated through [pyca/cryptography](https://cryptography.io/en/latest/) package for Python. These are the steps I am following:

1. Get the private key and converting the same to PEM format using [this tool](https://holtstrom.com/michael/tools/hextopem.php)
2. Loaded the Key in PEM format, signature and message 

Here's my code:

```
from cryptography.hazmat.primitives import hashes, serialization
from cryptography.hazmat.primitives.asymmetric import ec

loaded_public_key = serialization.load_pem_public_key(
    b"-----BEGIN PUBLIC KEY-----\nMFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAExIz65JHsnj88dq12Q/kysnh6lJiO26m5sYu2/7i\
vRjBJ0VeF+DWleQlue06HnnYc45uTFb2ulIGNLqOVcwvTbg==\n-----END PUBLIC KEY-----\n",
)

print(loaded_public_key.verify(b"304502210091f6879a797eaf1e5a46c0b7cdf3b78653eb0989b92577767d718f503666b82002207f012ff422e41ef8d08095df61b0637c21d17c808e3e9cb520dfbd5e3a30eeaa", b'B8:27:EB:BA:A1:CC', ec.ECDSA(hashes.SHA256())))
```

**Public key in DER:**
3059301306072a8648ce3d020106082a8648ce3d03010703420004c48cfae491ec9e3f3c76ad7643f932b2787a94988edba9b9b18bb6ffb8af463049d15785f835a579096e7b4e879e761ce39b9315bdae94818d2ea395730bd36e

The signature verification is not successful. Any idea on what is it that I am doing incorrectly here?

ayushev, 2021-11-26:
@tejaChintalapatiPD 

ok, now I understand it better, thank you for the example. The problem here is that that function accepts bytes, but not as a hex string, rather raw bytes.
I used your routines, but slightly modified it for simplicity, but I hope you can integrate them into your setup:

```,console
C:\Users\yushev >python
Python 3.9.6 (tags/v3.9.6:db3ff76, Jun 28 2021, 15:04:37) [MSC v.1929 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import optigatrust as optiga
>>> from optigatrust import objects, crypto
>>> from cryptography.hazmat.primitives import hashes, serialization
>>> from cryptography.hazmat.primitives.asymmetric import ec
>>> ECCKey_E0F1 = objects.ECCKey(0xE0F1)
Loaded: liboptigatrust-uart-win-i686.dll
>>> ECCPublicKey_E0F1, _ = crypto.generate_pair(ECCKey_E0F1, curve = "secp256r1", export = False)
>>> arbitraryDataObject = optiga.Object(0xF1DB)
>>> arbitraryDataObject.write(ECCPublicKey_E0F1)
>>> ECCPublicKey_E0F1 = arbitraryDataObject.read()
>>> deviceMACBytes = b'B8:27:EB:BA:A1:CC'
>>> ECDSASignature_E0F1 = crypto.ecdsa_sign(ECCKey_E0F1, deviceMACBytes)
>>> loaded_public_key = serialization.load_der_public_key(ECCPublicKey_E0F1)
>>> print(loaded_public_key.verify(ECDSASignature_E0F1.signature, b'B8:27:EB:BA:A1:CC', ec.ECDSA(hashes.SHA256())))
None
>>> print(loaded_public_key.verify(ECDSASignature_E0F1.signature, deviceMACBytes, ec.ECDSA(hashes.SHA256())))
None
>>>
>>>
>>>
>>> # Here are some values for reference
>>> print(ECCPublicKey_E0F1)
bytearray(b'0Y0\x13\x06\x07*\x86H\xce=\x02\x01\x06\x08*\x86H\xce=\x03\x01\x07\x03B\x00\x04\xcb\xd9C\xbbrx\xe3\xcf\xc0\xa3[\xb2\x8f\\\xe0s\x85i\x01\xa9\x02\x9dA*\xeb\xc1\x87\xdc\x9c\xd6\x05;\x17GS\x1a\xf0\xd1\xe07\xbb;6\xa8>\x85\xc1\xc3d\xba\xad\x8eM\xd19\xd6|.\xe8\x8e\xc4\x8eYK')
>>> print(ECDSASignature_E0F1.signature)
b'0D\x02 \x08{Iv\xf8\xae5i\x888\xc6\x8c7\xdf\x98\xd39U \xbd\x86\x9f\xff\xab\xcb\xe5\x0c\x12\x13\xc3\x8eu\x02 \t0\xae\xaaN\xde\xc8\xc0W\x93\xf6\x80\x98\x99\x80\x9f\xdf\xca=|.\xb2\xa8\\\xd3\x15\nUm];w'
>>> print(ECDSASignature_E0F1.signature.hex())
30440220087b4976f8ae35698838c68c37df98d3395520bd869fffabcbe50c1213c38e7502200930aeaa4edec8c05793f6809899809fdfca3d7c2eb2a85cd3150a556d5d3b77
>>> print(ECCPublicKey_E0F1.hex())
3059301306072a8648ce3d020106082a8648ce3d03010703420004cbd943bb7278e3cfc0a35bb28f5ce073856901a9029d412aebc187dc9cd6053b1747531af0d1e037bb3b36a83e85c1c364baad8e4dd139d67c2ee88ec48e594b
```

tejaChintalapatiPD, 2021-11-26:
Bingo! it works!

Here's my code to verify the ECDSA on the server side.

```
from cryptography.hazmat.primitives import hashes, serialization
from cryptography.hazmat.primitives.asymmetric import ec

publicKeyString =  '3076301006072a8648ce3d020106052b8104002203620004e4c36efb45f25bbbb33eb3735500f5f3352c181bf19f9a9730d97d4e24eb6162e690c59770549e5c5ffb5de2717427bdc1484929b7595d71fc0ff0f0ba231b59b9be1a7a8c0ab6dd9da5f27c96c6f811e4dd4cfb53d514e7c9c569af6334c455'
publicKeyByteArray = bytes.fromhex(publicKeyString)

ECDSADeviceIDString = "3064023019e01a92748c2a19b0e2a84a171b440f217956357c3324fd32d140b09e364f737ef3d1abd8a0ee294d9d671c196c6855023030c6bf98006fabe080164a0e8fe7edcf2addc4b7b82dfd5d8c19fd7b63f29047f7bc57d10e3315643634684ad638306a"
ECDSADeviceIDByteArray = bytes.fromhex(ECDSADeviceIDString)

print("Public Key: {0}\n".format(publicKeyByteArray))
print("ECDSA Signature: {0}\n".format(ECDSADeviceIDByteArray))

loaded_public_key = serialization.load_der_public_key(publicKeyByteArray)
print(loaded_public_key.verify(ECDSADeviceIDByteArray, b'B8:27:EB:BA:A1:CC', ec.ECDSA(hashes.SHA384())))
```

Actually, the solution didn't work initially. Here's why:

1. A while ago, on the same OPTIGA Chip, I generated the 384 bit keys and stored the pulic key in the arbitrary data object.
2. Without clearing the objects, I created new keys, this time, 256 bit keys

So, when I do ECDSA(Even with your example), the signature is failing, I belive this is because the difference in key length. @ayushev your thoughts?

**Solution**
Everything's working after I generated the 384 bit keys and updated the arbitrary object.

ayushev, 2021-11-26:
yes, with your data even correctly formed (it should be anyway as you noticed in the raw bytes form) that also didn't work on my side, that is why I paste an example which I could prove with functions you would like to use.

The key length shouldn't be the case, as the signature you provided 
ASN.1 Sequence TAG: `30`
Length:                        `45`
Encoded r component:`02210091f6879a797eaf1e5a46c0b7cdf3b78653eb0989b92577767d718f503666b820` 
Encoded s component: `02207f012ff422e41ef8d08095df61b0637c21d17c808e3e9cb520dfbd5e3a30eeaa`

In case of the NIST P384 the signature is longer

Also the public key you have showed corresponds to the correct curve for the same reason - size.
Maybe it was a typo sometime back, at least except that thing with the `bytes` representation I don't see a problem

closed 2021-11-26 by ayushev

tejaChintalapatiPD, 2021-11-26:
👍 

Thanks for your help, @ayushev!