Infineon/optiga-trust-m #92: Support for RSASSA-PSS signature scheme

robhancocksed, 2022-06-10:
**Is your feature request related to a problem? Please describe.**

TLS 1.3 requires the use of the RSASSA-PSS signature scheme when using RSA keys, as opposed to the RSASSA-PKCS1-v1_5 scheme which is currently supported in the optiga-trust-m library (and presumably hardware). This causes issues when trying to use an application such as OpenVPN, which now defaults to TLS 1.3, and private keys stored on the HSM. Inside OpenSSL when the PSS signature is being generated (in pkey_rsa_sign), the PSS padding is being added and then the signing request is passed on with padding mode RSA_NO_PADDING, which pkcs11-helper converts into a mechanism value of CKM_RSA_X_509, which the Optiga PKCS#11 implementation doesn’t support. There seems to be no way to add such support and either do the RSASSA-PSS signature type on the HSM, or have it do a raw RSA operation where the PSS could be done externally (the RSA encrypt/decrypt functionality seems to assume RSAES PKCS1-v1_5 encryption scheme, not raw).

**Describe the solution you'd like**

RSASSA-PSS signing support could potentially be added in a firmware update, and then added to the Optiga Trust M library. Or potentially there is some way to do a "raw" RSA operation on the HSM and handle the PSS in software, although I do recall seeing a mention that allowing non-padded RSA operations from a potentially hostile client could be a potential security risk.

**Describe alternatives you've considered**

We are looking into using ECC keys rather than RSA to avoid this problem. The other alternative would be limit to TLS 1.2, however this does not seem a good idea moving forward for security reasons.

snehapra, 2022-06-14:
Hi @robhancocksed,

Optiga Trust M only supports RSASSA-PKCS1-v1_5 signature scheme and unfortunately, without a business case, it is unlikely that the firmware update for RSASSA-PSS will happen. 
As you have mentioned, using ECC keys instead of RSA is a better alternative.

aoune-ayoub, 2024-11-08:
I hope you managed to solve your issue using the alternative solution mentioned by @snehapra. Due to inactivity this issue will be closed, feel free to reach to us if the problem is not solved yet.

closed 2024-11-08 by aoune-ayoub