Infineon/optiga-trust-m #93: ECDSA signing takes too much time

aectaan, 2022-07-28:
**Prerequisites**

Can you reproduce the problem reliably?
YES
Did you check current release notes for known issues?
YES
If this is not the latest release, have you checked newer releases? 
YES
**Description**:
TIme to sign data depends on `optiga_crypt_ecdsa_sign()` calling frequency. If you call it relatively often (1 sec), time to sign increases every call, from 75ms to 5 sec with step of ~40ms. If you call this function every 10 sec, time to sign decreases with the same step. Digest is fixed. PAL implementation attached.

**Steps to Reproduce:**
main.c
```c
K_SEM_DEFINE(se_sem, 0, 1);
uint8_t data[32] = {0};
uint8_t signature[80];
uint16_t sig_len = sizeof(signature);

static void optiga_lib_callback(void *context, optiga_lib_status_t return_status)
{
	LOG_DBG("callback status 0x%X", return_status);
	cb_status = return_status;
	k_sem_give(&se_sem);
}

int secure_element_sign_data(const uint8_t *data, uint8_t data_len, uint8_t *signature,
			     uint16_t *signature_len)
{
	int err = 0;

	uint64_t now = k_uptime_get();
	err = optiga_crypt_ecdsa_sign(crypt_h, data, data_len, OPTIGA_KEY_ID_E0F0, signature,
				      signature_len);
	k_sem_take(&se_sem, K_FOREVER);
	uint64_t signing_time = k_uptime_delta(&now);

	LOG_INF("sign duration %lld", signing_time);
	if (err != 0) {
		LOG_ERR("Failed to sign data (%d)", err);
		return err;
	}
	return err;
}

while (1) {
		secure_element_sign_data(data, sizeof(data), signature, &sig_len);
		k_sleep(K_SECONDS(10));
}
```
**Expected Result:**
sign duration is constant and is around 80msec.
**Actual Result:**
```log
# Sign data every 1 sec
[00:00:11.817,108] <inf> optiga: sign duration 76
[00:00:12.899,536] <inf> optiga: sign duration 82
[00:00:13.982,116] <inf> optiga: sign duration 83
[00:00:15.058,746] <inf> optiga: sign duration 76
[00:00:16.141,662] <inf> optiga: sign duration 83
[00:00:17.217,987] <inf> optiga: sign duration 76
[00:00:18.296,020] <inf> optiga: sign duration 78
[00:00:19.374,176] <inf> optiga: sign duration 78
[00:00:20.450,866] <inf> optiga: sign duration 76
[00:00:21.528,167] <inf> optiga: sign duration 78
[00:00:22.642,547] <inf> optiga: sign duration 114
[00:00:23.761,901] <inf> optiga: sign duration 119
[00:00:24.919,830] <inf> optiga: sign duration 157
[00:00:26.111,083] <inf> optiga: sign duration 192
[00:00:27.347,076] <inf> optiga: sign duration 236
[... 115 lines cut ...]
[00:05:18.954,528] <inf> optiga: sign duration 429
[00:05:29.345,275] <inf> optiga: sign duration 391
[00:05:39.693,878] <inf> optiga: sign duration 348
[00:05:50.005,737] <inf> optiga: sign duration 312
[00:06:00.279,724] <inf> optiga: sign duration 274
[00:06:10.516,601] <inf> optiga: sign duration 237
[00:06:20.710,449] <inf> optiga: sign duration 194
[00:06:30.865,600] <inf> optiga: sign duration 155
[00:06:40.985,809] <inf> optiga: sign duration 120
[00:06:51.068,817] <inf> optiga: sign duration 83
[00:07:01.144,989] <inf> optiga: sign duration 76
[00:07:11.226,806] <inf> optiga: sign duration 81
```
**Frequency (1/1, 1/50, 1/100):**
1/1
**Build/Commit:**
f64c42da2cc5ac59825bb995df4f0499557a1708
**Target:**
SoC: nRF52840, OS: Zephyr v3.0.0.
**Host OS and Version:**
Linux egor-WRT-WX9 5.15.0-41-generic #44-Ubuntu SMP Wed Jun 22 14:20:53 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
**Compiler:**
arm-zephyr-eabi, SDK v0.14.2

[PAL.zip](https://github.com/Infineon/optiga-trust-m/files/9206247/PAL.zip)

ayushev, 2022-07-28:
Hello @aectaan 
this is an expected behavior as the [Security Monitor](https://github.com/Infineon/optiga-trust-m/wiki/Security-Monitor) is in play here. For customized samples that behavior can be changed (relaxed)

aectaan, 2022-07-28:
WOW! Thank you so much for lighting fast response. Will check it now.

ayushev, 2022-07-28:
can we close the issue?

closed 2022-07-28 by aectaan