Infineon/optiga-trust-m #95: optiga_trust_init() - no init for the optiga_reset_0 pin

Youldboukhitine, 2022-09-13:
**Question/Issue**:

Hello, 
on the function "`void optiga_trust_init(void)`", 
the VDD pin  is initialize but not the reset pin. 

How should we consider this point, 
- the reset pin must be initialize outside the scope of the trust m component ? 
I modify the function to add the init of the reset pin 	`pal_gpio_init(&optiga_reset_0);`

Thanks

ayushev, 2022-09-13:
Hello @Youldboukhitine 

the optiga_trust_init() function calls a pal function implemented as part of your target board implementation, did you take on of the [existing](https://github.com/Infineon/optiga-trust-m/tree/develop/pal) or do you have your own? Anyway this pal initialization routine pretty much depends on the hardware you use, whether you have RST and VDD connected to the board

Youldboukhitine, 2022-09-13:
@ayushev Thanks for the feedback. 

I had a look on the current code and my implementation there are totally different (I may get an older version).
the call to pal_init should be fine now. 

I will close the issue

closed 2022-09-13 by Youldboukhitine