Infineon/optiga-trust-m #97: Suspicious typo in pal/linux.pal_i2c.c pal_i2c_init about LOG_HAL

versaloon, 2022-12-01:
```c
pal_status_t pal_i2c_init(const pal_i2c_t* p_i2c_context)
{
	int32_t ret = PAL_I2C_EVENT_ERROR;
	pal_linux_t *pal_linux;
	do
	{
		pal_linux = (pal_linux_t*) p_i2c_context->p_i2c_hw_config;
		pal_linux->i2c_handle = open(pal_linux->i2c_if, O_RDWR);
		LOG_HAL("IFX OPTIGA TRUST X Logs \n");
		
		// Assign the slave address
		ret = ioctl(pal_linux->i2c_handle, I2C_SLAVE, p_i2c_context->slave_address);
		if(PAL_STATUS_SUCCESS != ret)
		{
			LOG_HAL((uint32_t)pal_linux->i2c_handle, "ioctl returned an error = ", ret);
			break;
		}
		
		//start_transceive_thread();
	}while(0);
    return ret;
}
```

I originally think LOG_HAL is something like printf, but there is a LOG_HAL call in this function which accept i2c_handle as the first parameter:
```c
LOG_HAL((uint32_t)pal_linux->i2c_handle, "ioctl returned an error = ", ret);
```

Is it a typo?

ying-css, 2022-12-05:
@versaloon It is not typo, supposedly when you enable `printf(__VA_ARGS__)` in `#define LOG_HAL(...) //printf(__VA_ARGS__)` , it can be used to print out the debug message. Thanks.

versaloon, 2022-12-05:
> ```c
> LOG_HAL
> ```

Yes, I try the same as you said, but the problem is printf doesn't accept first parameter as uint32_t type. This is why I doubt it's a typo. Other calls to LOG_HAL is good.

versaloon, 2022-12-05:
> @versaloon It is not typo, supposedly when you enable `printf(__VA_ARGS__)` in `#define LOG_HAL(...) //printf(__VA_ARGS__)` , it can be used to print out the debug message. Thanks.

Yes, I try the same as you said, but the problem is printf doesn't accept first parameter as uint32_t type. This is why I doubt it's a typo. Other calls to LOG_HAL is good.

ying-css, 2022-12-05:
@versaloon Yes, You are right. We are evaluating this part now and may modify this for better debugging message printout/error handling. Thanks a lot.

closed 2022-12-06 by versaloon