Infineon/optiga-trust-m #98: Trust M V3, Example fails with error code 0x8010

bebroep, 2022-12-02:
Hello,

I have ported the OPTIGA library to STM32H7 to evaluete Trust M using the OPTIGA shell. It worked fine for a while but while debugging I2C and stepping the source code (IAR using JLink) at some point in time the protected update example does not work any more. When I run the example using the OPTIGA shell it fails with device error code 0x8010. All other V3 examples work fine.

Text from he solution reference manual for error 0x10:
Invalid/Wrong Payload Version
The Payload Version provided in the Manifest is not greater than
the version of the target object, or the last update was
interrupted and the restarted/retried update has not the same
version.

Maybe I have interrupted an update?
How can I resolve this issue ?

Thanks
[image]

ShaikKarishma0, 2022-12-05:
Hi @bebroep,

For every protected update call, the manifest should be generated by changing (incrementing) the payload version.
Please refer to the [protected update] (https://community.infineon.com/t5/Blogs/OPTIGA-Trust-M-Protected-Update/ba-p/363385) blog

bebroep, 2022-12-05:
Hi @ShaikKarishma0,

very useful hint. The protected update example works now!
Thanks

closed 2022-12-05 by bebroep