Infineon/optiga-trust-m #109: E0E0 Certificate can't be converted with openssl

ayushev, 2023-09-26:
Hi @ayushev , thank you for the information, according to encrypt, we have the next issue:

We have already got the certificate from E0E2 and it doesn’t seem to be a valid certificate for us. Maybe it has to do with the format or something like that, because if we try with the contents of E0E0 from the json in gitlab, we receive the same error.

We get the bin contents of that record and try to verify the certificate’s content with this command:

openssl x509  -inform der -in contents.bin

We get this error:
asn1 encoding routines:asn1_check_tlen:wrong tag

Should we use a different format/command to verify the contents? 

Regards

_Originally posted by @DIEGGOMEZ in https://github.com/Infineon/optiga-trust-m/issues/108#issuecomment-1735002983_

ayushev, 2023-09-26:
Hi @DIEGGOMEZ,

the public key certificate has by default a [TLS Identity Tags](https://github.com/Infineon/optiga-trust-m/blob/develop/documents/OPTIGA%E2%84%A2%20Trust%20M%20Solution%20Reference%20Manual.md#common-data-structures) which means the first 9 bytes can be skipped. The rest will be a usual X.509 certificate understood by the openssl.

Please also have a look on the option to work with optiga (for provisioning or data manipulation) using our Python Module - [optigatrust](https://github.com/Infineon/python-optiga-trust/tree/develop) ([pypi repo](https://pypi.org/project/optigatrust/)).
For that you can enable your Board to work as a bridge, see the figure where instead of PSoC your board can be used
[image]

you need simply to port two files ([main.c](https://github.com/Infineon/mtb-example-optiga-data-management/blob/master/source/main.c) + [provisioning.c](https://github.com/Infineon/mtb-example-optiga-data-management/blob/master/source/provisioning_sm.c) and the whole [project setup](https://github.com/Infineon/mtb-example-optiga-data-management/tree/master))

DIEGGOMEZ, 2023-09-26:
Hi @ayushev ,

As you said, we have keep your steps and certificate process now is sucessfull, thank you so much!

ayushev, 2023-09-26:
If the ticket is not relevant anymore (and the previous as well), please mark them as closed.

DIEGGOMEZ, 2023-09-26:
Hi @ayushev

I cannot close the ticket, because of you created it.

Regards.

closed 2023-09-26 by ayushev