Infineon/optiga-trust-m #111: Decrypt issue

DIEGGOMEZ, 2023-09-29:
Hi, 
I am triying to decrypt with my private key stored in 0xE0FD, but error 0x802E is occuring, here is my code:

		uint8_t encrypted_key[] = 
		{
		0x09,0xba,0x75,0x2f,0x5b,0x16,0xdc,0xd8,0xe6,0x40,0xfc,0x53,0x79,0xd5,0x07,0x1e,0x79,0xc8,0x3f,0xcc,0xbd,0x5e,0x35,0xc9,0x1c,0x58,0x7c,0x05,0x65,0x76,0xb7,0x55,0x6a,0x95,0x1e,0xca,0x8b,0x4b,0xbe,0xa9,0xa4,0x72,0xa1,0xad,0xd5,0x68,0x0e,0x5f,0x26,0x45,0xbc,0x86,0x30,0xa4,0xd4,0x53,0x0a,0xca,0x73,0x1e,0x06,0xde,0xa3,0xcb,0xb1,0x01,0x1d,0x46,0x35,0xac,0xa8,0xdf,0x08,0xd7,0xb6,0x9d,0x48,0x6f,0xdd,0x4f,0x15,0x85,0xde,0xeb,0x5f,0x78,0x4d,0x4f,0xb4,0x58,0x97,0xce,0xd2,0xec,0x13,0xe8,0xd7,0xdd,0xf9,0x2e,0xe6,0x49,0xd6,0xd5,0x3e,0x99,0x84,0x49,0x51,0xeb,0x82,0x4a,0x34,0x66,0xee,0x72,0xba,0x30,0x43,0x78,0x41,0xe8,0x27,0x59,0x0d,0x5f,0x97,0x56,0x6f,0x07,0x43,0xbb,0xe8,0x14,0x08,0xab,0x19,0xe0,0xad,0xc2,0xc3,0xfb,0x76,0x25,0xeb,0xc6,0x0b,0x7d,0x13,0xb1,0x4e,0x37,0x35,0xa4,0xd4,0x94,0x3f,0xf0,0xe7,0x93,0xb9,0x1e,0x24,0x9f,0xe5,0x37,0xdd,0x13,0x64,0x08,0x20,0xce,0x90,0x8f,0xd4,0x4d,0xd9,0x65,0xa9,0xc5,0xbd,0xa3,0x89,0x78,0x87,0xbd,0xf3,0x5e,0x56,0x78,0x3c,0x4d,0x59,0xb7,0x12,0x2f,0x5a,0x79,0xae,0xc3,0x9b,0x20,0x76,0x2e,0x47,0x96,0xe6,0x22,0xcf,0xc1,0x40,0xb5,0x63,0x59,0xd4,0x38,0xcc,0xec,0xbf,0x3f,0x9d,0x0b,0x78,0xc8,0x0f,0x36,0x3d,0x1f,0x47,0x74,0x6b,0x7d,0xe4,0x90,0xd3,0x4e,0x64,0x19,0xb4,0xb6,0xd4,0x3b,0x02,0x58,0x98,0xff,0x0e,0xd4,0x61,0x29,0x25,0x9c,0x22,0xa3,0x55,0x1b
};

HAL_StatusTypeDef status;
		optiga_lib_status_t return_status = 0;
		optiga_key_id_t optiga_key_id = 0xE0FD;
		optiga_util_t * util = NULL;
	  optiga_crypt_t * crypt = NULL;
		optiga_rsa_encryption_scheme_t encryption_scheme;
		uint8_t tries = 0;

		optiga_status = OPTIGA_LIB_BUSY;
		
//Read certificate
		do{
			util = optiga_util_create(OPTIGA_INSTANCE_ID_0,optiga_util_callback,NULL);
			optiga_status = optiga_util_open_application(util,0);	
			HAL_Delay(5000);

			tries++;
		}while(tries<2);
		
			tries = 0;
			optiga_status = OPTIGA_LIB_BUSY;
			encryption_scheme = OPTIGA_RSAES_PKCS1_V15;

			crypt = optiga_crypt_create(0, optiga_crypt_callback, NULL);
			HAL_Delay(5000);
			optiga_status = OPTIGA_LIB_BUSY;
			encryption_scheme = OPTIGA_RSAES_PKCS1_V15;
			optiga_status = optiga_crypt_rsa_decrypt_and_export(	crypt,
                                                            encryption_scheme,
                                                            encrypted_key,
                                                            sizeof(encrypted_key),
                                                            NULL,
                                                            0,
                                                            optiga_key_id,
                                                            decrypted_message,
                                                            &decrypted_message_length);

Any idea? 
Regards!

DIEGGOMEZ, 2023-09-29:
Anotation:

    static uint8_t decrypted_message[274] = {0};
    uint16_t decrypted_message_length = sizeof(decrypted_message);

koblibri, 2023-09-29:
Hi, 
the code seems to be correct. 
The reported error-code 0x802E indicates a "decryption failure", which should only happen, if the data could not be decrypted with the given key (key mismatch).

DIEGGOMEZ, 2023-10-02:
Hi @koblibri 

After these days, and multiple tries with different messages encryptions, i always have the same error, 802E, i can read public key from my certificate, but obviously(?) i cant read private key stored in E0FD to check that wverything is correct...

Any idea?

Regards!

DIEGGOMEZ, 2023-10-03:
Finally resolved, thank you.

closed 2023-10-03 by DIEGGOMEZ