Infineon/optiga-trust-m #117: example_optiga_crypt_rsa_generate_keypair   [optiga example]  : Failed with return value - 0x8007

ppradeepubi, 2024-02-06:
I have ported optiga pal layer for STM32L4 ( freeRTOS) , The following examples work fine for me "example_read_coprocessor_id", "example_optiga_crypt_random". So I hope the library is integrated fine. But, when I run "example_optiga_crypt_rsa_generate_keypair"
I get the following error => [optiga example]  : Failed with return value - 0x8007.

What could be the issue here ?

sss551, 2024-02-06:
Hi @ppradeepubi ,

The error code tells you that access conditions are not satisfied to generate the keypair. This may be due to the key slot Lcso might be in operational. 

Can you please share the metadata of the object that is used for generating key pair ?
 
Can you also try generating key pairs in other key slots?

Best regards,
Shiva Sai

sss551, 2024-02-07:
Hi @ppradeepubi ,

The metadata looks fine.
Does the metadata you have shared in your previous reply, is the response from the optiga trust m after executing optiga_util_read_metadata API ?
If not, you need to execute the optiga_util_read_metadata function for meta data of a object.
For more information, refer this [link](https://community.infineon.com/t5/Blogs/OPTIGA-Trust-M-Metadata/ba-p/363355).

Best regards,
Shiva Sai

ppradeepubi, 2024-02-07:
No, This is the meta data(const uint8_t E0FC_metadata[] = { 0x20, 0x06, 0xD0, 0x01, 0x00, 0xD3, 0x01, 0x00 };) is used in "example_optiga_crypt_rsa_generate_keypair" => "optiga_util_write_metadata" => and I get the error 8007 (access conditions are not satisfied ).

in "example_optiga_util_read_data" I can read data nd metadata from 0xE0E0 but not from 0xE0FC or E0FD. How to fix this issue. I want to test and get working "example_optiga_crypt_rsa_generate_keypair".

sss551, 2024-02-07:
Hi @ppradeepubi ,

Can you please tell what error you are getting when you tried to execute optiga_util_read_metadata for OID 0xE0FC in example_optiga_util_read_data.c file ?

Best regards,
Shiva Sai

ppradeepubi, 2024-02-07:
when I tried to execute "optiga_util_read_metadata" and "optiga_util_read_data" for OID 0xE0FC in example_optiga_util_read_data.c file, I get the following error => [optiga example] : Failed with return value - 0x8007.

=>[optiga example]  : example_optiga_init

[optiga example]  : Passed

[optiga example]  : example_optiga_util_read_data

[optiga example]  : Failed with return value - 0x8007

[optiga example]  : example_optiga_deinit

[optiga example]  : Passed

sss551, 2024-02-07:
Hi @ppradeepubi ,

Can you please share example_optiga_util_read_data.c file ?

Best regards,
Shiva Sai

ppradeepubi, 2024-02-07:
I am using this file 
https://github.com/Infineon/optiga-trust-m/blob/develop/examples/optiga/example_optiga_util_read_data.c

sss551, 2024-02-07:
Hi @sss551 ,

Can you please paste the below snippet of code before executing optiga_util_write_metadata function in example_optiga_crypt_rsa_generate_keypair.c file and share the values of read_data_buffer.
```c
uint16_t offset, bytes_to_read;
uint8_t read_data_buffer[1024];
optiga_oid = 0xE0FC;
        bytes_to_read = sizeof(read_data_buffer);
        optiga_lib_status = OPTIGA_LIB_BUSY;
        return_status = optiga_util_read_metadata(util_me,
                                                  optiga_oid,
                                                  read_data_buffer,
                                                  &bytes_to_read);

        WAIT_AND_CHECK_STATUS(return_status, optiga_lib_status);
        printf("\r\n");
        for (int i=0;i<bytes_to_read;i++){
         printf("%02x ",read_data_buffer[i]);
        }
        printf("\r\n");
``` 

I would like to read the metadata of the object.

Best regards,
Shiva Sai

ppradeepubi, 2024-02-07:
Hi @sss551 

when I paste the above code in "example_optiga_crypt_rsa_generate_keypair.c" before read_data_buffer I get this =>

[optiga example]  : example_optiga_crypt_rsa_generate_keypair

20 23 c0 01 07 d0 07 21 e0 ef fd 20 f1 db d3 07 20 e1 40 fd 23 f1 d4 d8 03 21 e0 ef e0 01 42 e1 01 02 f0 01 13

[optiga example]  : Failed with return value - 0x8007

sss551, 2024-02-08:
Hi @ppradeepubi ,
From the Metadata, I can see that the LCSO of the object is in operational. So you can't change the key in the data in the slot.
Please see the below screenshot from Solution reference Manual.

[image]

Now you need to change the metadata, but metadata cannot be changed because Lcso is operational.

In order to change metadata, you need to perform Protected Update for the keyslot. Please refer the article [here](https://community.infineon.com/t5/Blogs/OPTIGA-Trust-M-Protected-Update/ba-p/363385)for this.

Best regards,
Shiva Sai

ppradeepubi, 2024-02-08:
In the first place, if I had not used these slots before how did these LCSO became operational. Can you please share your thoughts.

sss551, 2024-02-08:
Hi @ppradeepubi ,

Can you please confirm whether you are using [OPTIGA™ Trust M](https://www.infineon.com/optiga-trust)Express variant ?
In case you are using [OPTIGA™ Trust M](https://www.infineon.com/optiga-trust) Express variant, the Lcso of 0xE0FC slot will be by default operational.
Please refer to the Comparison Table given [here](https://github.com/Infineon/optiga-trust-m?tab=readme-ov-file#comparison-table).
But you should be able to use the slot 0xE0FD.
Can you please share the Metadata of the OID 0xE0FD, similar to the procedure you have done previously for the OID 0xE0FC ? 

Best regards,
Shiva Sai

ppradeepubi, 2024-02-08:
Hi @sss551 

The metadata of 0xE0FD is 20 23 c0 01 07 d0 07 21 e0 ef fd 20 f1 db d3 07 20 e1 40 fd 23 f1 d4 d8 03 21 e0 ef e0 01 42 e1 01 02 f0 01 13

ppradeepubi, 2024-02-12:
Hi @sss551 , It a confidential document shared by Infineon. Can you provide you email so that I can mail ?

koblibri, 2024-02-13:
Hi @ppradeepubi, please reach out to me via email johannes.koblbauer@infineon.com and we can take it from there.
I will close this issue for now.

closed 2024-02-13 by koblibri