Infineon/optiga-trust-m #120: example_mbedtls_optiga_crypt_ecc_extendedcurve_calsign failure

ppradeepubi, 2024-03-01:
Hi,

I have the following examples work fine for me (TrustM on STM32)=>
example_mbedtls_optiga_crypt_ecc_extendedcurve_verify
example_mbedtls_optiga_crypt_ecc_extendedcurve_calcssec

But, this one fails =>
example_mbedtls_optiga_crypt_ecc_extendedcurve_calsign. error that I get is :

 TrustM util instance created.
TrustM crypt instance created.
 example_mbedtls_optiga_crypt_ecc_extendedcurve_calsign: using keyid E0F2
ubiwrap_optiga_crypt_ecc_generate_keypair: entry
authenticate: entryubiwrap_optiga_crypt_ecc_generate_keypair: key_oid=0xe0f2, curve_id = 0x5, key_usage=0x21
ááááááááááááL áááááááááááááááááááááááááááááL  áááááubiwrap_optiga_crypt_ecdsa_sign: entry
ubiwrap_optiga_crypt_ecdsa_sign has failed with error: 0x203
example_mbedtls_optiga_crypt_ecc_extendedcurve_calsign:590: error
Failed example_mbedtls_optiga_crypt_ecc_extendedcurve_calsign(), error: 0xffffb080
TrustM closed.

Please let me know your thoughts on this failure.

ppradeepubi, 2024-03-25:
Hi,

1. "optiga_crypt_ecdsa_sign has failed with error: 0x203"  -> key OID is 0xE0F2
2. mbedtls_ecdsa_sign is returning error from MBEDTLS_ERR_ECP_BAD_INPUT_DATA
3. key_usage 0x21 is the OR of 0x01 and 0x20.
4. code is almost is the same with just a wrapper written on top.

ShaikKarishma0, 2024-03-28:
Hi @ppradeepubi,

In the trustm_ecdsa.c the mbedtls_ecdsa_sign function is always using priv-key with OID 0xE0F0 - https://github.com/Infineon/optiga-trust-m/blob/develop/examples/mbedtls_port/trustm_ecdsa.c#L97
If your OID 0xE0F0 does not allow signatures, it may fail.

Can you please share your project so that we can replicate the issue on our side?

closed 2024-06-16 by ppradeepubi