Infineon/optiga-trust-m #121: AES ECB encryption does not provide expected result

Mangelmesa, 2024-03-06:
For a particular application, I am using the Optiga Trust M v3 to encrypt data using AES128 in ECB mode. To do this, first I've generated an AES128 Key with the Trust M v3, and I've exported it to be able to decrypt the data on another device. Then, using the Trust M function to encrypt data (optiga_crypt_symmetric_encrypt_ecb), I encrypt 16 bytes of data to obtain a 16 bytes ciphertext. The problem comes during the decryption, where the plaintext is not obtained when decrypting with anything other than the ECB decrypt function of the Optiga Trust M. That is, if I input the key and ciphertext generated by the Trust M into other crypto libraries  (tried with mbedTLS and PyCryptodome), the plaintext is not recovered.

I have also observed that, if I input the Trust M generated key and the plaintext in both mbedTLS and PyCryptodome, they return the same ciphertext, while the one obtained from the Optiga Trust M is completely different. Checking the ECB encryption function, it doesn't seem like the result should differ from the other libraries, as there is no IV or nonce, and the length of the plaintext is always 16 bytes, so no padding is added.

Is this the expected behavior? I supposed that, with the same symmetric key and ciphertext it would be possible to decrypt and obtain the plaintext back using another library. Am I missing something that the _optiga_crypt_symmetric_encrypt_ecb_ is doing that leads to a different result?

Any help would be greatly appreciated.

sss551, 2024-03-07:
Hi @Mangelmesa ,

Can you please tell how you are encrypting the plain text ?

I see that the function optiga_crypt_symmetric_encrypt_ecb accepts the symmetric key OID for encryption. It doesn't accept symmetric key from the user, only the key populated in the Symmetric Key slot. So the user keys cannot be used for the encryption. Please see the below screenshot for reference.
[image]

Best regards,
Shiva Sai

Mangelmesa, 2024-03-08:
Hi @sss551 ,

I generate the key once using optiga_crypt_symmetric_generate_key, following the example in example_optiga_crypt_symmetric_generate_key. Once the key is generated the function is not called again, and the encryption/decryption  is done by passing the corresponding OID (that should contain the key that was just generated), and the arrays with data and size, as you showed.

sss551, 2024-03-09:
Hi @Mangelmesa ,

In your previous post you have mentioned that you have exported the private key. If you export  it it will not store in the Optiga.
Please reler to the function definition below.
[image]

I think when you try to execute the API optiga_crypt_symmetric_encrypt_ecb, it is not using the generated key (may be it is using the random data in the Symmetric Key slot ). So the Cipher text that is generated is different.

Can you please try encrypting without exporting the key ?

Best regards,
Shiva Sai

Mangelmesa, 2024-03-09:
Hi @sss551,

I see my mistake now, somehow I didn’t get that if the key is exported it is not stored. I suppose that it is not possible to create a key and export it at the same time?  I have to use the protected update to input my own key into the symmetric key OID then, right? Thank you for your help clarifying this matter!

ShaikKarishma0, 2024-03-11:
I suppose that it is not possible to create a key and export it at the same time? I have to use the protected update to input my own key into the symmetric key OID then, right?
 -> Yes, your understanding is correct.

Mangelmesa, 2024-03-12:
Great, I got it to work by inputting my own AES 128 key in the corresponding OID of the Optiga Trust M. Thank you for your support!

closed 2024-03-12 by Mangelmesa