Infineon/optiga-trust-m #122: Optiga Trust M Express Authorization failed with return code 0x0107

SoftAvocado, 2024-04-12:
**Problem**
I am trying to perform Authorization for Optiga Trust M Express, but get the error 0x0107 from function optiga_crypt_hmac_verify. 

**Context**
Platform Binding Secret (0xE140) has the following metadata:
```
Raw:
20 1b c0 01 07 c4 01 40 c5 01 40 d0 07 20 e1 40
fd 23 f1 d0 d1 01 ff d3 01 00 e8 01 22

0xC0-LcsO
07
0xC4-Maximum size
40
0xC5-Used size
40
0xD0-Change access
20 e1 40 fd 23 f1 d0
0xD1-Read access
ff
0xD3-Execute access
00
0xE8-Object type
22
```
Autoref secret (0xF1D0) has the following metadata:
```
Raw:
20 1d c0 01 07 c4 01 8c c5 01 40 d0 07 20 e1 40
fd 23 f1 d0 d1 01 ff d3 03 20 e1 40 e8 01 31

0xC0-LcsO
07
0xC4-Maximum size
8c
0xC5-Used size
40
0xD0-Change access
20 e1 40 fd 23 f1 d0
0xD1-Read access
ff
0xD3-Execute access
20 e1 40
0xE8-Object type
31
```

I have Platform Binding Secret and Autoref secret for my Optiga as hexadecimal strings.

I'm using the following code to perform authorization (note: the keys in this example are mocked):

```c
uint8_t autoref_buffer[] = {0x96, 0x81, 0x02, 0x1F, 0x13, 0xE7, 0x43, 0x5B, 0xC1, 0x42, 0x29, 0x20, 0xD5, 0x2C, 0xBF, 0x79, 0xE5, 0x8C, 0xD1, 0x3E, 0xE3, 0x29, 0x5D, 0xAA, 0x24, 0xF8, 0x22, 0x93, 0x56, 0x38, 0x3F, 0x9F, 0xBD, 0x14, 0xEB, 0xC1, 0x4A, 0xA9, 0x35, 0x24, 0xE9, 0xF6, 0x60, 0x78, 0xC0, 0x96, 0x97, 0x4A, 0x06, 0x5F, 0x25, 0xDD, 0x05, 0x85, 0x8B, 0xAE, 0x98, 0x24, 0x4E, 0x46, 0x91, 0xA0, 0x35, 0x37};
uint32_t autoref_buffer_length = sizeof(autoref_buffer);

uint8_t pbs_buffer[] = {0x03, 0x91, 0x02, 0x1F, 0x13, 0xE7, 0x43, 0x5B, 0xC1, 0x42, 0x29, 0x20, 0xD5, 0x2C, 0xBF, 0x79, 0xE5, 0x8C, 0xD1, 0x3E, 0xE3, 0x29, 0x5D, 0xAA, 0x24, 0xF8, 0x22, 0x93, 0x56, 0x38, 0x3F, 0x9F, 0xBD, 0x14, 0xEB, 0xC1, 0x4A, 0xA9, 0x35, 0x24, 0xE9, 0xF6, 0x60, 0x78, 0xC0, 0x96, 0x97, 0x4A, 0x06, 0x5F, 0x25, 0xDD, 0x05, 0x85, 0x8B, 0xAE, 0x98, 0x24, 0x4E, 0x46, 0x91, 0xA0, 0x35, 0x37};
uint32_t pbs_buffer_length = sizeof(pbs_buffer);

// Set Pbs for secure connection
// This function makes the function pal_os_datastore_read read the passed secret.
DRV_OPTIGA_Status ret = DRV_OPTIGA_STATUS_RUNTIME_ERROR;
ret = pal_os_datastore_set_old_secret(pbs_buffer, pbs_buffer_length );

// Generate random data
const uint8_t arbitrary_data[] = {0xF0, 0xF1, 0xF2, 0xF3, 0xF4, 0xF5, 0xF6, 0xF7, 0xF8, 0xF9, 0xFA, 0xFB, 0xFC, 0xFD, 0xFE, 0xFF};
uint8_t optional_data[] = {0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0A, 0x0B, 0x0C, 0x0D, 0x0E, 0x0F, 0x10};
uint8_t random_data[32] = {0x00};
do
{
    g_optiga_lib_status = OPTIGA_LIB_BUSY;

    optiga_return_status = optiga_crypt_generate_auth_code(me_crypt, OPTIGA_RNG_TYPE_TRNG, optional_data, sizeof(optional_data), random_data, sizeof(random_data));
    WAIT_AND_CHECK_STATUS(optiga_return_status);
} while (FALSE);

// Calculate HMAC (hash-based message authentication code)
uint8_t input_data_buffer[64] = {0x00}; //optional_data || optiga_random ||  user arbitrary data
memcpy(input_data_buffer, optional_data, sizeof(optional_data));
memcpy(input_data_buffer+sizeof(optional_data), random_data, sizeof(random_data));
memcpy(input_data_buffer+sizeof(optional_data)+sizeof(random_data), arbitrary_data, sizeof(arbitrary_data));
uint8_t hmac_buffer[32] = {0x00};
optiga_hmac_type_t hmac_type = OPTIGA_HMAC_SHA_256;
uint16_t secret_oid = 0xF1D0;// default secret OID;
pal_crypt_mbedtsl_status = pal_crypt_hmac(hmac_type, autoref_buffer, autoref_buffer_length, input_data_buffer, sizeof(input_data_buffer), hmac_buffer);

// Perform authorization by verifying HMAC
do
{
    g_optiga_lib_status = OPTIGA_LIB_BUSY;

   OPTIGA_UTIL_SET_COMMS_PROTECTION_LEVEL(me_util, OPTIGA_COMMS_FULL_PROTECTION);
   OPTIGA_UTIL_SET_COMMS_PROTOCOL_VERSION(me_util, OPTIGA_COMMS_PROTOCOL_VERSION_PRE_SHARED_SECRET);

   OPTIGA_CRYPT_SET_COMMS_PROTECTION_LEVEL(me_crypt, OPTIGA_COMMS_FULL_PROTECTION);
   OPTIGA_CRYPT_SET_COMMS_PROTOCOL_VERSION(me_crypt, OPTIGA_COMMS_PROTOCOL_VERSION_PRE_SHARED_SECRET);

    optiga_return_status = optiga_crypt_hmac_verify(me_crypt, hmac_type, secret_oid, input_data_buffer, sizeof(input_data_buffer), hmac_buffer, sizeof(hmac_buffer));
    WAIT_AND_CHECK_STATUS(optiga_return_status);
} while (FALSE);
```
**Question**
So, it seems like the shielded connection (which is required to execute Autoref secret (0xF1D0), has been established, but I still get a handshake error. Could you please help me to figure out what am i doing wrong?

ShaikKarishma0, 2024-04-12:
Hi @SoftAvocado,

Before using the Shielded Connection, please ensure that the OPTIGA Trust M and the Host are paired using a shared secret (this secret key is used for the secure communication).
To share the secret, you have to execute the pair_host_and_optiga_using_pre_shared_secret() function. Please refer the existing example for details on this implementation - https://github.com/Infineon/optiga-trust-m/blob/develop/examples/optiga/usecases/example_pair_host_and_optiga_using_pre_shared_secret.c

SoftAvocado, 2024-05-15:
@ShaikKarishma0 thank you for response. It seems everything was correct in my implementation, but for some reason in the archive that came with OPTIGA Trust M there were pbs keys in auto_keys.txt and auto keys in PBS_keys.txt...

I have another question. I'd like to change metadata of Platform Binding Secret (0xE140), so it can be changed only in Shielded Connection (without authorization):
```
0xD0-Change access
20 e1 40
```

Since LcsO is already in Operational state, am I correct that I can only change metadata with  Confidentiality protected update?

ShaikKarishma0, 2024-05-15:
Hi @SoftAvocado,

Your understanding is correct. 
If the object is in operational state, you need to use the protected update method with necessary access conditions for the metadata update.
For Metadata update, Metadata Update Descriptor tag (0xD8) with the respective trust anchor OID must be present in the target OID metadata.
Please refer the following [blog](https://community.infineon.com/t5/Knowledge-Base-Articles/OPTIGA-TM-Trust-M-Protected-Update-KBA235464/ta-p/362531) for more details on the protected update process.

SoftAvocado, 2024-05-15:
Thanks a lot!

closed 2024-05-15 by SoftAvocado