Infineon/optiga-trust-m #124: Link for "OPTIGA™ Trust M Zephyr PAL & Application" on the main page is broken

farrellaultman, 2024-05-10:
**Prerequisites**

Can you reproduce the problem reliably? y

Did you check current release notes for known issues? y

If this is not the latest release, have you checked newer releases? 

**Description**:

A clear and concise description of what the bug is.

**Steps to Reproduce:**

1. Just click on the link for "OPTIGA™ Trust M Zephyr PAL & Application".

2. Page not found displayed

3.

**Expected Result:**
Zephyr example

**Actual Result:**
Page not found

**Frequency (1/1, 1/50, 1/100):**
1/1

**Build/Commit:**

Please specify the release, like 1.4.1. If there is no release, give us the commit for the code you used.

**Target:**

The chip or kit you are working with

**Host OS and Version:**

MacOS, Windows, Linux

**Compiler:**

Include the version!

**Environment**

Provide the context surrounding this problem. Perhaps an IDE, a debug probe, an RTOS, etc. (and versions)

###### THANKS!

aoune-ayoub, 2024-05-10:
We are aware of the missing link, the zephyr PAL & application will be released on Github very soon.

aoune-ayoub, 2024-05-21:
The zephyr application is now live and the link should send you to the repository where a zephyr application is present.

closed 2024-05-21 by aoune-ayoub