Infineon/optiga-trust-m #127: optiga_crypt_ecdsa_verify error for pre provisioned TrustM

ppradeepubi, 2024-06-16:
Hi.

I am working on a problem.

My TrustM chip is pre-provisioned with 
Private Key OID: E0F0 (ECC Key 1)
Public Key Certificate OID: E0E0 (Certificate 1)
Key Type: Elliptic Curve - NIST P 521
Certificate Format: TLS Identity Certificate

Length of Certificate (birth certificate ) at 0xE0E0 is around 879 byes.
I am trying to connect to my server ( Cipher SHA-256). During TLS handshake, we get 32 bytes hash message from the server.
This hash message is being sent to TrustM to sign using the E0F0 Private key, and I need to send this signature back to server to ensure that handshaking goes through. The Server fails saying signature mismatch.

Now, I am using the following API to sign = > optiga_crypt_ecdsa_sign(me_crypt, digest, digest_length, 0xE0F0, signature, signature_length);
Digest length = 32 bytes
Signature length received = 136 bytes (Elliptic Curve - NIST P 521)

For debugging, I would like to verify the signature using the following API => 
optiga_crypt_ecdsa_verify(me_crypt, digest, digest_length, signature, signature_length,OPTIGA_CRYPT_OID_DATA,0xE0E0);
I am getting following error message. 

OPTIGA_CMD_ERROR_MEMORY_INSUFFICIENT
APDU length formed is greater than OPTIGA_MAX_COMMS_BUFFER_SIZE.

Can you suggest what could be wrong with this approach?
My intention is to verify the signature over the given hash from Optiga itself, because the server says it’s a mismatch.

Regards,
Parvathi

koblibri, 2024-06-17:
Hi @ppradeepubi,

The “optiga_crypt_ecdsa_verify” function can only verify signatures via certificate objects, if the certificate object contains a single, DER encoded certificate (Data starts with 0x30)
In your case, you are saying that 0xE0E0 contains a “TLS Identity Certificate”. This would be the wrong format for verification. You could read the certificate, extract the public key on the host (e.g. via mbedTLS) and send it back to the OPTIGA for verification.

Regarding the potentially wrong signature (the root cause): We are aware of a bug within the host-library, which results in a potentially wrong ASN.1 encoding of P521 based signatures (#106). A fix for that will be released within the next few weeks.

ppradeepubi, 2024-06-20:
Hi @koblibri 

I am using these keys, digest and signature you provided to verify my API. 

I am getting the following error. Please see my attached code derived from ["example_optiga_crypt_ecdsa.c"]([example_optiga_crypt_ecdsa_verify.txt](https://github.com/user-attachments/files/15919971/example_optiga_crypt_ecdsa_verify.txt))

,[optiga example]  : example_optiga_init
[optiga example]  : pair_host_and_optiga_using_pre_shared_secret
[optiga example]  : Passed
[optiga example]  : Passed
[optiga example]  : example_optiga_crypt_ecdsa_verify
[optiga example]  : Failed with return value - 0x8005
[optiga example]  : example_optiga_deinit
[optiga example]  : Passed
[example_optiga_crypt_ecdsa_verify.txt](https://github.com/user-attachments/files/15919966/example_optiga_crypt_ecdsa_verify.txt)

Public Key:
        03 81 86 00 04 00 A8 9B E3 5F DE 80 14 32 D4 A4 
        DD E6 10 C4 0B 1B 49 91 91 91 5D 3F 65 D1 9E A1 
        FF 1C 90 6B 46 52 BF 7C BF 23 DF 7D ED 23 34 74 
        F8 28 86 75 45 D8 49 18 3D FB 8C 86 DA 0B E2 03 
        C9 A8 0E F6 6C 75 02 01 77 9C 9D 09 98 E7 09 77 
        2E F4 A7 D5 BD DC 5F 7F 11 C1 01 51 6E FA 23 D6 
        E8 95 50 27 26 16 FB 56 44 4A 84 AC A3 24 67 C7 
        4D CD 9D FD EE AC 48 E7 89 DE 04 3E 28 E8 FC 6F 
        4D F1 15 7B FC 8B 2A 7B 55 

Digest:
        0E B5 6B 44 BB BA E8 64 08 29 FF B6 F8 6C 71 C3 
        73 A3 13 10 92 E3 A3 76 7F 1B 01 61 76 53 14 2A 

Signature:
        02 42 01 73 18 DD CE EC C7 38 8F 24 6D 66 95 4D 
        45 2A 08 B9 CB 15 33 71 2B 5C F0 CA E7 CC 54 E0 
        9D E0 32 DE 1B 0B 3B 81 7D 24 B7 42 43 28 E3 3B 
        67 5B 7C E4 85 F7 33 8F C1 95 6F 99 14 BA 1B D4 
        21 D5 6C C3 02 42 00 86 02 A8 85 64 81 EF 85 2F 
        1F 26 0B 7D 48 D8 9F DA 4C CE 25 A1 83 8F 3B C6 
        E7 78 7E 1D 22 94 2C F0 63 A9 5B 4D 08 BE EB B0 
        68 03 7B 94 5E 38 02 73 A5 CE 14 79 62 4D A2 4A 
        EC 26 9E F0 A0 7B C0 BE

koblibri, 2024-06-26:
The bytes you have written do not match the bytes in your sourcecode-file. Please check again using the uint8_t arrays below: 
```c
//SHA-256 Digest
static const uint8_t digest [] =
{
    0x0E,0xB5,0x6B,0x44,0xBB,0xBA,0xE8,0x64,0x08,0x29,0xFF,0xB6,0xF8,0x6C,0x71,0xC3,
    0x73,0xA3,0x13,0x10,0x92,0xE3,0xA3,0x76,0x7F,0x1B,0x01,0x61,0x76,0x53,0x14,0x2A
};

// ECDSA Signature
static const uint8_t signature [] =
{
    0x02,0x42,
    0x01,0x73,0x18,0xDD,0xCE,0xEC,0xC7,0x38,0x8F,0x24,0x6D,0x66,0x95,0x4D,0x45,0x2A,
    0x08,0xB9,0xCB,0x15,0x33,0x71,0x2B,0x5C,0xF0,0xCA,0xE7,0xCC,0x54,0xE0,0x9D,0xE0,
    0x32,0xDE,0x1B,0x0B,0x3B,0x81,0x7D,0x24,0xB7,0x42,0x43,0x28,0xE3,0x3B,0x67,0x5B,
    0x7C,0xE4,0x85,0xF7,0x33,0x8F,0xC1,0x95,0x6F,0x99,0x14,0xBA,0x1B,0xD4,0x21,0xD5,
    0x6C,0xC3,
    0x02,0x42,
    0x00,0x86,0x02,0xA8,0x85,0x64,0x81,0xEF,0x85,0x2F,0x1F,0x26,0x0B,0x7D,0x48,0xD8,
    0x9F,0xDA,0x4C,0xCE,0x25,0xA1,0x83,0x8F,0x3B,0xC6,0xE7,0x78,0x7E,0x1D,0x22,0x94,
    0x2C,0xF0,0x63,0xA9,0x5B,0x4D,0x08,0xBE,0xEB,0xB0,0x68,0x03,0x7B,0x94,0x5E,0x38,
    0x02,0x73,0xA5,0xCE,0x14,0x79,0x62,0x4D,0xA2,0x4A,0xEC,0x26,0x9E,0xF0,0xA0,0x7B,
    0xC0,0xBE
};

//NIST-P521 Public Key
static const uint8_t ecc_public_key[] = {
    0x03,
    0x81,0x86,0x00,0x04,0x00,0xA8,0x9B,0xE3,0x5F,0xDE,0x80,0x14,0x32,0xD4,0xA4,0xDD,
    0xE6,0x10,0xC4,0x0B,0x1B,0x49,0x91,0x91,0x91,0x5D,0x3F,0x65,0xD1,0x9E,0xA1,0xFF,
    0x1C,0x90,0x6B,0x46,0x52,0xBF,0x7C,0xBF,0x23,0xDF,0x7D,0xED,0x23,0x34,0x74,0xF8,
    0x28,0x86,0x75,0x45,0xD8,0x49,0x18,0x3D,0xFB,0x8C,0x86,0xDA,0x0B,0xE2,0x03,0xC9,
    0xA8,0x0E,0xF6,0x6C,0x75,0x02,0x01,0x77,0x9C,0x9D,0x09,0x98,0xE7,0x09,0x77,0x2E,
    0xF4,0xA7,0xD5,0xBD,0xDC,0x5F,0x7F,0x11,0xC1,0x01,0x51,0x6E,0xFA,0x23,0xD6,0xE8,
    0x95,0x50,0x27,0x26,0x16,0xFB,0x56,0x44,0x4A,0x84,0xAC,0xA3,0x24,0x67,0xC7,0x4D,
    0xCD,0x9D,0xFD,0xEE,0xAC,0x48,0xE7,0x89,0xDE,0x04,0x3E,0x28,0xE8,0xFC,0x6F,0x4D,
    0xF1,0x15,0x7B,0xFC,0x8B,0x2A,0x7B,0x55
    };
```

closed 2024-07-03 by koblibri